arendst/Tasmota #6580: expressions in rules not working

th-s-op, 2019-10-06:
### BUG DESCRIPTION
Using Tasmota Version 6.6.0.15, fresh clone, with #define USE_RULES, #define USE_EXPRESSION and #define SUPPORT_IF_STATEMENT. Resulting firmware compiles with no errors and in overall works fine. But I can't use expressions in rules.

My first plead is to include a reference to how to use expressions into the "Rules" section of the Wiki. There is a new subpage for the IF statement mentioning but not describing expressions. I found a reference somewhere else and expected them to work like in the following example:

var1 1 + 1
but that results in
19:27:43 RSL: RESULT = {"Var1":"1 + 1"}

I found some hints here but that might be a completely different implementation:
https://git.koehlerweb.org/frodovdr/Sonoff-Tasmota/commit/4b4b3709ad1bfc1a78af205c4f48ebd8d6f77003

- is it not working for some reason? -> bug
- do I just not know how it should work -> documentation

Device: Blitzwolf / Gosund SP1 v23 Module.
Firmware: 6.6.0.15 self compiled with AtomPlatformIO, flashed with on-device Tasmota FW upgrade

Provide the output of command: ``Backlog Template; Module; GPIO``:
19:41:16 CMD: Backlog Template; Module; GPIO
19:41:17 RSL: RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
19:41:17 RSL: RESULT = {"Module":{"55":"Gosund SP1 v23"}}
19:41:17 RSL: RESULT = {"GPIO":"Not supported"}

Provide the output of this command: ``Status 0``:
  ```
  STATUS 0 output here:
19:42:13 CMD: status 0
19:42:13 RSL: STATUS = {"Status":{"Module":55,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
19:42:13 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:19:33","StartupUTC":"2019-10-06T17:22:40","Sleep":50,"CfgHolder":4617,"BootCount":1,"SaveCount":5,"SaveAddress":"F7000"}}
19:42:13 RSL: STATUS2 = {"StatusFWR":{"Version":"6.6.0.15(sonoff)","BuildDateTime":"2019-10-06T19:21:13","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)"}}
19:42:13 RSL: STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["hahn@opheys.de","hahn-w3@opheys.de"],"TelePeriod":0,"Resolution":"FEA181C0","SetOption":["00208001","280500000100060000005A64000000000000","00000000"]}}
19:42:13 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":448,"Free":552,"Heap":30,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":255,"Features":["00000809","8F082183","04100000","00000014","000000C0","00000000"],"Drivers":"1,2,3,4,6,9,10,21","Sensors":"2,3,4,5"}}
19:42:13 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-blitzwolf-1","IPAddress":"192.168.XXX.XXX","Gateway":"192.168.XXX.XXX","Subnetmask":"255.255.255.0","DNSServer":"192.168.4.1","Mac":"DC:4F:22:XX:XX:XX","Webserver":2,"WifiConfig":4}}
19:42:13 RSL: STATUS7 = {"StatusTIM":{"UTC":"Sun Oct 06 17:42:13 2019","Local":"Sun Oct 06 19:42:13 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":99,"Sunrise":"07:22","Sunset":"18:43"}}
19:42:13 RSL: STATUS9 = {"StatusPTH":{"PowerDelta":0,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
19:42:13 RSL: STATUS10 = {"StatusSNS":{"Time":"2019-10-06T19:42:13","ENERGY":{"TotalStartTime":"2019-10-06T19:22:39","Total":0.03011,"Yesterday":0.00000,"Today":0.03011,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":238,"Current":0.000}}}
19:42:13 RSL: STATUS11 = {"StatusSTS":{"Time":"2019-10-06T19:42:13","Uptime":"0T00:19:33","UptimeSec":1173,"Heap":30,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"XXX","BSSId":"E0:28:6D:XX:XX:XX","Channel":12,"RSSI":40,"LinkCount":1,"Downtime":"0T00:00:04"}}}  
  
  ```

meingraham, 2019-10-06:
Expressions are for the IF/ELSE conditional _rules_ statements only. I am pretty certain that arithmetic in expressions is not supported. This is why you **_don't_** see `+ - / *`  listed. Therefore, the expression syntax in the "subpage" is sufficient for the use of expressions. Arithmetic in commands (e.g., `var1 1 + 1`) is definitely not supported. All that does is set `var` to the string `1 + 1`.

th-s-op, 2019-10-06:
I disagree. Looking at the source code, I find accurences of "Add, Substract, Multiply" etc. operators (see below). And if you have a look at the if/else syntax, you find a reference to 

>\<expression\> {= | < | > | | | == | <= | >= | !=} \<expression\>

and there is no definition of exression. 

  #ifdef USE_EXPRESSION
  #include <LinkedList.h>                 // Import LinkedList library

  const char kExpressionOperators[] PROGMEM = "+-*/%^\0";
  #define EXPRESSION_OPERATOR_ADD         0
  #define EXPRESSION_OPERATOR_SUBTRACT    1
  #define EXPRESSION_OPERATOR_MULTIPLY    2
  #define EXPRESSION_OPERATOR_DIVIDEDBY   3
  #define EXPRESSION_OPERATOR_MODULO      4
  #define EXPRESSION_OPERATOR_POWER       5

From my point of view, there first should be a documentation of how and where expressions can be used, then I'll be able to check if my code is correct or not. Currently, I am only guessing.

th-s-op, 2019-10-06:
With a further look at the source code, mathematical expressions should be supported in RuleTimer, Var and Mem commands, too.
And checking the function CmndVariable in xdrv_10_rules.ino, I found the solution:

22:06:52 CMD: var1=1+1
22:06:52 RSL: RESULT = {"Var1":"2.0000000"}

=> Mathematical expressions are evaluated in a Var-Command in a rule if the "=" character is following the command!

=> should be documented if time permits.

Jason2866, 2019-10-06:
@th-s-op  thx for your findings! You are invited to add the infos to the wiki.
Tasmota is a Open Source project and every constructive help is highly welcome.

meingraham, 2019-10-06:
@th-s-op 

I agree with @Jason2866. We appreciate your efforts in researching this and would welcome you creating content (perhaps a separate page devoted to expressions). The wiki is always a work in progress and being refined. If you start the effort with expressions, I am certain that there will be further contributors in the near future (e.g. @laurentdong).

Mike

th-s-op, 2019-10-06:
@Jason2866 and @meingraham I got your message. While I completely agree to get part of the active community (and to be honest, I didn't know that the Wiki is open for everyone), I think it isn't a good idea in this case: I would have to completely "mentally reverse engineer" the source code to document the functionality. I am not blocking the effort, but I think the best, safest and most efficient way to document such a feature is to get it done by an author. I am not demanding this, just suggesting, appreciating all the authors' efforts and short time. There would be a high risk of me making errors due to wrong interpretation/understanding of the code.

Nevertheless: big thanks to the author(s) for the IF/ELSE + Expressions functionality to be included in Rules. This makes many of my more complex rules a lot simpler and more powerful and reduces "hacks" which made them close to unreadable.

meingraham, 2019-10-06:
@th-s-op 

We don't have to link the page initially. If you are willing, I would suggest give it your best first shot. Even if not complete and even if it has a few mistakes, having a start can get the ball rolling. @laurentdong is the author of this feature. Once you have a "draft", it will be easier for him to just make a few edits.

@Jason2866 can tell you, I've written lots of content for something I have not idea about... and with his help and that of others we have polished it to be useful for all users.

Mike

th-s-op, 2019-10-06:
@meingraham, who can say no if pushed in such a charming way :-)
I'll give it a try...

th-s-op, 2019-10-06:
Done. @laurentdong if you'd like to, please check if everything is correct.
https://github.com/arendst/Sonoff-Tasmota/wiki/Expressions-in-Rules

ascillato2, 2019-10-07:
Thanks everyone for working on this :+1:

closed 2019-10-07 by ascillato2