arendst/Tasmota #6587: Support for Link2Home (done)

g3gg0, 2019-10-07:
Hello,

some time ago i reverse engineered the protocol of the Link2Home devices (devices are available on https://www.l2h-rev.de/), especially the switchable wall plug and dimmable, colored garden lamp. (these devices can also be controlled via google etc)
I implemented the protocol on ESP8266 for my own devices, but didn't want to reinvent the wheel when i wanted to use the sonoff mini diy in my home.

So i tried tasmota and liked its "completeness". Thus I integrated my Link2Home emulator code into your project.
Tasmota can now be found with the Link2Home app and be controlled via google just like original devices.
(i.e.: open app, search for a new device, pair it and control it from everywhere)

Are you interested in merging that code into your project?
Or is it a bit too rude, emulating their hardware?

See: https://github.com/g3gg0/Sonoff-Tasmota

Regards,
g3gg0

ascillato, 2019-10-07:
Please, do a Pull Request with your driver

g3gg0, 2019-10-07:
well, before putting effort in cleaning up and seperating, i just wanted to see if there is some interest in it.

ascillato2, 2019-10-07:
Great!

If you need help please find us in the Tasmota Support Chat, or open the PR and we can check there.

Thanks for sharing :+1:

closed 2019-10-07 by ascillato2

mbecker, 2020-05-17:
> Hello,
>
> some time ago i reverse engineered the protocol of the Link2Home devices (devices are available on https://www.l2h-rev.de/), especially the switchable wall plug and dimmable, colored garden [...]

Hi @g3gg0 , any chane to see the protocol of Link2Home which you've reversed engineered? Unfortunately I do not find any additional links or further information in ypur profile or here in the code.

Thanks and best regards!

g3gg0, 2020-05-17:
its not documented in a readable way.
all i can give you is the source that emulates a device.
see https://github.com/g3gg0/Sonoff-Tasmota/blob/development/sonoff/l2h.h
and https://github.com/g3gg0/Sonoff-Tasmota/blob/development/sonoff/l2h.inc

mbecker, 2020-05-19:
Thanks @g3gg0 , great work! Unfortunately I was not able to identify how you communicate / send commands to the device via http (that was my assumption at the beginning). At the end I sniffed the netwrok communication and identified that the Link2Home devices are broadcasting their status via UDP and port 35932.

I see that you identified the commands for the Link2Home devices as well here: [https://github.com/g3gg0/Sonoff-Tasmota/blob/215ed4a7ce78aa38277f6049481e017064713ce9/sonoff/l2h.h#L33](https://github.com/g3gg0/Sonoff-Tasmota/blob/215ed4a7ce78aa38277f6049481e017064713ce9/sonoff/l2h.h#L33)

I guess the commands are bytes in hex value format. Is that correct?
Is the value 'L2H_CMD_SWITCH 0x01" to switch the lamp / power plug to 'on' / 'off'?

Based on my network packages I identified as well some commands for my shutter switch ("Wifi Jaluisenschalter"). I documented the commands here: [https://github.com/mbecker/mozilla-webthing-link2home#upd-data-paket](https://github.com/mbecker/mozilla-webthing-link2home#upd-data-paket)

g3gg0, 2020-05-19:
you are welcome.
see also https://github.com/g3gg0/Sonoff-Tasmota/blob/EmuL2H/sonoff/xdrv_29_l2h.ino
there might have been the missing pieces :)