arendst/Tasmota #6590: Display and Domoticz

mmaxx74, 2019-10-08:
Hi, can anyone help me?
i tried tasmota compiler, i need to load tasmota with lcd display and to have i domoticz settings. i tried different versions, but it doesn't work together.
I need the DS18b20 sensor to be displayed on the LCD and simultaneously sent the temperature to domoticz.
Can someone help me ? Thank you very much .

ascillato2, 2019-10-08:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-10-08 by ascillato2

mmaxx74, 2019-10-08:
Hello , where i find tasmota support chat ?
Can you get link . Thanks .

ascillato, 2019-10-08:
In the previous comment is the link

mmaxx74, 2019-10-08:
Ok thanks.