arendst/Tasmota #6593: IPv6 support and core_version

altelch, 2019-10-08:
Hi,

looked again into IPv6 support and stumbled over the following problem: 

What is the difference between the core versions? I can built against following platforms (platformio.ini):
* espressif8266@~2.2.2 aka core_2_5_2
* https://github.com/platformio/platform-espressif8266.git#feature/stage

but not against:
* https://github.com/Jason2866/platform-espressif8266.git#Tasmota aka core_2_5_0
* https://github.com/Jason2866/platform-espressif8266.git#Tasmota SDK22y aka core_pre

Looks like the v6-related defines are missing.
So I'd like to know what Jason2866 changed in his version what prevents IPv6 from working. Unfortunately there is no issues section in Jason2866 repository to ask my question there.

Thanks,

       Heiko.

altelch, 2019-10-08:
I know it's off topic but where should I ask!? I got some resources to try an IPv6 implementation again but I need to understand why I can't build against some core versions and if there is a chance that the tasmota "optimized" core can/will support IPv6 in future.

Greetings,
    
                Heiko.

andrethomas, 2019-10-08:
> if there is a chance that the tasmota "optimized" core can/will support IPv6 in future.

No - The pre-2.6 core is just for stability in favor of 2.5.2 in some cases and will be dropped once 2.6.0 or later is released.

ascillato2, 2019-10-08:
The Jason's version is just a freezed version from few commits behind to the actual STAGE version from Arduino Repo. You should try the actual arduino codebase (stage) that is the one that will become core 2.6.0 soon

altelch, 2019-10-08:
@ascillato2: Thanks for your response. Building against stage works. If this becomes the next version, then everything is ok for me but building against the currently set core_pre fails. Shouldn't core_pre be based on 2.6.0 then? Support for v6 startet in 2.5.x but none of the Json2866 versions support it, only the "original" versions work. I just want to avoid a situation where the code can't support relevant cores used during development (what seems to be Json2866 versions). For now I'll go with core_stage then.

Thanks,

              Heiko.

andrethomas, 2019-10-08:
@altelch It may just be the state in which the STAGE-2.6 repo was in at the time that it was cloned.

If I understand you correctly then you should be using the core_stage and not core_pre as this was pretty much made available exclusively for Sonoff-Tasmota and we don't use IPv6 so whether or not the headers and respective ip libraries are there is not relevant to any of the dev's on this specific project.

altelch, 2019-10-08:
Ok then. Thanks for your response!

closed 2019-10-08 by altelch