arendst/Tasmota #6594: Switchtopic/Butontopic '0' returns an empty field after reverting from a custom topic

effelle, 2019-10-08:
### BUG DESCRIPTION
Reverting `Switchtopic/Butontopic` to default (`0`) returns an empty field. 

### REQUESTED INFORMATION
- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Sonoff-Tasmota/issues)
- [X] Searched the problem in the [wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): D1 Mini / NodeMCU V3
- [X] Tasmota binary firmware version number used: 6.6.0.16
  - [X] Pre-compiled
  - [X] Self-compiled
    - [X] IDE / Compiler used: Gitpod
- [X] Flashing tools used: Esptool
- [X] Provide the output of command: ``Backlog Template; Module; GPIO``:

```
18:04:24 MQT: HA_DEV8/stat/RESULT = {"NAME":"Basic2","GPIO":[17,255,255,255,255,255,255,255,21,56,255,255,255],"FLAG":15,"BASE":18}
18:04:24 MQT: HA_DEV8/stat/RESULT = {"Module":{"18":"Generic"}}
18:04:24 MQT: HA_DEV8/stat/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO12":{"21":"Relay1"},"GPIO13":{"0":"None"},"GPIO14":{"9":"Switch1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}
```

- [ ] If using rules, provide the output of this command: ``Backlog Rule1; Rule2; Rule3``:

- [X] Provide the output of this command: ``Status 0``:

```
18:05:11 MQT: HA_DEV8/stat/STATUS = {"Status":{"Module":18,"FriendlyName":["HA_DEV8"],"Topic":"HA_DEV8","ButtonTopic":"","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
18:05:11 MQT: HA_DEV8/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/pre-2.6/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:21:28","StartupUTC":"2019-10-08T16:43:43","Sleep":50,"CfgHolder":4617,"BootCount":36,"SaveCount":92,"SaveAddress":"FA000"}}
18:05:11 MQT: HA_DEV8/stat/STATUS2 = {"StatusFWR":{"Version":"6.6.0.16(0062f7c-sonoff)","BuildDateTime":"2019-10-08T16:10:24","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)"}}
18:05:11 MQT: HA_DEV8/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":4,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["EffE-Domus",""],"TelePeriod":60,"Resolution":"558180C0","SetOption":["000A8009","280500000100060000005AFF000000000000","00000200"]}}
18:05:11 MQT: HA_DEV8/stat/STATUS4 = {"StatusMEM":{"ProgramSize":561,"Free":440,"Heap":26,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164068","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00000081"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
18:05:11 MQT: HA_DEV8/stat/STATUS5 = {"StatusNET":{"Hostname":"HA_DEV8-4216","IPAddress":"192.168.15.30","Gateway":"192.168.15.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.15.1","Mac":"CC:50:E3:47:10:78","Webserver":2,"WifiConfig":4}}
18:05:11 MQT: HA_DEV8/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.15.102","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_471078","MqttUser":"admin","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
18:05:11 MQT: HA_DEV8/stat/STATUS7 = {"StatusTIM":{"UTC":"Tue Oct 08 17:05:11 2019","Local":"Tue Oct 08 18:05:11 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"06:59","Sunset":"18:15"}}
18:05:11 MQT: HA_DEV8/stat/STATUS10 = {"StatusSNS":{"Time":"2019-10-08T18:05:11","Switch1":"ON"}}
18:05:11 MQT: HA_DEV8/stat/STATUS11 = {"StatusSTS":{"Time":"2019-10-08T18:05:11","Uptime":"0T00:21:28","UptimeSec":1288,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"EffE-Domus","BSSId":"94:EA:EA:97:2A:A7","Channel":1,"RSSI":50,"LinkCount":1,"Downtime":"0T00:00:04"}}}
```

- [X] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ ``weblog 4`` _for more debug information)_

```
18:08:31 CMD: switchtopic 
18:08:31 SRC: WebConsole from 192.168.15.199
18:08:31 CMD: Group 0, Index 1, Command "SWITCHTOPIC", Data ""
18:08:31 MQT: HA_DEV8/stat/RESULT = {"SwitchTopic":"0"}
18:08:35 CMD: switchtopic test
18:08:35 SRC: WebConsole from 192.168.15.199
18:08:35 CMD: Group 0, Index 1, Command "SWITCHTOPIC", Data "test"
18:08:35 MQT: HA_DEV8/stat/RESULT = {"SwitchTopic":"test"}
18:08:37 CMD: switchtopic 0
18:08:37 SRC: WebConsole from 192.168.15.199
18:08:37 CMD: Group 0, Index 1, Command "SWITCHTOPIC", Data "0"
18:08:37 MQT: HA_DEV8/stat/RESULT = {"SwitchTopic":""}
18:10:03 CMD: buttontopic 
18:10:03 SRC: WebConsole from 192.168.15.199
18:10:03 CMD: Group 0, Index 1, Command "BUTTONTOPIC", Data ""
18:10:03 MQT: HA_DEV8/stat/RESULT = {"ButtonTopic":"0"}
18:10:06 CMD: buttontopic test
18:10:06 SRC: WebConsole from 192.168.15.199
18:10:06 CMD: Group 0, Index 1, Command "BUTTONTOPIC", Data "test"
18:10:06 MQT: HA_DEV8/stat/RESULT = {"ButtonTopic":"test"}
18:10:08 CMD: buttontopic 0
18:10:08 SRC: WebConsole from 192.168.15.199
18:10:08 CMD: Group 0, Index 1, Command "BUTTONTOPIC", Data "0"
18:10:08 MQT: HA_DEV8/stat/RESULT = {"ButtonTopic":""}
```

### TO REPRODUCE
Change switchtopic / buttontopic to anything else than default and then revert.

### EXPECTED BEHAVIOUR
When changing  Switchtopic / Buttontopic to `0` Tasmota should revert them to the original state.

### ADDITIONAL CONTEXT
Seems when a *topic returns an empty string Home Assistant Discovery will not create entities. Under investigation due to other issues present on `xdrv_12_home_assistant.ino` relative to button/switches.

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2019-10-08:
As designed. A 0 is also an empty string for both buttontopic and switchtopic.

effelle, 2019-10-08:
I have an `if (strlen(sw_topic) != 0)` not triggered with an empty field.

arendst, 2019-10-08:
Then find another solution. There are plenthy within the tasmota code. Also learn the difference between NULL and nullptr.

effelle, 2019-10-08:
Is the original code since 6.3... Bu yes, I need another solution.

closed 2019-10-08 by effelle