arendst/Tasmota #6600: Relais option for RX, switch or button for TX?

jmar83, 2019-10-09:
Hello

So i saw, there are all the same options for all pins. (GPIO, RX, TX)

**Question:** How can i connect a relais on a RX (=receive, input) and how can i connect a switch or button on a TX (=transmit, output) ? That makes no sense in my humble opinion...?

I don't only need a simple on/off or true/false on the RX/TX pins, i need to a send (on/off/brightness) and receive (on/off/brightness states) some hex strings for a dimmer over the RX/TX pins. (NOT GPIO, the dimmer needs a RS232 signal)

Is that possible to send/receive that with tasmota? Or just simple on/off-things?? (E.g. a on/off-bridge (contact/no contact) on RX, and 0 or 5 volts on TX)

Thank you for your feedbacks! :-)

Best regards,
Jan

meingraham, 2019-10-09:
Our ability to provide assistance is greatly hampered since you did not take a few minutes to complete the  [troubleshooting template](https://github.com/arendst/Sonoff-Tasmota/issues/new/choose) with the requested information. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help you.

Please close this issue and address this to the [Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are best used for Tasmota **_software feature requests and bug reporting_**. Troubleshooting and setup assistance is more effective using an interactive forum.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-10-09 by jmar83

jmar83, 2019-10-21:
solved!! :-)