arendst/Tasmota #6602: MQTT daemon does not anwser to telnet 192.168.2.245 1883 ?

jmar83, 2019-10-09:
MQTT daemon does not anwser to telnet 192.168.2.245 1883 ? And what i do need to define for "host", and why i have to define that? Is it a "whitelist" for IP they can access the tasmota mqtt server on port 1883 ?

However, `*` does not work... still no anwser from telnet...?

Thank you for your feedbacks.

jmar83, 2019-10-09:
So i saw, i have to define the name from the sonoff itself on "WLAN Settings" - so the host from my above question can't be that... so it must be like a whitelist, but what do i need for accepting connection from ALL IPs / hosts?

jmar83, 2019-10-09:
Off-topic-question: what means `%s-%04d` on WLAN host name option field? Looks like a bug for me...(?)

jmar83, 2019-10-09:
"However, * does not work... still no anwser from telnet...?"

Same problem with `%` ...

meingraham, 2019-10-09:
Our ability to provide assistance is greatly hampered since you did not take a few minutes to complete the  [troubleshooting template](https://github.com/arendst/Sonoff-Tasmota/issues/new/choose) with the requested information. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help you.

Please close this issue and address this to the [Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are best used for Tasmota **_software feature requests and bug reporting_**. Troubleshooting and setup assistance is more effective using an interactive forum.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-10-09 by jmar83

jmar83, 2019-10-21:
solved!! :-)