arendst/Tasmota #6605: Tuya MCU add support for 2 bytes brightness

Nonsense1112, 2019-10-10:
Hello,
I'm using tasmota 6.6.016 with Moes Dimmer DS01C. It seems that this is a new model and the brightness is using 2 bytes instead of 1 byte as other models. 
Brightness range is between 10-1000.

The minimum brightness command is:
55 AA 00 06 00 08 02 02 00 04 00 00 00 0A 1F

and for maximum brightness command is:
55 AA 00 06 00 08 02 02 00 04 00 00 03 E8 00

The question is, can this be implemented in existing Tuya MCU module?

ascillato, 2019-10-10:
@Nonsense1112

Your request has been added by @shantur on PR #6606

Please, can you test it? Thanks

Nonsense1112, 2019-10-10:
Hi,
It works now.
The only problem is with the range below touch slider minimum.
the touch minimum is 55aa0307000802020004000000b4cd

If I want to dim the lights below the touch minimum using the switch, I have to double press the button, but the mcu send a package of 55aa0307000802020004000000b4cd and updates tasmota brightness.
But I don't think there is something that it can be done, because tasmota can't know the difference between the slider minimum and button double press, as the mcu send the same response for both.

Thank you!

PR #6606 (2019-10-10): Add DimmerRange command and support longer dimmer ranges. Fixes #6605
More and more serial dimmers are coming up with different dimming ranges, implemented DimmerRange command to make this setting generic and bring support for 2 byte dimming ranges which isn't supported by SetOptions.

## Description:

**Related issue (if applicable):** fixes #6605 

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  - [ ] The code change is tested and works on core 2.3.0, 2.4.2, 2.5.2, and pre-2 [... cut]
commit 01f6df91 (2019-10-10): Add DimmerRange command and support longer dimmer ranges. Fixes #6605

More and more serial dimmers are coming up with different dimming ranges, implemented DimmerRange command to make this setting generic and bring support for 2 byte dimming ranges which isn't supported by SetOptions.
commit 3cd6abe0 (2019-10-10): Merge pull request #6606 from shantur/tuya-2-byte-dimmer

Add DimmerRange command and support longer dimmer ranges. Fixes #6605
closed 2019-10-10 by arendst