arendst/Tasmota #6608: WiFi/PING issue when using H801/PWM

delinend, 2019-10-10:
I see some WiFi/ping problem with the new 6.6.0 version, when using the H801 modul.

Have tryed Core 2.5.2, Core 2.4.2 and 2.3.0, but all with same issue.

Problem:
When all LED's are OFF I see ping replyes from 2 to 8ms. (normal) over WiFi.
When one color or all LED's are dimmed or full, I see ping replyes around 200-500ms, and offen the PING's is timing out, to the H801 modul.

This result in, that I sometimes can't controle the H801 modul, when the H801 has LED's ON.

Maybe there is an internal PWM WiFi problem?

Have updated 8 pcs. H801 modul, and all now have the same issue, with release 6.6.0

I have an old 6.0.0c/core 2.3.0 release, and there I do not see the problem.

And I only use the precompiled sonoff.bin release, from here: http://thehackbox.org/tasmota/release/

Best regards.

Jason2866, 2019-10-10:
Try latest development version http://thehackbox.org/tasmota/pre-2.6/sonoff.bin

delinend, 2019-10-10:
Have just tryed pre-2.6 / 6.6.17/STAGE, and it have the same behavior as 6.6.0/all core builds.

Large PING replys, when LED's are ON. Now 90 to 450 ms. and sometimes no replys.

When LED's are OFF, then normal PING replys 2 to 8 ms. over WiFi to the H801 modul.

Btw. I have also testet with a large/another power supply, and only 6 LED's, to prevent maybe a power problem. But same problem. Also when LED's are weak dimming. When PWM is ON, then there is a PING/WiFi problem.

Jason2866, 2019-10-10:
Have you tried to use not all PWM channels? 
Use template function for and start testing with just one configured PWM to a GPIO

jziolkowski, 2019-10-10:
I have H801 with SetOption68 enabled, using 3 channels currently, at various PWM setting.

First test: 6.6.0.12 on core 2.3.0

Started ping and started toggling the outputs. When inactive, ping is ~1-2ms, during toggling one or two packets jump to ~5-15ms and then return to previous low value.

Second test: 6.6.0.17 on core pre2.6 gave similar results.

So here I can't reproduce your problem, sorry.

delinend, 2019-10-10:
Hmm. Now it's getting weird..

I always use channel 8 on my AP. If I switch to channel 1 or channel 12, the PING problem dissapears ?!?

Why is channel 8 working perfect on old version 6.0.0.c but not from 6.6.0+ ?

I'll test all channels tomorrow

meingraham, 2019-10-10:
Have you changed Cores?

ascillato, 2019-10-11:
@delinend Sorry, but seems to be a Wifi issue.

Could you be so kind on completing the [troubleshooting template](https://github.com/arendst/Sonoff-Tasmota/issues/new/choose) in order to have more information so as to properly help you?

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

Th0maz, 2019-10-11:
Could be related to my #5506 issue with AiLight. Sounds kind of similar.

ascillato, 2019-10-11:
@Th0maz

Seems not related. This issue is related with wifi disconnections on wifi channel 8.

delinend, 2019-10-11:
### ISSUE DESCRIPTION - TROUBLESHOOTING
I see some WiFi/ping problem with the new 6.6.0 version, when using the H801 modul.

Have tryed Core 2.5.2, Core 2.4.2 and 2.3.0, and now allso pre-2.6/6.6.0.17 but all with same issue.

Problem:
When all LED's are OFF I see ping replyes from 2 to 8ms. (normal) over WiFi.
When one color or all LED's are dimmed or full, I see ping replyes around 2-500ms, and offen the PING's is timing out, to the H801 modul.

This result in, that I sometimes can't controle the H801 modul, when the H801 has LED's ON.

Maybe there is an internal PWM WiFi problem?

Have updated 8 pcs. H801 modul, and all now have the same issue, with release 6.6.0 and pre-2.6/6.6.0.17

I have an old 6.0.0c/core 2.3.0 release, and there I do not see the problem.

Same problem on pre-2.6/6.6.0.17

Now I see the issue is related , only on some WiFi channels.

Here a matrix when one or more LED's are on or low dimmed:
**Channel , 6.0.0.c, 6.6.0, 6.6.0.17**
1, OK, OK, OK
2, OK, OK, OK
3, OK, OK, OK
4, OK, OK, OK
5, OK, **FAIL**, **FAIL**
6, OK, **FAIL**, **FAIL**
7, OK, **FAIL**, **FAIL**
8, OK, **FAIL**, **FAIL**
9, OK, OK, OK
10, OK, OK, OK
11, OK, OK, OK
12, OK, OK, OK

  
Here a matrix when ALL LED's are **OFF**:
**Channel , 6.0.0.c, 6.6.0, 6.6.0.17**
1, OK, OK, OK
2, OK, OK, OK
3, OK, OK, OK
4, OK, OK, OK
5, OK, OK, OK
6, OK, OK, OK
7, OK, OK, OK
8, OK, OK, OK
9, OK, OK, OK
10, OK, OK, OK
11, OK, OK, OK
12, OK, OK, OK

Here my  UniFi controller "WiFi experience" matrix, when one or more LED's are on or low dimmed:
**Channel , 6.0.0.c, 6.6.0, 6.6.0.17**
1, 90-99%, 90-99%, 90-99%
2, 90-99%, 90-99%, 90-99%
3, 90-99%, 90-99%, 90-99%
4, 90-99%, 90-99%, 90-99%
5, 90-99%, **0-5%**, **0-5%**
6, 90-99%, **0-5%**, **0-5%**
7, 90-99%, **0-5%**, **0-5%**
8, 90-99%, **0-5%**, **0-5%**
9, 90-99%, 90-99%, 90-99%
10, 90-99%, 90-99%, 90-99%
11, 90-99%, 90-99%, 90-99%
12, 90-99%, 90-99%, 90-99%

Here my  UniFi controller "WiFi experience" matrix, when ALL LED's are **OFF**:
**Channel , 6.0.0.c, 6.6.0, 6.6.0.17**
1, 90-99%, 90-99%, 90-99%
[... 115 lines cut ...]
  Reply from 192.168.1.30: bytes=32 time=2ms TTL=255
  Reply from 192.168.1.30: bytes=32 time=4ms TTL=255
  Reply from 192.168.1.30: bytes=32 time=3ms TTL=255
  Reply from 192.168.1.30: bytes=32 time=2ms TTL=255
  Reply from 192.168.1.30: bytes=32 time=7ms TTL=255
  Reply from 192.168.1.30: bytes=32 time=3ms TTL=255
  Reply from 192.168.1.30: bytes=32 time=4ms TTL=255
  Reply from 192.168.1.30: bytes=32 time=4ms TTL=255
  Reply from 192.168.1.30: bytes=32 time=5ms TTL=255
  Reply from 192.168.1.30: bytes=32 time=5ms TTL=255
  Reply from 192.168.1.30: bytes=32 time=2ms TTL=255
  Reply from 192.168.1.30: bytes=32 time=4ms TTL=255
  Reply from 192.168.1.30: bytes=32 time=6ms TTL=255
  Reply from 192.168.1.30: bytes=32 time=5ms TTL=255
  Reply from 192.168.1.30: bytes=32 time=2ms TTL=255
  Reply from 192.168.1.30: bytes=32 time=4ms TTL=255
  Reply from 192.168.1.30: bytes=32 time=4ms TTL=255
  Reply from 192.168.1.30: bytes=32 time=5ms TTL=255
  Reply from 192.168.1.30: bytes=32 time=3ms TTL=255
  Reply from 192.168.1.30: bytes=32 time=3ms TTL=255
  Reply from 192.168.1.30: bytes=32 time=3ms TTL=255
  Reply from 192.168.1.30: bytes=32 time=3ms TTL=255
  Reply from 192.168.1.30: bytes=32 time=5ms TTL=255
   
  ```

Jason2866, 2019-10-11:
I think the PWM will disturb wifi. 
The timing has changed a litte in Tasmota because code changed, so this effect was not there before.
Not a Tasmota issues! It is a Hf problem of the hardware. You can try to change the PWM frequency to change the generated disturbing HF waves.

delinend, 2019-10-11:
But why then only problems on WiFi channel 5, 6, 7 and 8 ?
And no problems on WiFi channel 1, 2, 3, 4, 9, 10, 11 and 12

ascillato2, 2019-10-11:
>But why then only problems on WiFi channel 5, 6, 7 and 8 ?

Different wifi channel = different frequency

ascillato2, 2019-10-11:
Weird problem. I couldn't reproduce it. Anyway, you have found a workaround for it.

Jason2866, 2019-10-11:
> But why then only problems on WiFi channel 5, 6, 7 and 8 ?
> And no problems on WiFi channel 1, 2, 3, 4, 9, 10, 11 and 12
> 
> Btw. The problem is regardless with low or high brightness. And also with 8 LED's or 80 LED's. Same problem. Only when all LED's are OFF, and the red LED on the H801 mainboard goes OFF, the problem is gone, when using WiFi channels 5, 6, 7 and 8.

All your findings fits to a HF issue through PWM. There is a Power MosFET as driver for the LEDs
FET´s are switching fast with very steep flanks. Steep flanks generates HF noise.
As i already said you cant try to change the PWM frequency to get it out of wifi  frequency spectrum

delinend, 2019-10-11:
To change the PWM frequency.... do I have do compile ?

I'm 100% a technician guy, and only a 50% programmer.

I also found this, maybe same problem. [https://github.com/xoseperez/espurna/issues/914](url)

Jason2866, 2019-10-11:
It is a command. `PwmFrequency` See comamands section in wiki
https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#management

delinend, 2019-10-11:
Thanks.

Have now tryed PwmFrequency 100/500/1000/4000, but with same result.

If I presse the "Toggle" botton OFF then it works. If I press ON again, then it fails again.

And another stange discovery: If I remove the LED's (connector) from the H801 modul (no driver load), then I still see the problem i ON stage.

Jason2866, 2019-10-11:
Driver load is irrelevant since the MosFet is switching on/off if there is a load connected or not.
Sorry if this didnt solve, was a try. HF issues are most of the time not easy to solve.
From Tasmota side nothing more can be done.  It is a bad hardware design!

ascillato2, 2019-10-11:
Closing this issue as it is hardware related due to it can be replicated on few devices and not in others. Sorry.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-10-11 by ascillato2

jziolkowski, 2019-10-12:
There is HF issue for sure, because I've noticed that when my headphones cable is near H801, when PWM channels are ON in various configurations I can hear a hum, changing according to PWM output.

delinend, 2019-10-12:
I have tryed to measure around the ESP8266 chip with my oscilloscope, decopled with small, mid and large caps around supply pins, but all looks very "clean".

Have tryed another location, that also have a H801 setup, with another AP device... changed to v. 6.6.0 and channel 8 on the AP. When we have a distance of around 8 meters between AP and the H801, we also see PING issues there.

Strange that the problem only is in the middel WiFi channel range (ch. 5, 6, 7, 8).

Back to v. 6.0.0.c and all channels works fine again.

delinend, 2019-10-12:
I have now tryed to switch to use templates.

If I remove all 5 pcs. PWM GPIO channels (R,G,B,CW,WW), then the proglem is gone.

If I just add one PWM/GPIO ex. W1 that normal is off... then the problem starts again.

16:27:42 CMD: Backlog Template; Module; GPIO:
16:27:42 MQT: stat/H801_4212B9/RESULT = {"NAME":"Generic","GPIO":[255,52,255,255,0,57,0,0,0,0,0,0,0],"FLAG":0,"BASE":20}
16:27:43 MQT: stat/H801_4212B9/RESULT = {"Module":"0 (Generic)"}
16:27:43 MQT: stat/H801_4212B9/RESULT = {"Command":"Unknown"}

andrethomas, 2019-10-12:
What does the `sleep` command return when its having these ping issues?

delinend, 2019-10-12:
17:17:46 CMD: sleep
17:17:46 MQT: stat/H801_4212B9/RESULT = {"Sleep":"0 (50)"}

andrethomas, 2019-10-12:
@arendst Looks like the H801 suffers some of the same issues we saw on the sonoff basic R1

I did PR https://github.com/arendst/Sonoff-Tasmota/pull/5264 to make it possible to disable the `sleep = 0;` which was in my case using ws2812 led's without the need for animation or schemes (it is just turning them all on white or off for a light in a dark passage with a PIR sensor on it)

My best guess for the different behaviour on different wifi channels is that the noise floor is likely higher on those channels and as a result the esp8266 is increasing its gain on the wifi side to try and maintain a better quality connection but then the heat is getting the better of the esp8266... hows that for self inflicted injury... just a theory.

jziolkowski, 2019-10-12:
I have a few more H801 at home (although bought some time ago; OTOH I don't know if they have different revisions, might be worth looking into), I might have some time tomorrow to test because I have some new led strips to install.

delinend, 2019-10-13:
First... I'm not a programmer shark, but I managed to compile my one BIN file.

Using Arduino IDE 1.8.1 with ESP8266/2.3.0 core and the Master Tasmota/6.6.0 zip.

Then using NodeMCU PyFlasher 4.0 to flash the H801 module and always erase all data before flash.

Result: Same issue as with precompiled BIN on Tasmota. All PING replys over 40ms when LED's ON.

Next I tryed to remove the // from **#define PWM_LIGHTSCHEME0_IGNORE_SLEEP** line in the sonoff.h file, as @andrethomas has mentioned.

Result: It now works on all WiFi channels !! Normal PING replys from 2ms to 5ms when LED's ON.

I don't know what this does, when I enable the line.

Tryed flashing back and forward between those two BIN's I maded 10 times now (testing), and I'm 100% sure, that the line is resolving the issue.

andrethomas, 2019-10-13:
@delinend Thx, you basically confirmed my theory. Happy it works for you.

delinend, 2019-10-13:
This can have a huge issue range, of people reporting H801 not responding or not switching on/off. Reporting of WiFi problem and some other don't see problems, regaring there AP setup's.

Maybe someone can dig deeper into this. I will gladly test.

The enabling of **#define PWM_LIGHTSCHEME0_IGNORE_SLEEP** in the sonoff.h file, is working for me.

Tnx. for your help @andrethomas

andrethomas, 2019-10-13:
Enabling the directive is just forcing the ESP chip to stay on a target loop of 50ms whilst using scheme 0 (the default) so that's the reason it is behaving differently with that change.

Your compiled firmware is using `sleep 50` at the moment - I think the question is how low can you go with the sleep setting before experiencing unexpected behavior.

Try lowering the sleep setting using the command `sleep 40` to set it to 40 and then down by 5 each time until you get to a sleep setting of 1 (if the chip will still function reliably on that)

delinend, 2019-10-13:
Have now tryed sleep 50 to 0.

0:57:27 CMD: sleep
10:57:27 MQT: stat/H801_4212B9/RESULT = {"Sleep":"0 (50)"}
10:57:30 CMD: sleep 40
10:57:30 MQT: stat/H801_4212B9/RESULT = {"Sleep":"40 (40)"}
10:57:33 CMD: sleep
10:57:33 MQT: stat/H801_4212B9/RESULT = {"Sleep":"0 (40)"}
10:57:51 CMD: sleep 35
10:57:51 MQT: stat/H801_4212B9/RESULT = {"Sleep":"35 (35)"}
10:57:55 CMD: sleep
10:57:55 MQT: stat/H801_4212B9/RESULT = {"Sleep":"0 (35)"}
10:58:15 CMD: sleep 30
10:58:15 MQT: stat/H801_4212B9/RESULT = {"Sleep":"30 (30)"}
10:58:28 CMD: sleep 25
10:58:28 MQT: stat/H801_4212B9/RESULT = {"Sleep":"25 (25)"}
10:58:47 CMD: sleep 20
10:58:47 MQT: stat/H801_4212B9/RESULT = {"Sleep":"20 (20)"}
10:59:02 CMD: sleep 15
10:59:02 MQT: stat/H801_4212B9/RESULT = {"Sleep":"15 (15)"}
10:59:17 CMD: sleep 10
10:59:17 MQT: stat/H801_4212B9/RESULT = {"Sleep":"10 (10)"}
10:59:35 CMD: sleep 5
10:59:35 MQT: stat/H801_4212B9/RESULT = {"Sleep":"5 (5)"}
10:59:47 CMD: sleep 0
10:59:47 MQT: stat/H801_4212B9/RESULT = {"Sleep":"0 (0)"}

But still large PING replys and Time out's:
Reply from 192.168.1.39: bytes=32 time=4ms TTL=128
Reply from 192.168.1.39: bytes=32 time=34ms TTL=128
Reply from 192.168.1.39: bytes=32 time=77ms TTL=128
Request timed out.
Reply from 192.168.1.39: bytes=32 time=67ms TTL=128
Request timed out.
Reply from 192.168.1.39: bytes=32 time=22ms TTL=128
Reply from 192.168.1.39: bytes=32 time=62ms TTL=128
Request timed out.
Reply from 192.168.1.39: bytes=32 time=559ms TTL=128
Reply from 192.168.1.39: bytes=32 time=181ms TTL=128
Reply from 192.168.1.39: bytes=32 time=59ms TTL=128
Reply from 192.168.1.39: bytes=32 time=41ms TTL=128
Reply from 192.168.1.39: bytes=32 time=7ms TTL=128
Request timed out.
Reply from 192.168.1.39: bytes=32 time=185ms TTL=128
Reply from 192.168.1.39: bytes=32 time=8ms TTL=128
Reply from 192.168.1.39: bytes=32 time=23ms TTL=128
Reply from 192.168.1.39: bytes=32 time=14ms TTL=128
Reply from 192.168.1.39: bytes=32 time=3ms TTL=128
Reply from 192.168.1.39: bytes=32 time=29ms TTL=128
Reply from 192.168.1.39: bytes=32 time=34ms TTL=128
Reply from 192.168.1.39: bytes=32 time=3ms TTL=128
Reply from 192.168.1.39: bytes=32 time=34ms TTL=128

andrethomas, 2019-10-13:
No need to reboot after changing sleep setting - just a case of giving it a few seconds to settle on the current setting.

Also please provide the LoadAvg value reported by command `status 11` on sleep 50

delinend, 2019-10-13:
**Here status 11 from Precompiled 6.6.0. and sleep 50**

`11:20:50 CMD: status 11
11:20:50 MQT: stat/H801_4212B9/STATUS11 = {"StatusSTS":{"Time":"2019-10-13T11:20:50","Uptime":"0T00:23:59","Heap":13,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"POWER":"ON","Dimmer":100,"Color":"1A1A1AFFFF","HSBColor":"0,0,10","Channel":[10,10,10,100,100],"CT":327,"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"dims","BSSId":"B4:FB:E4:44:09:D2","Channel":8,"RSSI":90,"LinkCount":1,"Downtime":"0T00:00:04"}}}
11:21:33 CMD: status 11
11:21:33 MQT: stat/H801_4212B9/STATUS11 = {"StatusSTS":{"Time":"2019-10-13T11:21:33","Uptime":"0T00:24:42","Heap":13,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"POWER":"ON","Dimmer":100,"Color":"1A1A1AFFFF","HSBColor":"0,0,10","Channel":[10,10,10,100,100],"CT":327,"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"dims","BSSId":"B4:FB:E4:44:09:D2","Channel":8,"RSSI":88,"LinkCount":1,"Downtime":"0T00:00:04"}}}
11:21:38 CMD: status 11
11:21:38 MQT: stat/H801_4212B9/STATUS11 = {"StatusSTS":{"Time":"2019-10-13T11:21:38","Uptime":"0T00:24:47","Heap":13,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"POWER":"ON","Dimmer":100,"Color":"1A1A1AFFFF","HSBColor":"0,0,10","Channel":[10,10,10,100,100],"CT":327,"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"dims","BSSId":"B4:FB:E4:44:09:D2","Channel":8,"RSSI":90,"LinkCount":1,"Downtime":"0T00:00:04"}}}
11:21:43 CMD: status 11
11:21:43 MQT: stat/H801_4212B9/STATUS11 = {"StatusSTS":{"Time":"2019-10-13T11:21:43","Uptime":"0T00:24:52","Heap":13,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"POWER":"ON","Dimmer":100,"Color":"1A1A1AFFFF","HSBColor":"0,0,10","Channel":[10,10,10,100,100],"CT":327,"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"dims","BSSId":"B4:FB:E4:44:09:D2","Channel":8,"RSSI":88,"LinkCount":1,"Downtime":"0T00:00:04"}}}
11:21:46 CMD: status 11
11:21:46 MQT: stat/H801_4212B9/STATUS11 = {"StatusSTS":{"Time":"2019-10-13T11:21:46","Uptime":"0T00:24:55","Heap":13,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"POWER":"ON","Dimmer":100,"Color":"1A1A1AFFFF","HSBColor":"0,0,10","Channel":[10,10,10,100,100],"CT":327,"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"dims","BSSId":"B4:FB:E4:44:09:D2","Channel":8,"RSSI":86,"LinkCount":1,"Downtime":"0T00:00:04"}}}
11:21:49 CMD: status 11
11:21:50 MQT: stat/H801_4212B9/STATUS11 = {"StatusSTS":{"Time":"2019-10-13T11:21:49","Uptime":"0T00:24:58","Heap":13,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"POWER":"ON","Dimmer":100,"Color":"1A1A1AFFFF","HSBColor":"0,0,10","Channel":[10,10,10,100,100],"CT":327,"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"dims","BSSId":"B4:FB:E4:44:09:D2","Channel":8,"RSSI":88,"LinkCount":1,"Downtime":"0T00:00:04"}}}
11:21:52 CMD: status 11
11:21:53 MQT: stat/H801_4212B9/STATUS11 = {"StatusSTS":{"Time":"2019-10-13T11:21:52","Uptime":"0T00:25:01","Heap":13,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"POWER":"ON","Dimmer":100,"Color":"1A1A1AFFFF","HSBColor":"0,0,10","Channel":[10,10,10,100,100],"CT":327,"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"dims","BSSId":"B4:FB:E4:44:09:D2","Channel":8,"RSSI":88,"LinkCount":1,"Downtime":"0T00:00:04"}}}
11:21:55 CMD: status 11
11:21:55 MQT: stat/H801_4212B9/STATUS11 = {"StatusSTS":{"Time":"2019-10-13T11:21:55","Uptime":"0T00:25:04","Heap":13,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"POWER":"ON","Dimmer":100,"Color":"1A1A1AFFFF","HSBColor":"0,0,10","Channel":[10,10,10,100,100],"CT":327,"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"dims","BSSId":"B4:FB:E4:44:09:D2","Channel":8,"RSSI":88,"LinkCount":1,"Downtime":"0T00:00:04"}}}
11:22:00 CMD: status 11
11:22:00 MQT: stat/H801_4212B9/STATUS11 = {"StatusSTS":{"Time":"2019-10-13T11:22:00","Uptime":"0T00:25:09","Heap":13,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"POWER":"ON","Dimmer":100,"Color":"1A1A1AFFFF","HSBColor":"0,0,10","Channel":[10,10,10,100,100],"CT":327,"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"dims","BSSId":"B4:FB:E4:44:09:D2","Channel":8,"RSSI":88,"LinkCount":1,"Downtime":"0T00:00:04"}}}
11:22:03 CMD: status 11
11:22:03 MQT: stat/H801_4212B9/STATUS11 = {"StatusSTS":{"Time":"2019-10-13T11:22:03","Uptime":"0T00:25:12","Heap":13,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"POWER":"ON","Dimmer":100,"Color":"1A1A1AFFFF","HSBColor":"0,0,10","Channel":[10,10,10,100,100],"CT":327,"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"dims","BSSId":"B4:FB:E4:44:09:D2","Channel":8,"RSSI":88,"LinkCount":1,"Downtime":"0T00:00:04"}}}
11:22:06 CMD: status 11
11:22:06 MQT: stat/H801_4212B9/STATUS11 = {"StatusSTS":{"Time":"2019-10-13T11:22:06","Uptime":"0T00:25:15","Heap":13,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"POWER":"ON","Dimmer":100,"Color":"1A1A1AFFFF","HSBColor":"0,0,10","Channel":[10,10,10,100,100],"CT":327,"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"dims","BSSId":"B4:FB:E4:44:09:D2","Channel":8,"RSSI":88,"LinkCount":1,"Downtime":"0T00:00:04"}}}
11:22:11 MQT: tele/H801_4212B9/STATE = {"Time":"2019-10-13T11:22:11","Uptime":"0T00:25:20","Heap":14,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"POWER":"ON","Dimmer":100,"Color":"1A1A1AFFFF","HSBColor":"0,0,10","Channel":[10,10,10,100,100],"CT":327,"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"dims","BSSId":"B4:FB:E4:44:09:D2"," [... cut ...] nkCount":1,"Downtime":"0T00:00:04"}}}
11:30:24 CMD: status 11
11:30:24 MQT: stat/H801_4212B9/STATUS11 = {"StatusSTS":{"Time":"2019-10-13T11:30:24","Uptime":"0T00:01:13","Heap":14,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Dimmer":100,"Color":"1A1A1AFFFF","HSBColor":"0,0,10","Channel":[10,10,10,100,100],"CT":327,"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"dims","BSSId":"B4:FB:E4:44:09:D2","Channel":8,"RSSI":86,"LinkCount":1,"Downtime":"0T00:00:04"}}}`

delinend, 2019-10-13:
**Her again status 11, from Precompiled 6.6.0. The LoadAvg is increases after reboot !?!**

`00:00:05 MQT: Connected
00:00:05 MQT: tele/sonoff/LWT = Online (retained)
00:00:05 MQT: cmnd/sonoff/POWER = 
00:00:05 MQT: tele/sonoff/INFO1 = {"Module":"H801","Version":"6.6.0(sonoff)","FallbackTopic":"cmnd/H801_4212B9_fb/","GroupTopic":"sonoffs"}
00:00:05 MQT: tele/sonoff/INFO2 = {"WebServerMode":"Admin","Hostname":"H801_4212B9","IPAddress":"192.168.1.39"}
00:00:06 MQT: tele/sonoff/INFO3 = {"RestartReason":"External System"}
00:00:06 MQT: stat/sonoff/RESULT = {"POWER":"ON"}
00:00:06 MQT: stat/sonoff/POWER = ON
11:44:07 MQT: tele/sonoff/STATE = {"Time":"2019-10-13T11:44:07","Uptime":"0T00:00:14","Heap":14,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":107,"POWER":"ON","Dimmer":100,"Color":"1A1A1AFFFF","HSBColor":"0,0,10","Channel":[10,10,10,100,100],"CT":327,"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"dims","BSSId":"B4:FB:E4:44:09:D2","Channel":8,"RSSI":82,"LinkCount":1,"Downtime":"0T00:00:04"}}
11:44:45 CMD: status 11
11:44:45 MQT: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-10-13T11:44:45","Uptime":"0T00:00:52","Heap":13,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":352,"POWER":"ON","Dimmer":100,"Color":"1A1A1AFFFF","HSBColor":"0,0,10","Channel":[10,10,10,100,100],"CT":327,"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"dims","BSSId":"B4:FB:E4:44:09:D2","Channel":8,"RSSI":82,"LinkCount":1,"Downtime":"0T00:00:04"}}}
11:44:52 CMD: status 11
11:44:52 MQT: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-10-13T11:44:52","Uptime":"0T00:00:59","Heap":13,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":453,"POWER":"ON","Dimmer":100,"Color":"1A1A1AFFFF","HSBColor":"0,0,10","Channel":[10,10,10,100,100],"CT":327,"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"dims","BSSId":"B4:FB:E4:44:09:D2","Channel":8,"RSSI":82,"LinkCount":1,"Downtime":"0T00:00:04"}}}
11:44:55 CMD: status 11
11:44:55 MQT: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-10-13T11:44:55","Uptime":"0T00:01:02","Heap":13,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":493,"POWER":"ON","Dimmer":100,"Color":"1A1A1AFFFF","HSBColor":"0,0,10","Channel":[10,10,10,100,100],"CT":327,"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"dims","BSSId":"B4:FB:E4:44:09:D2","Channel":8,"RSSI":82,"LinkCount":1,"Downtime":"0T00:00:04"}}}
11:44:57 CMD: status 11
11:44:58 MQT: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-10-13T11:44:57","Uptime":"0T00:01:04","Heap":13,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":533,"POWER":"ON","Dimmer":100,"Color":"1A1A1AFFFF","HSBColor":"0,0,10","Channel":[10,10,10,100,100],"CT":327,"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"dims","BSSId":"B4:FB:E4:44:09:D2","Channel":8,"RSSI":82,"LinkCount":1,"Downtime":"0T00:00:04"}}}
11:45:00 CMD: status 11
11:45:00 MQT: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-10-13T11:45:00","Uptime":"0T00:01:07","Heap":13,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":575,"POWER":"ON","Dimmer":100,"Color":"1A1A1AFFFF","HSBColor":"0,0,10","Channel":[10,10,10,100,100],"CT":327,"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"dims","BSSId":"B4:FB:E4:44:09:D2","Channel":8,"RSSI":82,"LinkCount":1,"Downtime":"0T00:00:04"}}}
11:45:04 CMD: status 11
11:45:04 MQT: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-10-13T11:45:04","Uptime":"0T00:01:11","Heap":13,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":614,"POWER":"ON","Dimmer":100,"Color":"1A1A1AFFFF","HSBColor":"0,0,10","Channel":[10,10,10,100,100],"CT":327,"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"dims","BSSId":"B4:FB:E4:44:09:D2","Channel":8,"RSSI":82,"LinkCount":1,"Downtime":"0T00:00:04"}}}
11:45:06 CMD: status 11
11:45:06 MQT: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-10-13T11:45:06","Uptime":"0T00:01:13","Heap":13,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":655,"POWER":"ON","Dimmer":100,"Color":"1A1A1AFFFF","HSBColor":"0,0,10","Channel":[10,10,10,100,100],"CT":327,"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"dims","BSSId":"B4:FB:E4:44:09:D2","Channel":8,"RSSI":82,"LinkCount":1,"Downtime":"0T00:00:04"}}}
11:45:15 CMD: status 11
11:45:15 MQT: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-10-13T11:45:15","Uptime":"0T00:01:22","Heap":13,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":720,"POWER":"ON","Dimmer":100,"Color":"1A1A1AFFFF","HSBColor":"0,0,10","Channel":[10,10,10,100,100],"CT":327,"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"dims","BSSId":"B4:FB:E4:44:09:D2","Channel":8,"RSSI":82,"LinkCount":1,"Downtime":"0T00:00:04"}}}
11:45:25 CMD: status 11
11:45:25 MQT: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-10-13T11:45:25","Uptime":"0T00:01:32","Heap":13,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":792,"POWER":"ON","Dimmer":100,"Color":"1A1A1AFFFF","HSBColor":"0,0,10","Channel":[10,10,10,100,100],"CT":327,"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"dims","BSSId":"B4:FB:E4:44:09:D2","Channel":8,"RSSI":82,"LinkCount":1,"Downtime":"0T00:00:04"}}}
11:45:35 CMD: status 11
11:45:35 MQT: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-10-13T11:45:35","Uptime":"0T00:01:42","Heap":13,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":867,"POWER":"ON","Dimmer":100,"Color":"1A1A1AFFFF","HSBColor":"0,0,10","Channel":[10,10,10,100,100],"CT":327,"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":" [... cut ...]
11:45:57 MQT: stat/sonoff/RESULT = {"Command":"Unknown"}
11:46:00 CMD: status 11
11:46:00 MQT: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-10-13T11:46:00","Uptime":"0T00:02:07","Heap":13,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"POWER":"ON","Dimmer":100,"Color":"1A1A1AFFFF","HSBColor":"0,0,10","Channel":[10,10,10,100,100],"CT":327,"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"dims","BSSId":"B4:FB:E4:44:09:D2","Channel":8,"RSSI":82,"LinkCount":1,"Downtime":"0T00:00:04"}}}`

andrethomas, 2019-10-13:
yes because it is using sleep 0 which is too low for the ESP to keep up - you need to find a sleep value where the loadavg settles on <= 100

delinend, 2019-10-13:
Have now tryed sleep 0 and up to 50 in 5 steps. Reboot after every change, and I can't stop LoadAvg from climing to over 999 after around 60 seconds. It always goes over 999 after around 60 seconds.

jziolkowski, 2019-10-13:
My H801 with standard settings goes to LoadAvg 999 all the time and pings/wifi work normally

delinend, 2019-10-13:
I stille see LoadAvg 999, even with WiFi channel shift from ch.8 to ch.9. But the PING issue disapears when changed to channel 9.

@jziolkowski What's your distance from AP to H801, and WiFi channel? Do you see PING replys over 10ms when using channel 5, 6, 7 ,8 with LED's ON ? It's clear to see in the PING reply list, if switching LED's OFF, and the PING's is responding normal (2mms->5ms.).

But with **#define PWM_LIGHTSCHEME0_IGNORE_SLEEP** enabled, the LoadAvg is only around ~20 and all channels PING respond fine from 2ms to 5ms.

**Here channel change from 8 to 9:**
`11:49:26 CMD: status 11
11:49:26 MQT: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-10-13T11:49:26","Uptime":"0T00:05:33","Heap":14,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"POWER":"ON","Dimmer":100,"Color":"1A1A1AFFFF","HSBColor":"0,0,10","Channel":[10,10,10,100,100],"CT":327,"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"dims","BSSId":"B4:FB:E4:44:09:D2","Channel":8,"RSSI":82,"LinkCount":1,"Downtime":"0T00:00:04"}}}
11:50:15 MQT: Attempting connection...
11:50:15 MQT: Connected
11:50:15 MQT: tele/sonoff/LWT = Online (retained)
11:50:15 MQT: cmnd/sonoff/POWER = 
11:52:49 CMD: status 11
11:52:49 MQT: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-10-13T11:52:49","Uptime":"0T00:08:56","Heap":14,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"POWER":"ON","Dimmer":100,"Color":"1A1A1AFFFF","HSBColor":"0,0,10","Channel":[10,10,10,100,100],"CT":327,"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"dims","BSSId":"B4:FB:E4:44:09:D2","Channel":9,"RSSI":82,"LinkCount":2,"Downtime":"0T00:00:10"}}}
11:53:39 CMD: status 11
11:53:39 MQT: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-10-13T11:53:39","Uptime":"0T00:09:46","Heap":14,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"POWER":"ON","Dimmer":100,"Color":"1A1A1AFFFF","HSBColor":"0,0,10","Channel":[10,10,10,100,100],"CT":327,"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"dims","BSSId":"B4:FB:E4:44:09:D2","Channel":9,"RSSI":84,"LinkCount":2,"Downtime":"0T00:00:10"}}}`

jziolkowski, 2019-10-13:
Distance is around 2m :) I'll look into channel config later because I have some wifi issues currently. I'm on channel 1 right now.

andrethomas, 2019-10-13:
The wifi channel is not firmware related - its about how much other wifi noise you have in your location.