arendst/Tasmota #6609: Power monitoring (Blitz wolf SHP module) Not showing correct Amps & watts value even after calibration

ceaswaran, 2019-10-10:
**Power monitoring ERROR (Blitz wolf SHP module) Not showing correct Amps & watts value even after calibration**

Recently bought a 16A power monitoring plug from aliexpress.
After tuya converting, was able to figure out from the Module to be BlitzWolf SHP on Tasmota.

**It uses BL0937 Single Phase Energy Monitor Chip** 

After doing Power monitoring calibration using 100W incandescent bulb , i observed the power values as follows, which seem to work well.

[image]

But, when I connected it to 1000W water heater after calibration, I see the W=95W (As against 1000W) & A=0.416 Amps (As against 3.5Amps approx)
[image]

l tried connecting the plug to different wattage devices (Air conditioner/iron box) & I observed its watts value never changed beyond 101W, amps beyond 0.4 Amps 

I have a set of 3 plugs & all seem to show the same problem.
Below is my Tasmota configuration: Blitzwolf SHP module.

[image]

[](https://www.aliexpress.com/item/33008720018.html?spm=a2g0o.cart.0.0.40d33c00TyhIxF&mp=1) 

What is going wrong? Please help

ascillato2, 2019-10-10:
Please, could you be so kind on completing the [troubleshooting template](https://github.com/arendst/Sonoff-Tasmota/issues/new/choose) in order to have more information so as to properly help you?

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

ceaswaran, 2019-10-11:
@ascillato below is template details for this plug
`{"NAME":"BlitzWolf SHP","GPIO":[57,255,56,255,0,134,0,0,131,17,132,21,0],"FLAG":0,"BASE":45}`

ascillato, 2019-10-11:
We need the troubleshooting issue template that ask you for several information that we need for helping you. Please, click on the link provided for the template.

When you open a new issue you have 4 options: bug, troubleshooting, features request and blank. You are using blank issue that don't provide to us much information.

ascillato2, 2019-10-11:
Please, try latest Tasmota version (6.6.0.18) that have several fixs on power monitoring.

If after that, your issue persist, address this to the Tasmota support chat or open a new issue with the troubleshooting template completed. Thanks

closed 2019-10-11 by ascillato2