arendst/Tasmota #6614: Shutter Position and slider not showed in homeassistant

snakuzzo, 2019-10-10:
### BUG DESCRIPTION
_Position is not showed on homeassistant_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Sonoff-Tasmota/issues)
- [X] Searched the problem in the [wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [Sonoff Dual R2] Device used (e.g., Sonoff Basic): _____
- [6.6.0.17] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [X] Self-compiled
    - [X] IDE / Compiler used: VS Code
- [X] Flashing tools used: VS Code and also esptool
- [X] Provide the output of command: ``Backlog Template; Module; GPIO``:
  ```
  Configuration output here:
  23:54:43 MQT: stat/sonoff-cletto/RESULT = {"NAME":"Sonoff Dual R2","GPIO":[17,255,0,255,0,22,18,0,21,56,0,0,0],"FLAG":0,"BASE":39}
  23:54:43 MQT: stat/sonoff-cletto/RESULT = {"Module":{"0":"Sonoff Dual R2"}}
  23:54:43 MQT: stat/sonoff-cletto/RESULT = {"GPIO1":{"0":"None"},"GPIO3":{"0":"None"}}
  ```

- [ ] If using rules, provide the output of this command: ``Backlog Rule1; Rule2; Rule3``:
  ```
  Rules output here:
  No rules
  ```
- [ ] Provide the output of this command: ``Status 0``:
  ```
  STATUS 0 output here:
  23:55:50 MQT: stat/sonoff-cletto/STATUS = {"Status":{"Module":0,"FriendlyName":["Sonoff","Sonoff2"],"Topic":"sonoff-cletto","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1}}
  23:55:50 MQT: stat/sonoff-cletto/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:24:26","StartupUTC":"2019-10-10T22:31:24","Sleep":50,"CfgHolder":4617,"BootCount":7,"SaveCount":41,"SaveAddress":"FB000"}}
  23:55:50 MQT: stat/sonoff-cletto/STATUS2 = {"StatusFWR":{"Version":"6.6.0.17(sonoff)","BuildDateTime":"2019-10-10T23:43:29","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
  23:55:50 MQT: stat/sonoff-cletto/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["SnakeRM_24",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000C029","2805C8000100060000005A64000000000000","40002000"]}}
  23:55:50 MQT: stat/sonoff-cletto/STATUS4 = {"StatusMEM":{"ProgramSize":521,"Free":480,"Heap":18,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","8FDAE397","003683A1","22B617CD","01001BC0","00000181"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
  23:55:50 MQT: stat/sonoff-cletto/STATUS5 = {"StatusNET":{"Hostname":"sonoff-cletto-2783","IPAddress":"192.168.1.40","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"1.1.1.1","Mac":"DC:4F:22:97:6A:DF","Webserver":2,"WifiConfig":4}}
  23:55:50 MQT: stat/sonoff-cletto/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.10","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_976ADF","MqttUser":"snakuzzo","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
  23:55:50 MQT: stat/sonoff-cletto/STATUS7 = {"StatusTIM":{"UTC":"Thu Oct 10 22:55:50 2019","Local":"Thu Oct 10 23:55:50 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"07:02","Sunset":"18:11"}}
  23:55:50 MQT: stat/sonoff-cletto/STATUS10 = {"StatusSNS":{"Time":"2019-10-10T23:55:50","Shutter1":{"Position":36,"direction":0}}}
  23:55:50 MQT: stat/sonoff-cletto/STATUS11 = {"StatusSTS":{"Time":"2019-10-10T23:55:50","Uptime":"0T00:24:26","UptimeSec":1466,"Heap":18,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"SnakeRM_24","BSSId":"10:13:31:BC:5D:59","Channel":6,"RSSI":74,"LinkCount":1,"Downtime":"0T00:00:04"}}}
  ```
- [X] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ ``weblog 4`` _for more debug information)_
  ```
  Console output here:
  
  23:56:34 MQT: stat/sonoff-cletto/RESULT = {"WebLog":4}
  23:56:34 CFG: Saved to flash at FA, Count 42, Bytes 4096
  23:56:39 MQT: tele/sonoff-cletto/STATE = {"Time":"2019-10-10T23:56:39","Uptime":"0T00:25:15","UptimeSec":1515,"Heap":18,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":21,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"SnakeRM_24","BSSId":"10:13:31:BC:5D:59","Channel":6,"RSSI":78,"LinkCount":1,"Downtime":"0T00:00:04"}}
  23:56:39 MQT: tele/sonoff-cletto/SENSOR = {"Time":"2019-10-10T23:56:39","Shutter1":{"Position":36,"direction":0}}
  23:56:47 HTP: Main Menu
  23:56:50 SRC: WebGui from 192.168.1.2
  23:56:50 CMD: Group 0, Index 1, Command "SHUTTERPOSITION", Data "60"
  23:56:50 SHT: Position in: payload 60, index 1, source 7
  23:56:50 SHT: lastsource 7:, realpos 14000, target 22800, payload 60
  23:56:50 SHT: Start shutter in direction 1
  23:56:50 SRC: Shutter
  23:56:50 SHT: Switched relay: 1 by Shutter
  23:56:50 MQT: stat/sonoff-cletto/RESULT = {"POWER1":"ON"}
  23:56:50 MQT: stat/sonoff-cletto/POWER1 = ON (retained)
  23:56:50 MQT: stat/sonoff-cletto/RESULT = {"ShutterPosition1":60}
  23:56:50 WIF: Checking connection...
  23:56:50 WIF: Connected
  23:56:50 SHT: Shutter 0: Real Pos: 15000, Target 22800, source: Shutter, start-pos: 36 %, direction: 1, rtcshutter: 0.5  [s]
  23:56:50 CFG: Saved to flash at F9, Count 43, Bytes 4096
  23:56:51 SHT: Shutter 0: Real Pos: 17000, Target 22800, source: Shutter, start-pos: 36 %, direction: 1, rtcshutter: 1.5  [s]
  23:56:52 SHT: Shutter 0: Real Pos: 19000, Target 22800, source: Shutter, start-pos: 36 %, direction: 1, rtcshutter: 2.5  [s]
  23:56:53 SHT: Shutter 0: Real Pos: 21000, Target 22800, source: Shutter, start-pos: 36 %, direction: 1, rtcshutter: 3.5  [s]
  23:56:54 SHT: Shutter 0: Real Pos. 22800, Stoppos: 60, relay: 0, direction 1, pulsetimer: 0, rtcshutter: 4.4 [s], operationtime 4
  23:56:54 MQT: stat/sonoff-cletto/SHUTTER1 = 60 (retained)
  23:56:54 SRC: Shutter
  23:56:54 SHT: Switched relay: 1 by Shutter
  23:56:54 MQT: stat/sonoff-cletto/RESULT = {"POWER1":"OFF"}
  23:56:54 MQT: stat/sonoff-cletto/POWER1 = OFF (retained)
  23:56:54 MQT: tele/sonoff-cletto/RESULT = {"Shutter1":{"Position":60,"direction":0}}
  23:56:54 CFG: Saved to flash at F8, Count 44, Bytes 4096
  ```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Just open or close the shutter

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Value showed on HA

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

I followed every step in the wiki and I used the "Another Home Assistant integration example:" to configure my cover. No slider and position showed on homeassistant

```
- platform: mqtt
  name: "Test"
  availability_topic: "tele/sonoff-cletto/LWT"
  state_topic: "stat/sonoff-cletto/RESULT"
  command_topic: "cmnd/sonoff-cletto/Backlog"
  value_template: '{{value_json.ShutterPosition1}}'
  qos: 1
  retain: true
  payload_open: "ShutterOpen1"
  payload_close: "ShutterClose1"
  payload_stop: "ShutterStop1"
  state_open: "ON"
  state_closed: "OFF"
  payload_available: "Online"
  payload_not_available: "Offline"
  optimistic: true
  tilt_command_topic: 'cmnd/sonoff-cletto/ShutterPosition1'
  tilt_status_topic: 'cmnd/sonoff-cletto/ShutterPosition1'
  set_position_topic: 'cmnd/sonoff-cletto/ShutterPosition1'
  tilt_min: 0
  tilt_max: 100
  tilt_closed_value: 0
  tilt_opened_value: 100
```

**(Please, remember to close the issue when the problem has been addressed)**

meingraham, 2019-10-11:
There may be an error in the documentation. I think that the tilt_status_topic should be stat/<topic>/ShutterPosition1

ascillato2, 2019-10-11:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you to config your device.

Thanks.

---------------------------

### Support Information 
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-10-11 by ascillato2

snakuzzo, 2019-10-14:
@ascillato I think the third example in the wiki is incomplete.
If I use the third example after homeassistant 0.100.2 update, I've got an ERROR beacuse `position_topic` missing.
Maybe it can help someone...I solved using this configuration under homeassistant. Look at `value_template` and `position_topic`

```
- platform: mqtt
  name: "matrimoniale"
  availability_topic: "tele/sonoff-cletto/LWT"
  state_topic: "stat/sonoff-cletto/RESULT"
  command_topic: "cmnd/sonoff-cletto/Backlog"
  value_template: '{{ value | int }}'
  qos: 1
  retain: true
  payload_open: "ShutterOpen1"
  payload_close: "ShutterClose1"
  payload_stop: "ShutterStop1"
  state_open: "ON"
  state_closed: "OFF"
  payload_available: "Online"
  payload_not_available: "Offline"
  optimistic: true
  set_position_topic: 'cmnd/sonoff-cletto/ShutterPosition1'
  position_topic: "stat/sonoff-cletto/SHUTTER1"
```

ascillato, 2019-10-14:
Hi, thanks for sharing your config. Please, can you complete the example of the wiki?

Another point. Retain to true is not recommended. Really. Will give you a lot of headaches. You should put it to false.

snakuzzo, 2019-10-14:
Ok...thanks for the retain suggestion.
Wiki page modified. I've changed also retain and optimistic to false...is it correct?

ascillato, 2019-10-14:
Yes, it is better.

The retain option is for other type of automations where you can't have a real-time feedback of the actual value or state of a sensor. In this case, you can ask Tasmota for its state and you don't need to "remember" the last known state.

The problem with retain in distributed control systems (like this one that yours because you have smart devices) is that if your wifi drops and your Tasmota change any of its states, then when your wifi or your home assistant come on line again, it will force your Tasmota to go to the previous retained state.

With retained to false, you will have the actual state. On, off, offline etc. 

In the case your home assistant is restarted but your mqtt broker is kept online, you will need to make an automation in home assistant to ask to all your devices for its state. That automation is explained also in the wiki.

snakuzzo, 2019-10-14:
are you talking about announce automation...right?

This is my automation triggered on homassistant start (it's based on group topic)

```
- id: '1552268240479'
  alias: Shelly and Sonoff Announce
  trigger:
  - event: start
    platform: homeassistant
  condition: []
  action:
  - data:
      topic: cmnd/sonoffs/POWER
    service: mqtt.publish
  - data:
      topic: tele/sonoff-sensors/SENSOR
    service: mqtt.publish

```

ascillato, 2019-10-14:
Yes, that one. You can add to that automation also to check sensors, etc.

snakuzzo, 2019-10-14:
the second topic is for a temperature nodemcu+dht22 sensor

Gifford47, 2024-10-06:
@stefanbode 
Is it possible to get a cover entity (for a configured shutter) in Homeassistent via autodiscovery?
F.e. a pwm gpio on tasmota site could be interpreted as cover entity on tasmota site.

stefanbode, 2024-10-06:
Yes this should be the default. All my covers show up correctly.