arendst/Tasmota #6615: Buzzer 0 should not beep

CyiDev, 2019-10-11:
**Have you looked for this feature in other issues and in the wiki?**  
Yes

**Is your feature request related to a problem? Please describe.**  
When using the Buzzer function the only way to currently disable it that I am aware of is to issue Buzzer 0;  This itself creates a beep.  When implementing this in a few projects it would be desirable to not have it beep.

**Describe the solution you'd like**  
Buzzer 0; stops existing buzzer commands but does not issue a beep itself.  If Buzzer 0; is desired please provide an alternate solution like Buzzer stop or StopBuzzer;

**Describe alternatives you've considered**  
Power cycling the buzzer (takes too long)

andrethomas, 2019-10-11:
I'm not totally sure I understand what you are saying... under what circumstances could you possibly want to abort a buzzer sequence?

meingraham, 2019-10-11:
Probably less of abort (although a long tune?) but a temporary/permanent disabling of it?

CyiDev, 2019-10-11:
i issue the sequence buzzer 200,5,5;. if i need to stop it i would like to stop it.  this is being integrated into ha as an alarm of sorts.

ascillato2, 2019-10-12:
Closing this issue as it has been fixed. Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-10-12 by ascillato2