arendst/Tasmota #6617: Sensor which can read a value from a webserver (electricity price)

rest77, 2019-10-11:
I would like a sensor which can read the electricity spot price from a website an start or stop a device according to some limits. In python I could extract the value from this response:

resp = requests.get(url="http://www.nordpoolspot.com/api/marketdata/page/10")

How should I start if I want to use Tasmota?

Any ideas?

/Reik

andrethomas, 2019-10-11:
I believe the simplest would be to implement this on your automation software and have it send an on/off as required to the Tasmota device.

rest77, 2019-10-11:
That would be my first choise as well, but... This application should be used in a standalone sonoff. It will be connected to wifi and internet but no homeassiant or similar system will be available.

andrethomas, 2019-10-11:
It's not that I don't understand what you would like to do.

That link provides a JSON file which is ~320KB in size and then you still want to perform some post-processing on this data to make a "decision" on whether something should be on or off.

jziolkowski, 2019-10-11:
With how limited the ESP is, I don't see this working on standalone Tasmota.

Jason2866, 2019-10-11:
It is not possible with Tasmota since the ESP82xx has not enough hardware resources for.
Please close!
P.S. With a RPi this should be doable.

rest77, 2019-10-11:
With a Pi, no problem. I only need one value out of the 320kb. If I can solve to just get this value with the right request, would it be possible than?

ascillato2, 2019-10-11:
Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-10-11 by ascillato2

rest77, 2019-10-11:
Thanks for the quick respons anyway!

joba-1, 2019-10-11:
it is definitely not impossible, if you want to give it a try. It doesn’t matter how big the page is you need to read.

Read the http request body as a stream in small chunks (as is best practice anyways) and stop decoding once you have the value you are after.

Not sure if reusable code for this already exists in tasmota (read http stream, decode json stream, trigger internal action). There are better experts than me on this here. But if not, all of this is not a big deal to implement.

andrethomas, 2019-10-11:
I can see how simple wget like applications for updating dyndns systems being something that might be useful to some users but I don't think there's a way to make a solution useful to others which will do what this request would require.

Maybe with sd card and some enhancements on the scripting engine, it could work... I don't know for sure though.