arendst/Tasmota #6620: Add support for Xiaomi induction cooker

arzaman, 2019-10-11:
Could be very interesting to tasmotize the Xiaomi Mi Induction cooker

the device is a smart induction cooker based on ESP8266 controller where you can regulate power form 0 to 99, switch between the various modes, set the timer and other basic functions by using dedicated App.

[image]

https://xiaomi-mi.com/appliances/mijia-induction-cooker/

Here an interesting teardown that show internal construction and ESP8266 controller that exposes the TX/RX PIN

https://aquarat.co.za/xiaomi-precision-induction-cooker-teardown/

using Tasmota to control sensor and  output power of  the induction coli could be very useful

thanks for your interest
br
Davide

ascillato2, 2019-10-11:
Hi,

You have to check how are controlled the electronics (if there is anothe MCU besides the ESP8266 and how it communicate) to see if it can be reversed engineer for making a template or a Driver for Tasmota.

Also, before testing, make sure to make a copy of the flash of the ESP8266 (using esptool.py as explained in the wiki)

If you need assistance, please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Thanks.

---------------------------

### Support Information 
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-10-11 by ascillato2

arzaman, 2019-10-11:
Ok thanks for considering the Xiaomi induction cooker as potential interesting device for tasmota project. 
Not sure I have all the needed skill to make a complete reverse engineering but I can try to collect more info and make a copy of ESP8266 flash
I will ask for sure help and the discord chat 

thanks again
Davide

ascillato, 2019-10-11:
Also remember that tinkering a device like that will void the warranty. Besides, all safety considerations has to be taken in place. We don't know if the ESP8266 has some safety routines to prevent damage or fire to the induction cooker. In case it has, they must be programmed in a driver.

So a deep understanding of this appliance is needed in order to replace the original software. I personally don't like much the idea just for safety concerns. I would stay with the original firmware just in case.

arzaman, 2019-10-11:
Totally agree on your suggestion especially on safety
What is potential interesting that the communication between the controller and the induction coil power supply is done on I2C inerface...so most likely understanding the dialog between the controller and power supply we can see how the "power" part is manged
My idea is to add a sort of PID temperature controller loop to the original device. Now you can regulate (PWM ??) just the power applied to the heater but no feedback on temperature

hossF, 2019-10-12:
I am trying to control to start, stop and control the temperature of the Xiamoi Induction stove.
I succeded to stop it using the raw command: 
mirobo --ip [the_IP] --token [the_Token] raw-command set_func '["end02"]'. 
Does anyone know if it's possible to start the induction cooker or/and control the temperatures using the raw commands? If not, I would like to contribute to make this control possible.  
Thanks.
Houssem