arendst/Tasmota #6621: Feature Request: Script engine - save code from web interface compacted by Javascript

th-s-op, 2019-10-11:
The optional script engine in Tasmota is a great and extremely powerful enhancement.
From my understanding, the script from the ("Edit Script") web interface's text field is saved to the device 1:1. Realistically, complex scripts need some comments due to their small footprint's nature. But comments take away all the valuable space. But while it makes sense to have the original script in a local editor, filled with comments and made well readable, when pasting to the UI editor and saving, it would be a cool feature to shorten/compact the code before sending by calling a Javascript function. This could:
- completely strip comments
- strip white space, at least from beginning/end of lines

I would offer myself to write such a Javascript function which takes a (script code) string and returns the compacted version.

I just don't have enough knowledge of the Tasmota code internals to be able to build it into the frontend, so someone would have to take the .js and (optionally, with a compiler #define?)
- include the JS code into the HTML header
- add the funcion call to the save button
- probably add a check box to enable/disable automatic compation

Please tell me if that's something you also like. If you're not going to include this into Tasmota directly, I'll just create a User Script for myself, intercepting and live changing the DOM/Javascript of stock Tasmota. This still also might me interesting for hard core developers.

ascillato, 2019-10-11:
Hi,

So, your feature request is to have an optimized editor for the scripter that reduces the use of spaces and comments right?

In that case, there is no need to have exclusively JS code inside the HTML, due to that "optimization" can be also done in C and inside the scripter driver before saving.

Inside the scripter code, this request is already in the TO DO list from @gemu2015 :

https://github.com/arendst/Sonoff-Tasmota/blob/7c8f2daa10a24e15e34b86c89011509794b560a0/sonoff/xdrv_10_scripter.ino#L22-L38

Anyway, the actual max script size is 1535 bytes and it is up to the user if he/she wants their comments and spaces to be erased or not, so, this optimization should be optional.

th-s-op, 2019-10-11:
Yes, that's what I meant - script code compaction so that an author can use comments for readability but they are not saved into the Tasmota memory.

I don't understand the code fragment above to reflect this as a todo. He for sure means that he wants to keep his C code optimized for space, not the script code.

Plus, my thought was that those max size of 1535 would still be the limit - if the compaction code would be in C, it would first have to be loaded into memory, then shortened. If it's dont in the user's browser, it would be downloaded in an already compact version.

Example:
- I have a script that including a good amount of comments could be 3000 characters length
- before saving, Javascript would strip comments and Tasmota only sees a shortened version of let's say 1530 characters

Are you sure that in this example, it would be possible to download 3000 characters into a Tasmota buffer and do the stripping to 1530 chars in C code?

ascillato, 2019-10-12:
>it would be possible to download 3000 characters into a Tasmota buffer and do the stripping to 1530 chars in C code?

With the actual scripter implementation, Tasmota will process only the first 1535 bytes.

gemu2015, 2019-10-12:
i added this as an option
#define SCRIPT_STRIP_COMMENTS

however there are potential problems.
the html variable textsize is normally set to script size and limits the text size on entry.
since i do not know how many chars to strip i now set text size to twice the actual size.
if your real code size is larger than the Tasmota limit the code is shortened.

if you enable this option in the compiler you can switch it on and off by
script 3 switch option on
script 2 switch option off
 
i myself and a couple of users however expand the buffer by the EEPROM option.
simply add a cheap (1 Euro) EEPROM to the i2c Bus and have at least 4k (8k) script buffer

https://www.ebay.de/itm/AT24C256-Serial-EEPROM-I2C-Interface-EEPROM-Data-Storage-Module-for-Arduino-NEU/233026828988?hash=item36417b3abc:g:tq4AAOSwMO9cALBC

update on my fork, ill do a pr later

gemu2015, 2019-10-12:
@th-s-op 
i implemented the Javascript version of compression. (currently no ON,OFF switch)
#define SCRIPT_STRIP_COMMENTS

the c version is still present with
#define xSCRIPT_STRIP_COMMENTS

after more testing i will do a pr

th-s-op, 2019-10-12:
@gemu2015 great, really looking forward to use this! And thanks for the EEPROM tip, I'll have a look at it.

While I am at it - a little bit off topic - is there a way to compile Tasmota in a way that a script is "hard burned" into it, ie if a Reset Configuration is done, this script is back to how it was set at compile time?
I do such things in settings.ino in a function which initializes the defaults. Is there a way to set the script string also?

gemu2015, 2019-10-13:
@th-s-op 
according the javascript version i am struggling with several issues. (the current version only pastes exactly one time, then again only after a page reload)
i am not very experienced in javascript. You may look at the code and find a better solution. (https://github.com/arendst/Sonoff-Tasmota/blob/development/sonoff/xdrv_10_scripter.ino)
the desired final implementation would be to have a special paste (e.g. shift paste) that removes comments and normal paste stays as usual.

you may set a default script in settings.ino by initializing the settings.rules[0]; space with a cast to a default script. (i treat the 3 consecutive rules buffers as one script buffer)

gemu2015, 2019-10-15:
i now implemented the following:
if you paste anything with fewer chars than maxlength paste acts as usual.
else clipboard is compressed  and inserted into complete textarea.
you may give it a try

PR #6656 (2019-10-16): scripter button events, remove comments on paste
## Description:

**Related issue (if applicable):** fixes #6643 >
section >b catches button change event   bt[x] (x=1.. MAX_KEYS) gets button state
#define USE_BUTTON_EVENT

**Related issue (if applicable):** fixes #6621>
initial support for stripping comments on pastes larger then max length

support for \n \r on commands text

small fixes

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  - [x] The code change is tested and work [... cut]
closed 2019-10-16 by arendst