arendst/Tasmota #6623: merkury bw942 bulb - can not switch

or78rXxZ, 2019-10-12:
> **GUIDE**  
>
> This troubleshooting issue template is meant to help Tasmota users with difficult problems. It is aimed to be opened if using the wiki and the support chat could not solve the issue. The Github Issue tracker is NOT a general discussion forum!
>   
> Please DO NOT OPEN AN ISSUE:  
>  - If you have general questions or you need help on Tasmota usage, go to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4)
>  - If your Tasmota version is not the latest from the development branch, please update your device before submitting your issue. Your problem might already be solved. The latest precompiled binaries of Tasmota can be downloaded from http://thehackbox.org/tasmota/  
>  - If your issue is about a new device, please use the Tasmota [Template](../wiki/Templates) feature.
>  - If your issue is a flashing issue, please address it to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4)
>  - If your issue is compilation problem, please address it to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4)
>  - If your issue has been addressed before (i.e., duplicated issue), please ask in the original issue
>  - If your issue is a Wi-Fi problem or MQTT problem, please try the steps provided in the [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) and troubleshooting wiki articles  
>
> Please take a few minutes to complete the requested information below. Our ability to provide assistance is greatly hampered without it. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help you.

### ISSUE DESCRIPTION - TROUBLESHOOTING
_A clear description of what the issue is and be as extensive as possible_
this is a merkury mi-bw942-999w smart wifi dimmable led bulb
i installed tasmota with no problems and configured it
i set module as generic (18)
i tried gpios 2, 4, 5, 12, 13, 14, 15, 16 as pwm1 and relay1
in all cases the bulb will not switch on
is there a way to trouble shoot this short of destroying the bulb

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): merkury mi-bw942-999w
- [x] Tasmota binary firmware version number used: 
Program Version | 6.6.0.7(digiMIN)
Build Date & Time | 2019-10-01T20:40:34
Core/SDK Version | 2_3_0/1.5.3(aec24ac9)
  -[x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [x] Provide the output of this command: ``Backlog Template; Module; GPIO``:
  ```
13:27:35 CMD: Backlog Template; Module; GPIO
13:27:35 RSL: stat/sonoff/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
13:27:36 RSL: stat/sonoff/RESULT = {"Module":"18 (Generic)"}
13:27:36 RSL: stat/sonoff/RESULT = {"GPIO0":"0 (None)","GPIO1":"0 (None)","GPIO2":"0 (None)","GPIO3":"0 (None)","GPIO4":"0 (None)","GPIO5":"37 (PWM1)","GPIO12":"0 (None)","GPIO13":"0 (None)","GPIO14":"0 (None)","GPIO15":"0 (None)","GPIO16":"0 (None)"}
  Configuration output here:
  
  
  ```
- [ ] If using rules, provide the output of this command: ``Backlog Rule1; Rule2; Rule3``:
  ```
  Rules output here:
  
  
  ```
- [ ] Provide the output of this command: ``Status 0``:
  ```
  STATUS 0 output here:
  13:29:03 CMD: Status 0
13:29:03 RSL: stat/sonoff/STATUS = {"Status":{"Module":18,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1}}
13:29:03 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Power on","Uptime":"0T00:28:22","StartupUTC":"2019-10-12T12:00:41","Sleep":50,"CfgHolder":4617,"BootCount":21,"SaveCount":83,"SaveAddress":"F9000"}}
13:29:03 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.6.0.7(digiMIN)","BuildDateTime":"2019-10-01T20:40:34","Boot":7,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
13:29:03 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Yv4Yo5xjegtbnjrb","ht1Kgdw0Hfltbkxa"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008029","280500000100060000005A00000000000000","00000000"]}}
13:29:03 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":401,"Free":600,"Heap":21,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","08008082","00248000","00000015","000000C0","00000001"],"Drivers":"1,2,3,4,6,10,16,22,24","Sensors":"1,2,3,4,5"}}
13:29:03 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-6546","IPAddress":"192.168.2.100","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.1","Mac":"2C:F4:32:B2:B9:92","Webserver":2,"WifiConfig":2}}
13:29:03 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_B2B992","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
13:29:03 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Sat Oct 12 12:29:03 2019","Local":"Sat Oct 12 13:29:03 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00"}}
13:29:03 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-10-12T13:29:03","Epoch":1570883343}}
13:29:03 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-10-12T13:29:03","Epoch":1570883343,"Uptime":"0T00:28:22","UptimeSec":1702,"Heap":21,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Dimmer":52,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"Yv4Yo5xjegtbnjrb","BSSId":"58:EF:68:48:92:87","Channel":6,"RSSI":94,"LinkCount":1,"Downtime":"0T00:00:06"}}}
  
  ```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ ``weblog 4`` _for more debug information)_
  ```
  Console output here:
  
  
  ```

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2019-10-12:
Please try latest development version from http://thehackbox.org/tasmota/pre-2.6/sonoff.bin
There where several changes in light driver since your version.

or78rXxZ, 2019-10-12:
i got http://thehackbox.org/tasmota/pre-2.6/sonoff-minimal.bin and loaded it
after restart i can not communicate with the device
i can see it requesting an address from dhcp and i can ping it
but no web interface

Jason2866, 2019-10-12:
Is it reachable via mqtt? If yes do the upgrade over mqtt via backlog command.
See wiki.

or78rXxZ, 2019-10-12:
i am not using mqtt so i never configured it

Jason2866, 2019-10-12:
:-(  If you cant reach tx rx for serial flash it looks like it is bricked

ascillato2, 2019-10-12:
Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-10-12 by ascillato2

or78rXxZ, 2019-11-01:
i have resurrected this device.
i have tried the generic 18 module and assigning pwm1 to each gpio.
none of them have any affect.
i tried the tuyamcu module.
the log shows no dpid's.
any suggestions as to how to proceed?