arendst/Tasmota #6625: Power monitoring (Blitz wolf SHP module) Not showing correct Amps & watts value even after calibration

ceaswaran, 2019-10-12:
### ISSUE DESCRIPTION - TROUBLESHOOTING
_Power monitoring ERROR (Blitz wolf SHP module) Not showing correct Amps & watts value even after calibration_

Recently bought a 16A power monitoring plug from aliexpress  (Model No: NX-SM223) & flashed with Tasmota (Using Tuya convert
After wiki & some google found that its Module to be BlitzWolf SHP on Tasmota.
**It uses BL0937 Single Phase Energy Monitor Chip** 

After performing Power monitoring calibration using 100W incandescent bulb, I observed the power values as follows, which seem to work well. (Everything Looks OK as per standards)
(https://github.com/arendst/Sonoff-Tasmota/wiki/Power-monitoring-calibration)

[image]  

But, when I connected it to 1000W water heater after calibration, I see the W=95W (As against 1000W) & A=0.416 Amps (As against 3.5Amps approx)
[image]

l tried connecting the plug to different wattage devices (Air conditioner/iron box) & I observed its watts value never changed beyond 101W, amps beyond 0.4 Amps 

I have a set of 3 plugs & all seem to show the same problem.
Below is my Tasmota configuration: Blitzwolf SHP module.

[image]

What is going wrong? 
Only for the calibrated appliance is shows the right Amps & Watts, for all other appliances its wrong.
Voltage seem to be OK for all devices.
Please help

### REQUESTED INFORMATION
- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): [https://www.aliexpress.com/item/33008720018.html?spm=a2g0s.9042311.0.0.737a4c4d9XgEEF]
  - [X] Pre-compiled (Sonoff.bin used v6.6.0)
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [X] Flashing tools used: EspEasy
- [X] Provide the output of this command: ``Backlog Template; Module; GPIO``:
  ```
`{"NAME":"BlitzWolf SHP","GPIO":[57,255,56,255,0,132,0,0,131,17,134,21,0],"FLAG":0,"BASE":45}`
`{"Module":"45 (BlitzWolf SHP)"}`
`{"GPIO1":"0 (None)","GPIO3":"0 (None)"}`
  
  
  ```
- `No rules used`
  STATUS 0 output here:
  Console output here:
`4:52:10 CMD: STATUS 0`
`14:52:10 RSL: stat/sonoffheat01/STATUS = {"Status":{"Module":45,"FriendlyName":["Sonoff"],"Topic":"sonoffheat01","ButtonTopic":"0","Power":1,"PowerOnState":1,"LedState":1,"Led``Mask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1}}`
  
  
  ```

ascillato2, 2019-10-12:
Closing this issue as it is **duplicated**. Sorry. Please update to latest Tasmota.

If you need further assistance, please address this to the Tasmota Support Chat

Please, read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-10-12 by ascillato2

ceaswaran, 2019-10-12:
I have updated to latest tasmota which is `Sonoff-Tasmota v6.6.0`
Still, the issue persist, hence I recreated the case here, no duplication. please open the same

Thanks in advance.

ascillato, 2019-10-12:
Latest Tasmota is 6.6.0.18. Please upgrade (http://thehackbox.org/tasmota/pre-2.6/sonoff.bin)

ceaswaran, 2019-10-13:
@ascillato I Found something Wierd/interesting. Am not able to upgrade my device to Latest Tasmota is 6.6.0.18.

**OTA method:** Tried both sonoff.minimal & sonoff.bin after uploading the bin my device 
is connected to wifi (Saw thru fing app) but No tasmota UI appears after i key in the IP address.
but am able to ping the device using app link FING. 
Also my TAsmo Admin (Home assistant) not discovering the device.

**FTDI wired method:** Same as above Nodemcupyflahser & Easpeasy flasher shows Flash successful but no UI seen or sonoffXXX hotspot identified.

Still, am able to flash lower version of Tasmota V6.5.0 using FTDI adapter & it works well to show UI and change the settings.
Is the chip memory too low to upgrade to next version??

[image]

Why is this happening? How do i upgrade to Latest Tasmota is 6.6.0.18

ascillato, 2019-10-13:
Something is corrupt in your flash. You need to full erase your flash with esptool.py as explained in the wiki and flash again with the precompiled bins with esptool.py also.

If you need further assistant, please find us in the Tasmota support chat. Thanks