arendst/Tasmota #6626: Tasmota is not subscribing to GroupTopic

celsworth, 2019-10-12:
# BUG DESCRIPTION

I noticed that sending commands to my group topic was not working, and on investigating why, I am not seeing GroupTopic subscribes from Tasmota devices to my broker. Coincidentally it doesn't seem to be subscribing to the backup topic either.

To confirm this, I turned on complete logging in mosquito (with `log_type all`) and saw the following after restarting it:

```
# grep -A1 SUBSCRIBE.*DVES /var/log/mosquitto/mosquitto.log
1570891580: Received SUBSCRIBE from DVES_60129F
1570891580: 	cmnd/sonoff-4767/# (QoS 0)
--
1570891580: Received SUBSCRIBE from DVES_60186B
1570891580: 	cmnd/sonoff-6251/# (QoS 0)
--
1570891580: Received SUBSCRIBE from DVES_6004E7
1570891580: 	cmnd/sonoff-1255/# (QoS 0)```

There is more, but they're all the same, each of my sockets is only subscribing on the full topic. I have customised my MQTT full topic as you can see in the screenshots below - I added `%hostname%` so that I can tell which socket messages have come from. But I don't think this should prevent the GroupTopic working? I didn't see anything in the documentation about it.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Sonoff-Tasmota/issues)
- [ ] Searched the problem in the [wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): [Anoopsyche JH-G01B1](https://blakadder.github.io/templates/anoopsyche_JH-G01B1.html)
- [ ] Tasmota binary firmware version number used: 6.6.0(release-sonoff)
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: ``Backlog Template; Module; GPIO``:

```
16:26:09 MQT: stat/sonoff-1255/RESULT = {"NAME":"JH-G01B1","GPIO":[0,145,0,146,0,0,0,0,17,56,21,0,0],"FLAG":0,"BASE":41}
16:26:09 MQT: stat/sonoff-1255/RESULT = {"Module":"0 (JH-G01B1)"}
16:26:10 MQT: stat/sonoff-1255/RESULT = {"GPIO":"Not supported"}  
```

- [ ] If using rules, provide the output of this command: ``Backlog Rule1; Rule2; Rule3``:
  ```
  Rules output here:
  
  
  ```

- [ ] Provide the output of this command: ``Status 0``:

```
16:26:36 MQT: stat/sonoff-1255/STATUS = {"Status":{"Module":0,"FriendlyName":["Lounge Corner"],"Topic":"sonoff","ButtonTopic":"0","Power":1,"PowerOnState":0,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":0,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1}}
16:26:36 MQT: stat/sonoff-1255/STATUS1 = {"StatusPRM":{"Baudrate":4800,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software Watchdog","Uptime":"2T03:29:13","StartupUTC":"2019-10-10T11:57:23","Sleep":50,"CfgHolder":4617,"BootCount":9,"SaveCount":31,"SaveAddress":"FB000"}}
16:26:36 MQT: stat/sonoff-1255/STATUS2 = {"StatusFWR":{"Version":"6.6.0(release-sonoff)","BuildDateTime":"2019-07-06T13:10:20","Boot":4,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
16:26:36 MQT: stat/sonoff-1255/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":2,"LogHost":"192.168.0.125","LogPort":514,"SSId":["AirPort",""],"TelePeriod":60,"Resolution":"569380C0","SetOption":["04008028","280500000100068000005A00000000000000","00000028"]}}
16:26:36 MQT: stat/sonoff-1255/STATUS4 = {"StatusMEM":{"ProgramSize":503,"Free":500,"Heap":14,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","0FDAE396","001783A0","22B617CD","01003BC0"]}}
16:26:36 MQT: stat/sonoff-1255/STATUS5 = {"StatusNET":{"Hostname":"sonoff-1255","IPAddress":"192.168.0.244","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.125","Mac":"84:0D:8E:60:04:E7","Webserver":2,"WifiConfig":2}}
16:26:36 MQT: stat/sonoff-1255/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.125","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_6004E7","MqttUser":"DVES_USER","MqttCount":12,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
16:26:36 MQT: stat/sonoff-1255/STATUS7 = {"StatusTIM":{"UTC":"Sat Oct 12 15:26:36 2019","Local":"Sat Oct 12 16:26:36 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"07:05","Sunset":"18:07"}}
16:26:36 MQT: stat/sonoff-1255/STATUS9 = {"StatusPTH":{"PowerDelta":0,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
16:26:36 MQT: stat/sonoff-1255/STATUS10 = {"StatusSNS":{"Time":"2019-10-12T16:26:36","ENERGY":{"TotalStartTime":"2019-10-08T17:54:03","Total":0.59366,"Yesterday":0.36411,"Today":0.05109,"Power":0.99,"ApparentPower":16.07,"ReactivePower":16.00,"Factor":0.06,"Voltage":247.0,"Current":0.065}}}
16:26:36 MQT: stat/sonoff-1255/STATUS11 = {"StatusSTS":{"Time":"2019-10-12T16:26:36","Uptime":"2T03:29:13","Heap":14,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"ON","Wifi":{"AP":1,"SSId":"AirPort","BSSId":"70:56:81:CE:38:FD","Channel":6,"RSSI":64,"LinkCount":1,"Downtime":"0T00:00:03"}}}
  ```

- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ ``weblog 4`` _for more debug information)_
  ```
  Console output here:
  
  
  ```

### TO REPRODUCE

Boot the device. Observe in the mosquito log that group topics are not subscribed to. And then obviously commands to the GroupTopic are ignored by all sockets.

### EXPECTED BEHAVIOUR

Expected GroupTopic to be subscribed :)

### SCREENSHOTS

![](https://i.imgur.com/llYhHkx.png)

arendst, 2019-10-12:
Bug???? Why. What is the grouptopic set to? Provide output of a restart with logging set to 4.

ascillato2, 2019-10-12:
Hi,

On my first test this was not working but was my mistake, I was issueing the command wrongly.

After retesting sending commands to grouptopic works fine:

```
12:46:24 CMD: grouptopic
12:46:24 SRC: WebConsole from 192.168.1.125
12:46:24 MQT: stat/luzescalera/RESULT = {"GroupTopic":"sonoffs"}
12:46:36 CMD: publish cmnd/sonoffs/power
12:46:36 SRC: WebConsole from 192.168.1.125
12:46:36 MQT: cmnd/sonoffs/power = 
12:46:37 MQT: Topic Recibido cmnd/sonoffs/power, Tamaño de Datos 0, Datos 
12:46:37 SRC: MQTT
12:46:37 MQT: stat/luzescalera/RESULT = {"POWER1":"OFF"}
12:46:37 MQT: stat/luzescalera/POWER1 = OFF
```

closed 2019-10-12 by ascillato2

celsworth, 2019-10-12:
GroupTopic is `sonoffs` it's in the status output above. Here's the restart log.. it got pretty noisy..

```
Oct 12 16:47:52 sonoff-1255 ESP-CMD: SysLog 4
Oct 12 16:47:52 sonoff-1255 ESP-MQT: stat/sonoff-1255/RESULT = {"SysLog":"4 (Active 4)"}
Oct 12 16:47:52 sonoff-1255 ESP-DMP: 55 5A 02 EE C8 00 03 12 00 3F 2F 05 0F F6 50 CB 98 33 EA 14 E1 C6 63 33
Oct 12 16:47:53 sonoff-1255 ESP-DMP: 55 5A 02 EE C8 00 03 12 00 3F 2F 05 0F F6 50 CB 98 35 83 79 E1 C6 63 33
Oct 12 16:47:53 sonoff-1255 ESP-DMP: 55 5A 02 EE C8 00 03 12 00 3F 2F 05 0F F6 50 CB 98 37 1B DF E1 C6 63 33
Oct 12 16:47:53 sonoff-1255 ESP-DMP: 55 5A 02 EE C8 00 03 12 00 3F 2F 05 0F F6 50 CB 98 38 B6 43 E1 C6 63 33
Oct 12 16:47:53 sonoff-1255 ESP-CFG: Saved to flash at F6, Count 36, Bytes 3584
Oct 12 16:47:53 sonoff-1255 ESP-DMP: 55 5A 02 EE C8 00 03 12 00 3F 2F 05 0F F6 50 CB 98 3A 4F A8 E1 C6 63 33
Oct 12 16:47:53 sonoff-1255 ESP-DMP: 55 5A 02 EE C8 00 03 12 00 3F 2F 05 0F F6 50 CB 98 3B E9 0D E1 C6 63 33
Oct 12 16:47:53 sonoff-1255 ESP-DMP: 55 5A 02 EE C8 00 03 12 00 3F 2F 05 0F F6 50 CB 98 3D 81 73 E1 C6 63 33
Oct 12 16:47:53 sonoff-1255 ESP-DMP: 55 5A 02 EE C8 00 03 12 00 3F 2F 05 0F F6 50 CB 98 3F 1A D8 E1 C6 63 33
Oct 12 16:47:53 sonoff-1255 ESP-DMP: 55 5A 02 EE C8 00 03 12 00 3F 2F 05 0F F6 50 CB 98 40 B5 3C E1 C6 63 33
Oct 12 16:47:53 sonoff-1255 ESP-DMP: 55 5A 02 EE C8 00 03 12 00 3F 2F 05 0F F6 50 CB 98 42 4D A2 E1 C6 63 33
Oct 12 16:47:53 sonoff-1255 ESP-DMP: 55 5A 02 EE C8 00 03 12 00 3F 2F 05 0F F6 50 CB 98 43 E7 07 E1 C6 63 33
Oct 12 16:47:54 sonoff-1255 ESP-DMP: 55 5A 02 EE C8 00 03 12 00 3F 2F 05 0F F6 50 CB 98 45 80 6C E1 C6 63 33
Oct 12 16:47:54 sonoff-1255 ESP-DMP: 55 5A 02 EE C8 00 03 12 00 3F 2F 05 0F F6 50 CB 98 47 1A D0 E1 C6 63 33
Oct 12 16:47:54 sonoff-1255 ESP-DMP: 55 5A 02 EE C8 00 03 12 00 3F 2F 05 0F F6 50 CB 98 48 B3 36 E1 C6 63 33
Oct 12 16:47:54 sonoff-1255 ESP-DMP: 55 5A 02 EE C8 00 03 12 00 3F 2F 05 0F F6 50 CB 98 4A 4C 9B E1 C6 63 33
Oct 12 16:47:54 sonoff-1255 ESP-DMP: 55 5A 02 EE C8 00 03 12 00 3F 2F 05 0F F6 50 CB 98 4B E6 FF E1 C6 63 32
Oct 12 16:47:54 sonoff-1255 ESP-DMP: 55 5A 02 EE C8 00 03 12 00 3F 2F 05 0F F6 50 CB 98 4D 80 64 E1 C6 63 33
Oct 12 16:47:54 sonoff-1255 ESP-DMP: 55 5A 02 EE C8 00 03 12 00 3F 2F 05 0F F6 50 CB 98 4F 18 CA E1 C6 63 33
Oct 12 16:47:54 sonoff-1255 ESP-DMP: 55 5A 02 EE C8 00 03 12 00 3F 2F 05 0F F6 50 CB 98 50 57 C9 F1 C6 63 82
Oct 12 16:47:54 sonoff-1255 ESP-DMP: 55 5A 02 EE C8 00 03 12 00 3F 2F 05 0F F6 50 CB 98 50 57 C9 F1 C6 64 83
Oct 12 16:47:55 sonoff-1255 ESP-DMP: 55 5A 02 EE C8 00 03 12 00 3F 2F 04 02 A4 50 CB 98 50 57 C9 F1 C6 64 23
Oct 12 16:47:55 sonoff-1255 ESP-DMP: message repeated 7 times: [ 55 5A 02 EE C8 00 03 12 00 3F 2F 04 02 A4 50 CB 98 50 57 C9 F1 C6 64 23]
Oct 12 16:47:55 sonoff-1255 ESP-DMP: 55 5A 02 EE C8 00 03 10 00 3F 2F 04 02 A4 50 CB 98 10 58 97 E1 C6 64 A0
Oct 12 16:47:55 sonoff-1255 ESP-DMP: 55 5A 02 EE C8 00 03 10 00 3F 2F 04 02 A4 50 CB 98 11 F1 FC E1 C6 64 9F
Oct 12 16:47:56 sonoff-1255 ESP-DMP: 55 5A 02 EE C8 00 03 10 00 3F 2F 04 02 A4 50 CB 98 13 8B 61 E1 C6 64 A0
Oct 12 16:47:56 sonoff-1255 ESP-MQT: tele/sonoff-1255/STATE = {"Time":"2019-10-12T16:47:55","Uptime":"0T00:01:15","Heap":14,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"ON","Wifi":{"AP":1,"SSId":"AirPort","BSSId":"70:56:81:CE:38:FD","Channel":6,"RSSI":64,"LinkCount":1,"Downtime":"0T00:00:04"}}
Oct 12 16:47:57 sonoff-1255 ESP-MQT: tele/sonoff-1255/SENSOR = {"Time":"2019-10-12T16:47:55","ENERGY":{"TotalStartTime":"2019-10-08T17:54:03","Total":0.59404,"Yesterday":0.36411,"Today":0.05147,"Period":0.01,"Power":1.02,"ApparentPower":15.09,"ReactivePower":15.10,"Factor":0.07,"Voltage":245.2,"Current":0.062}}
Oct 12 16:47:57 sonoff-1255 ESP-DMP: 55 5A 02 EE C8 00 03 10 00 3F 2F 04 02 A4 50 CB 98 15 23 C7 E1 C6 64 A0
Oct 12 16:47:57 sonoff-1255 ESP-DMP: 55 5A 02 EE C8 00 03 10 00 3F 2F 04 02 A4 50 CB 98 16 BE 2B E1 C6 64 A0
Oct 12 16:47:57 sonoff-1255 ESP-DMP: 55 5A 02 EE C8 00 03 10 00 3F 2F 04 02 A4 50 CB 98 18 57 90 E1 C6 64 A0
Oct 12 16:47:57 sonoff-1255 ESP-DMP: 55 5A 02 EE C8 00 03 10 00 3F 2F 04 02 A4 50 CB 98 19 F0 F5 E1 C6 64 9F
Oct 12 16:47:57 sonoff-1255 ESP-DMP: 55 5A 02 EE C8 00 03 10 00 3F 2F 04 02 A4 50 CB 98 1B 89 5B E1 C6 64 A0
Oct 12 16:47:57 sonoff-1255 ESP-DMP: 55 5A 02 EE C8 00 03 10 03 10 00 3F 2F 04 02 A4 50 CB 98 25 21 B9 E1 C6
Oct 12 16:47:57 sonoff-1255 ESP-CSE: Checksum failure
Oct 12 16:47:57 sonoff-1255 ESP-DMP: 55 5A 02 EE C8 00 55 5A 02 EE C8 00 03 10 00 3F 2F 04 02 A4 50 CB 98 29
Oct 12 16:47:57 sonoff-1255 ESP-CSE: Checksum failure
Oct 12 16:47:57 sonoff-1255 ESP-DMP: 55 5A 02 EE C8 00 03 10 00 3F 2F 04 02 A4 50 CB 98 29 EE E7 E1 C6 64 9F
Oct 12 16:47:57 sonoff-1255 ESP-DMP: 55 5A 02 EE C8 00 03 10 00 3F 2F 04 02 A4 50 CB 98 2B 88 4C E1 C6 64 A0
Oct 12 16:47:57 sonoff-1255 ESP-DMP: 55 5A 02 EE C8 00 03 10 00 3F 2F 04 02 A4 50 CB 98 2D 21 B1 E1 C6 64 A0
Oct 12 16:47:57 sonoff-1255 ESP-DMP: 55 5A 02 EE C8 00 03 10 00 3F 2F 03 60 F4 50 CB 98 2E BB 16 E1 C6 64 4D
Oct 12 16:47:57 sonoff-1255 ESP-DMP: 55 5A 02 EE C8 00 03 10 00 3F 2F 03 60 F4 50 CB 98 30 54 7B E1 C6 64 4D
Oct 12 16:47:58 sonoff-1255 ESP-DMP: 55 5A 02 EE C8 00 03 10 00 3F 2F 03 60 F4 50 CB 98 31 ED E0 E1 C6 64 4C
Oct 12 16:47:58 sonoff-1255 ESP-DMP: 55 5A 02 EE C8 00 03 10 00 3F 2F 03 60 F4 50 CB 98 33 87 45 E1 C6 64 4D
Oct 12 16:47:58 sonoff-1255 ESP-DMP: 55 5A 02 EE C8 00 03 10 00 3F 2F 03 60 F4 50 CB 98 35 1F AB E1 C6 64 4D
[... cut ...]
Oct 12 16:48:25 sonoff-1255 ESP-DMP: 55 5A 02 EE C8 00 03 11 00 3F 2F 03 31 6A 50 CB 98 65 FE EE E1 C6 67 EA
Oct 12 16:48:25 sonoff-1255 ESP-DMP: 55 5A 02 EE C8 00 03 11 00 3F 2F 03 31 6A 50 CB 98 67 99 52 E1 C6 67 EB
Oct 12 16:48:26 sonoff-1255 ESP-CMD: SysLog 2
Oct 12 16:48:26 sonoff-1255 ESP-SRC: WebConsole from 192.168.0.150
Oct 12 16:48:26 sonoff-1255 ESP-RSL: Received Topic /SysLog, Data Size 1, Data 2
Oct 12 16:48:26 sonoff-1255 ESP-RSL: Group 0, Index 1, Command SYSLOG, Data 2
Oct 12 16:48:26 sonoff-1255 ESP-MQT: stat/sonoff-1255/RESULT = {"SysLog":"2 (Active 2)"}
```

celsworth, 2019-10-12:
So in that log I can see:

```
Oct 12 16:48:18 sonoff-1255 ESP-MQT: Subscribe to cmnd/sonoff-1255/#
```

But no other subscribe lines?

celsworth, 2019-10-12:
Is there any setting I might have changed that is preventing the subscription to the group topic? I've scoured the documentation looking for anything but come up blank..

arendst, 2019-10-12:
If your fulltopic does not contain %topic% it won't subscribe to grouptopic and fallbacktopic.

celsworth, 2019-10-12:
Ah! Thank you. I can put `%topic%` back in then, and just keep `%hostname%` too. Could be worth a little mention in the `FullTopic` documentation, as I thought I could just replace that with no ill-effects.

Confirmed this does fix it:

```
1570895947: Received SUBSCRIBE from DVES_6004E7
1570895947: 	cmnd/sonoff/sonoff-1255/# (QoS 0)
--
1570895947: Received SUBSCRIBE from DVES_6004E7
1570895947: 	cmnd/sonoffs/sonoff-1255/# (QoS 0)
--
1570895947: Received SUBSCRIBE from DVES_6004E7
1570895947: 	cmnd/DVES_6004E7_fb/# (QoS 0)
````

arendst, 2019-10-12:
You might want to try and change topic to ``%hostname%``

celsworth, 2019-10-12:
I tried this, but it looks like `%hostname%` is only parsed in FullTopic, but not Topic. Could this be added?

arendst, 2019-10-12:
It works with my version 6.6.0.18

celsworth, 2019-10-12:
Definitely in `Topic`, not `FullTopic`?

The only mention of `%hostname%` in the code I can find is https://github.com/arendst/Sonoff-Tasmota/blob/0f72a93170c974ea55760eefe9f8918beda80247/sonoff/sonoff.ino#L267 which is for `FullTopic`. And with 6.6.0 release here it's definitely not being replaced in `Topic`..

ascillato, 2019-10-12:
6.6.0 is old, please update to 6.6.0.18

celsworth, 2019-10-12:
Sorry, maybe I'm missing something, the latest I see on the releases page is 6.6.0 - would I need to build 6.6.0.18 myself?

celsworth, 2019-10-12:
Ah found it in http://thehackbox.org/tasmota/. Will give it a go.

celsworth, 2019-10-12:
No, sorry, I don't know if we're doing something different but it's not working in Topic even with 6.6.0.18:

```
Oct 12 17:32:54 192.168.0.244 %hostname%-1255 ESP-WIF: Connected
Oct 12 17:32:54 192.168.0.244 %hostname%-1255 ESP-DNS: Initialized
Oct 12 17:32:54 192.168.0.244 %hostname%-1255 ESP-HTP: Web server active on %hostname%-1255.local with IP address 192.168.0.244
Oct 12 17:32:56 192.168.0.244 %hostname%-1255 ESP-DNS: Query done. MQTT services found 0
Oct 12 17:32:56 192.168.0.244 %hostname%-1255 ESP-MQT: Attempting connection...
Oct 12 17:32:56 192.168.0.244 %hostname%-1255 ESP-MQT: Connected
Oct 12 17:32:56 192.168.0.244 %hostname%-1255 ESP-MQT: tele/%hostname%-1255/LWT = Online (retained)
Oct 12 17:32:57 192.168.0.244 %hostname%-1255 ESP-MQT: cmnd/%hostname%-1255/POWER =
Oct 12 17:32:57 192.168.0.244 %hostname%-1255 ESP-MQT: tele/%hostname%-1255/INFO1 = {"Module":"JH-G01B1","Version":"6.6.0.18(7b316f4-sonoff)","FallbackTopic":"cmnd/DVES_6004E7_fb/","GroupTopic":"sonoffs"}
Oct 12 17:32:57 192.168.0.244 %hostname%-1255 ESP-MQT: tele/%hostname%-1255/INFO2 = {"WebServerMode":"Admin","Hostname":"%hostname%-1255","IPAddress":"192.168.0.244"}
Oct 12 17:32:57 192.168.0.244 %hostname%-1255 ESP-MQT: tele/%hostname%-1255/INFO3 = {"RestartReason":"Software/System restart"}
Oct 12 17:32:58 192.168.0.244 %hostname%-1255 ESP-MQT: stat/%hostname%-1255/RESULT = {"POWER1":"ON"}
Oct 12 17:32:58 192.168.0.244 %hostname%-1255 ESP-MQT: stat/%hostname%-1255/POWER1 = ON (retained)
```

That's from syslog and you can see the `%hostname%` hasn't been translated.

[image]

celsworth, 2019-10-12:
I can't help wondering if I'm barking up the wrong tree here by trying to change my topics in the first place. How do people normally tell which socket an MQTT message has come from, if not by using the topic?

celsworth, 2019-10-12:
Okay, so after some fiddling I've arrived at a working solution with the configuration as follows:

Hostname is set to `sonoff-1255`. Topic is `sonoff-1255`. FullTopic is `%prefix%/%topic%/`.

With this setup, Tasmota subscribes to all the topics correctly (`cmnd/sonoff-1255/#`, `cmnd/sonoffs/#`, `cmnd/DVES_6004E7_fb/#`). Sorted :) It's just a bit of a shame I have to set the numbers manually but its the only way I can see to make it work.

A feature request here would be that Tasmota subscribes to the group topic if `%hostname%` is present in `FullTopic`, then I could leave all the other boxes at their defaults and I think it would all just work.

But anyway I've taken enough of your time, so thanks for your help in troubleshooting, much appreciated!