arendst/Tasmota #6628: (Sometimes) large delay when switching on/off over HTTP command

peterjuras, 2019-10-13:
> **GUIDE**
>
> This BUG issue template is meant to REPORT Tasmota software BUGS ONLY>
>  
> Please DO NOT OPEN AN ISSUE:
>  - If your Tasmota version is not the latest from the development branch, please update your device before submitting your issue. Your problem might already be solved. The latest precompiled binaries of Tasmota can be downloaded from http://thehackbox.org/tasmota/  
>  - If your issue is a flashing issue, please address it to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4)  
>  - If your issue is compilation problem, please address it to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4)  
>  - If your issue has been addressed before (i.e., duplicated issue), please ask in the original issue  
>  - If your issue is a Wi-Fi problem or MQTT problem, please try the steps provided in the [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) and troubleshooting wiki articles  
>
> Please take a few minutes to complete the requested information below. Our ability to provide assistance is greatly hampered without it. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help you.

### BUG DESCRIPTION

Sometimes web requests to power switches on or off are taking a very long time (10 - 15 seconds), where as other times they are resolving very quickly (within a few milliseconds). I can't figure out an exact reproduction or reason that leads to this behavior. It also appears that normal ping messages are not arriving during the time of a blocked request.

The commands I am sending have the following structure:

```
GET http://192.168.XX.XX/cm?cmnd=Power%20On&user=admin&password=my-password
GET http://192.168.XX.XX/cm?cmnd=Power%20Off&user=admin&password=my-password
```

I read a lot that the MQTT connection could be an issue, but **the MQTT connection is fully disabled for all my devices since I don't need them.**

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Sonoff-Tasmota/issues)
- [X] Searched the problem in the [wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): [Teckin Smart socket](https://www.amazon.de/gp/product/B07D77LFMJ/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1)
- [X] Tasmota binary firmware version number used: 6.6.0(release-basic)
  - [X] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [X] Flashing tools used: [tuya-convert](https://github.com/ct-Open-Source/tuya-convert)
- [X] Provide the output of command: ``Backlog Template; Module; GPIO``:
  ```
  Configuration output here:
07:09:57 CMD: Backlog Template; Module; GPIO
07:09:57 RSL: RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":1,"BASE":18}
07:09:57 RSL: RESULT = {"Module":"45 (BlitzWolf SHP)"}
07:09:58 RSL: RESULT = {"GPIO1":"0 (None)","GPIO3":"0 (None)"}
  
  ```
- [X] If using rules, provide the output of this command: ``Backlog Rule1; Rule2; Rule3``:
  ```
  Rules output here:
07:10:30 RSL: RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
07:10:30 RSL: RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
07:10:30 RSL: RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
  
  ```
- [X] Provide the output of this command: ``Status 0``:
  ```
  STATUS 0 output here:
07:10:51 RSL: STATUS = {"Status":{"Module":45,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
07:10:51 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff-basic.bin","RestartReason":"Power on","Uptime":"16T01:29:33","StartupUTC":"2019-09-27T04:41:18","Sleep":50,"CfgHolder":4617,"BootCount":8,"SaveCount":261,"SaveAddress":"F8000"}}
07:10:51 RSL: STATUS2 = {"StatusFWR":{"Version":"6.6.0(release-basic)","BuildDateTime":"2019-07-06T13:11:28","Boot":4,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
07:10:51 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Rutzheim",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008001","280500000100060000000000000000000000","00000000"]}}
07:10:51 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":422,"Free":580,"Heap":18,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144068","FlashMode":3,"Features":["00000809","0F082382","001083A0","20021706","010001C0"]}}
07:10:51 RSL: STATUS5 = {"StatusNET":{"Hostname":"sonoff-small-l","IPAddress":"192.168.178.77","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.1","Mac":"B4:E6:2D:42:7F:3F","Webserver":2,"WifiConfig":4}}
07:10:51 RSL: STATUS7 = {"StatusTIM":{"UTC":"Sun Oct 13 06:10:51 2019","Local":"Sun Oct 13 07:10:51 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"07:07","Sunset":"18:05"}}
07:10:51 RSL: STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
07:10:51 RSL: STATUS10 = {"StatusSNS":{"Time":"2019-10-13T07:10:51","ENERGY":{"TotalStartTime":"2019-08-11T16:01:40","Total":0.565,"Yesterday":0.000,"Today":0.000,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}}
07:10:51 RSL: STATUS11 = {"StatusSTS":{"Time":"2019-10-13T07:10:51","Uptime":"16T01:29:33","Vcc":3.182,"Heap":18,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":31,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Rutzheim","BSSId":"7C:FF:4D:8B:87:D2","Channel":1,"RSSI":36,"LinkCount":3,"Downtime":"0T00:01:28"}}}
  
  ```
- [X] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ ``weblog 4`` _for more debug information)_

**Note: Command was sent around 06:45:45!**

  ```
  Console output here:
06:44:27 WIF: Checking connection...
06:44:27 WIF: Connected
06:44:47 WIF: Checking connection...
06:44:47 WIF: Connected
06:45:07 WIF: Checking connection...
06:45:07 WIF: Connected
06:45:27 WIF: Checking connection...
06:45:27 WIF: Connected
06:45:47 WIF: Checking connection...
06:45:47 WIF: Connected
06:46:01 HTP: Command
06:46:01 SRC: WebCommand from 192.168.178.29
06:46:01 RSL: Received Topic /Power, Data Size 3, Data off
06:46:01 RSL: Group 0, Index 1, Command POWER, Data off
06:46:01 RSL: RESULT = {"POWER":"OFF"}
06:46:01 RSL: POWER = OFF
06:46:01 CFG: Saved to flash at FB, Count 258, Bytes 3584
06:46:07 WIF: Checking connection...
  ```

### TO REPRODUCE
_Steps to reproduce the behavior:_

It is not possible to clearly reproduce it every time, however it would be the following steps:
1. Wait a random amount of time
1. Switch device on or off (both are lagging sometimes) using a web request:
```
GET http://192.168.XX.XX/cm?cmnd=Power%20On&user=admin&password=my-password
GET http://192.168.XX.XX/cm?cmnd=Power%20Off&user=admin&password=my-password
```
1. Experience 10 - 15 second lag, then device turns on or off

Note: Usually after the lag appeared, it will work fine for at least the next 5 minutes and then might or might not lag again. I sometimes had another lag directly after a lagging command was executed however.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

It should never take more than 3 seconds. (Less than 1 second would be desirable - and is possible most of the time).

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

Let me know if you need more information.

**(Please, remember to close the issue when the problem has been addressed)**

stefanbode, 2019-10-13:
In the log you can see, that as soon as the command reaches the device it executes. For me it looks like the http initiate takes the time. Can you add, if you get the http 200 = ok immediately and then it takes 10 seconds or if you send the command it takes 10 seconds until you get on ok. Currently this looks like a network issue.

peterjuras, 2019-10-13:
When I start the request it takes the 10 - 15 seconds to get the 200 status response.

Jason2866, 2019-10-13:
It is a network issue. Have you already rebooted all your network devices and the router?

stefanbode, 2019-10-13:
I do see an rssi from 36. And a total downtime from more than a minute. I had also some problems with the far away devices until yesterday. I now use the STAGE core of esp. for me it looks much more stable right now. Please change the platform.ini and enable the stage core. Give it a try.

Jason2866, 2019-10-13:
@stefanbode  dont expect wonders from core pre 2.6 It solves some wifi bugs from earlier cores but it cant solve bad wifi issues

peterjuras, 2019-10-13:
Thanks, you were right it was rather related to wifi issues. I have 5 plugs and the same two were making problems. It turns out these were connected to a repeated instead of my router.

Now, being connected with the router directly everything appears to go more smoothly. Thank you for pointing me in the right direction, I'll close this issue now and will re-open only if I have issues again.

closed 2019-10-13 by peterjuras