arendst/Tasmota #6629: allow rules for unknown IR receive protocols

jacmet, 2019-10-13:
The rule handling for "on IrReceived#Data= is very nice, thanks. Unfortunately it doesn't work for signals with an unknown protocol because of the conditional in xdrv_05_irremote.ino:

```
      if (iridx) {
        XdrvRulesProcess();
```
E.G. Nerf lasertags "shoot" with codes like:

`{"IrReceived":{"Protocol":"UNKNOWN","Bits":18,"Data":"0x67228B44"}}
`

I understand why some use cases may want to filter away unknown protocol events (but this can largely be handled by setoption38). It would be great to have an option to also allow rules processing for the unknown events.

jacmet, 2019-10-13:
Or alternatively, perhaps document this restriction more clearly? It took some head scratching and digging into the code to understand why the rule wasn't triggering

Jason2866, 2019-10-13:
The info unknow means the code cant NOT decoded correctly. The code is just a part of the code.
Saying this it makes no sense for setuping a rule to this crippled code fragment
To receive the raw code you have to enable ir receive raw. This is described in wiki.
See commands section in wiki too

s-hadinger, 2019-10-13:
Please use sonoff-ir that has a wide range of decoded protocols including Laser tag. 

I will look why the rules don't fire. In the latest versions "Data" is replaced with "Hash" parameter. Even if the ir protocol can't be decoded, the hash should be semi-unique and able to recognize the same message.

ascillato2, 2019-10-13:
Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-10-13 by ascillato2

jacmet, 2019-10-13:
Well, unknown just means that it doesn't match any known pattern. Even with "unknown", if the data is the same for multiple presses of the same key (and different for other keys) I don't see any reason to not handle it. The unknown received codes ARE transmitted over MQTT, so ignoring them for the rules processing seems a bit inconsistent.

jacmet, 2019-10-13:
@s-hadinger the rules don't fire because of the conditional in xdrv_05_irremote.ino I quoted above

jacmet, 2019-10-13:
@s-hadinger sorry, what do you mean about sonoff-ir? As far as I can see, there isn't a tasmota binary variant named like that?

Jason2866, 2019-10-13:
You can find here http://thehackbox.org/tasmota/pre-2.6/

s-hadinger, 2019-10-13:
Sonoff-ir is a Tasmota variant that includes almost all protocols from IRremoteESP8266. See here: https://github.com/arendst/Sonoff-Tasmota/wiki/Tasmota-IR

I looked at sonoff-ir variant, and there is a bug: no rules are fired at all.

Let me fix this for both variants.

jacmet, 2019-10-13:
@s-hadinger thanks (for both), I will monitor the development branch and give it a try once your commit a fix

s-hadinger, 2019-10-13:
@jacmet Once Theo merges this PR, you can self-compile or wait a couple of hours for the pre-compiled binaries.

jacmet, 2019-10-18:
@s-hadinger verified with sonoff-ir.bin commit 3bb58b3. IR events are still listed as unknown, but a rule matching the hash value does trigger:
E.G. ON IrReceived#Hash=0x67228B44 DO POWER TOGGLE ENDON

tele/sonoff-ir/RESULT {"IrReceived" "Protocol":"UNKNOWN","Bits":18,"Hash":"0x67228B44","Repeat":0}}
cmnd/sonoff-ir/POWER TOGGLE

s-hadinger, 2019-10-18:
Thanks for the feedback. If you have IR raw recordings and you know about the encoding, you can submit the new IR protocol to https://github.com/crankyoldgit/IRremoteESP8266

When it is added to this lib, Tasmota will support it as well.

PR #6635 (2019-10-13): Fix Rules were not triggered with IR unknown protocol or in sonoff-ir
## Description:

Rules are now correctly triggered when IR unknown protocol is received.
Rules are now correctly triggered with "sonoff-ir" and "sonoff-ircustom"

**Related issue (if applicable):** fixes #6629 

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  - [x] The code change is tested and works on core 2.3.0, 2.4.2, 2.5.2, and pre-2.6
  - [x] The code change pass travis tests. **Your PR cannot be merge [... cut]