arendst/Tasmota #6636: Expose FanSpeed#State for IFAN03/IFAN02

SirGoodenough, 2019-10-13:
I have spent some time trying to re-create the 1 beep for low, 2 beeps for med 3 beeps for high that the IFAN02 provides (With Tasmota) but with the IFAN03.  My daughter is handicapped and she finds the beeping helps her quite a bit when controlling the fan remotely.  I have been trying to isolate the FanSpeed#State and trigger off of that, but that does not appear to be exposed.  The rule does not syntax error, however the trigger never happens so I assume it is not exposed.
Trying to control the buzzer triggering off the power commands is not easily doable either as the 3 relays come on at odd times to kick startup of the motor at speed changes.  I do not want to go out to an external program like Node Red or Home Assistant to tell the buzzer to go, but that may be where I am headed without FanSpeed#State available.
```
03:14:55 MQT: Jens_Fan/stat/RESULT = {"FanSpeed":2}
03:14:59 MQT: Jens_Fan/stat/RESULT = {"FanSpeed":3}
03:17:16 CMD: rule1
03:17:16 MQT: Jens_Fan/stat/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":472,"Rules":"on fanspeed#state=3 do buzzer 3,3 endon"}
```
And no trigger...

Jason2866, 2019-10-14:
You can try the provided firmware. Your rule should work with. Since i didnt have a sonoff ifan i couldnt test it. If it works give me a feedback, and i will do a PR to include it as standard in Tasmota.
[Sonoff_ifan_rule.zip](https://github.com/arendst/Sonoff-Tasmota/files/3723579/Sonoff_ifan_rule.zip)

arendst, 2019-10-14:
I'll see if I can do it in code. Makes it a lot simpler for the script kiddies ;0)

arendst, 2019-10-14:
Try the latest dev version. Precompiles are available within an hour.

arendst, 2019-10-14:
Oh BTW it is no longer needed to use a script; it will beep anyway.

meingraham, 2019-10-14:
So, this adds beeps to the FanSpeed setting; yes?

There is still no `FanSpeed#State` rule trigger; right?

arendst, 2019-10-14:
Yes, Right.

ascillato2, 2019-10-14:
Closing this issue as it has been added. Thanks 

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-10-14 by ascillato2

SirGoodenough, 2019-10-15:
Thanks very much for the quick response.
I really hate to be a whiner, but this didn't quite solve my problem.
At this point it looked like you replaced the single beep that happened previously with the remote with 1, 2, or 3 beeps on the 3 fan speed changes.  Yes, I did setoption67 to on and tried off as well.
As my Daughter cannot use the remote, and only interfaces with the fan at this point from the console or home Assistant, the beep doesn't happen there.  I checked from the console and from mqtt commands and the beep only happens when using the 433 remote.
A minor thing that we can be fine without, the ifan02 puts out a long beep with fanspeed 0, and in this iteration there is no beeps when the fan is set to fanspeed 0 from any method.
Let me know what you think, if exposing fanspeed#status is easier, I am fine scripting to it.  It may well be of use to others for other things as well.  I am not a programmer, so I don't know what this takes.
Thanks for your time.

arendst, 2019-10-15:
The latest commit solves the RC button beeps by providing a longer beep when speed 0 is pressed.

In addition al commands are now also fed through the rule processor solving your initial issue. Only change is that you should use ``#data`` instead of ``#state``.

So your rule should look like this:
```
on fanspeed#data=3 do buzzer 3,3 endon
```

SirGoodenough, 2019-10-16:
Thankyou!
I tested and all works perfect with setoption67 on for use with the button and with setoption67 off I tested with rules.

Here's a good rule for this in case someone else looks this up and wants some hints...
```
rule1 on fanspeed#data=0 do buzzer 1,5 endon on fanspeed#data=1 do buzzer 1,2 endon on fanspeed#data=2 do buzzer 2,2 endon on fanspeed#data=3 do buzzer 3,2 endon
```
Thanks again for your help and I sent you a paypal yesterday for your trouble.