arendst/Tasmota #6641: Cannot connect to Wifi on latest development

shantur, 2019-10-14:
> **GUIDE**
>
> This BUG issue template is meant to REPORT Tasmota software BUGS ONLY>
>  
> Please DO NOT OPEN AN ISSUE:
>  - If your Tasmota version is not the latest from the development branch, please update your device before submitting your issue. Your problem might already be solved. The latest precompiled binaries of Tasmota can be downloaded from http://thehackbox.org/tasmota/  
>  - If your issue is a flashing issue, please address it to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4)  
>  - If your issue is compilation problem, please address it to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4)  
>  - If your issue has been addressed before (i.e., duplicated issue), please ask in the original issue  
>  - If your issue is a Wi-Fi problem or MQTT problem, please try the steps provided in the [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) and troubleshooting wiki articles  
>
> Please take a few minutes to complete the requested information below. Our ability to provide assistance is greatly hampered without it. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help you.

### BUG DESCRIPTION
Cannot connect to any Wifi AP on latest development branch
It seems like some issue with IP after IPV6 implementation.

This shows up in serial log

```00:08:01 WIF: Connecting to AP1 SKY91125 in mode 11N as sonoff-6429...
00:08:09 WIF: Connect failed as no IP address received
00:08:09 WIF: Connecting to AP1 SKY91125 in mode 11N as sonoff-6429...
00:08:17 WIF: Connect failed as no IP address received```

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the [wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): NodeMcu
- [x] Tasmota binary firmware version number used: Latest Development
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: `esptool`
- [ ] Provide the output of command: ``Backlog Template; Module; GPIO``:
  ```
  Configuration output here:
  
00:12:55 RSL: stat/sonoff/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
00:12:56 RSL: stat/sonoff/RESULT = {"Module":{"1":"Sonoff Basic"}}
00:12:56 RSL: stat/sonoff/RESULT = {"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO14":{"0":"None"}}

  
  ```
- [ ] If using rules, provide the output of this command: ``Backlog Rule1; Rule2; Rule3``:
  ```
  Rules output here:
  
  
  ```
- [ ] Provide the output of this command: ``Status 0``:
  ```
  STATUS 0 output here:
  
tonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
00:12:33 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:12:33","StartupUTC":"","Sleep":50,"CfgHolder":4617,"BootCount":2,"SaveCount":5,"SaveAddress":"F7000"}}
00:12:33 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.6.0.18(sonoff)","BuildDateTime":"2019-10-14T10:56:26","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)"}}
00:12:33 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["SKY91125","AndroidAP"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000"]}}
00:12:33 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":562,"Free":440,"Heap":31,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00000081"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
00:12:33 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-6429","IPAddress":"(IP unset)","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"2C:F4:32:19:79:1D","Webserver":2,"WifiConfig":4}}
00:12:33 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_19791D","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
00:12:33 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 01 00:12:33 1970","Local":"Thu Jan 01 00:12:33 1970","StartDST":"Thu Jan 01 00:00:00 1970","EndDST":"Thu Jan 01 00:00:00 1970","Timezone":"+00:00","Sunrise":"07:43","Sunset":"16:03"}}
00:12:33 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:12:33"}}
00:12:33 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:12:33","Uptime":"0T00:12:33","UptimeSec":753,"Heap":31,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":30,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"SKY91125","BSSId":"A0:BD:CD:5E:19:1A","Channel":4,"RSSI":100,"LinkCount":0,"Downtime":"0T00:00:00"}}}
  
  ```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ ``weblog 4`` _for more debug information)_
  ```
  Console output here:
  

00:08:01 WIF: Connecting to AP1 SKY91125 in mode 11N as sonoff-6429...
00:08:09 WIF: Connect failed as no IP address received
00:08:09 WIF: Connecting to AP1 SKY91125 in mode 11N as sonoff-6429...
00:08:17 WIF: Connect failed as no IP address received  

  ```

### TO REPRODUCE
- Install latest development

### EXPECTED BEHAVIOUR

Wifi connects.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2019-10-14:
Pls retry as I just fixed the IPV6 compile issue.

shantur, 2019-10-14:
Thanks its fixed

closed 2019-10-14 by shantur