arendst/Tasmota #6642: CH340 not visable anymore

agoralive, 2019-10-14:
Hi

Not the right forum I KNOW!!!!

But after I updated my windows, it no longer is able to find my NodeMCU, ESP8266, WeMos

Drivers have been installed again, but it does not do anything besides power my ESP's?

Anybody have the same issue, or at workaround?

localhost61, 2019-10-14:
-> [Chat on Discord](https://discordapp.com/invite/Ks2Kzd4)

effelle, 2019-10-14:
Please close this issue and address this to the [Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are best used for **Tasmota software feature requests and bug reporting**. Troubleshooting and setup assistance is more effective using an interactive forum.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-10-14 by agoralive