arendst/Tasmota #6643: Add new section to scripts that is run on button change

BCNelson, 2019-10-14:
**Have you looked for this feature in other issues and in the wiki?**  
Yes

**Is your feature request related to a problem? Please describe.** 
No

**Describe the solution you'd like**  
I would like to add another section to the scripts that is run anytime that FUNC_BUTTON_PRESSED event is fired by XdrvCall()
also I think that two new variables need to be added namely button and button state that are valid only in the button section.

**Describe alternatives you've considered**  
Checking the buttons in the >F section that is executed every 100ms while this is serviceable it is slow and takes more room in the script storage.

**Additional context**  
_Add any other context or screenshots about the feature request here._

BCNelson, 2019-10-14:
Mostly I'm looking to see if this would be acceptable before I go to the work of making it. As well as discussing the specifics of the addition

gemu2015, 2019-10-15:
this is already working.
the >E section receives all events like rules does, but in contrast to rules JSON strings are case sensitive.

for buttons use:

>D
b1=0

>E
b1=Button1#State

if upd[b1]>0
then
print button 1 pressed
endif

gemu2015, 2019-10-15:
ok, added section >b
is executed on any state change of button

#define USE_BUTTON_EVENT

bt[x]   x=1-MAX_KEYS
\>b
print button event =>%bt[1]%

gemu2015, 2019-10-15:
@BCNelson 

https://github.com/gemu2015/Sonoff-Tasmota/blob/universal7/sonoff/xdrv_10_scripter.ino

will do a pr soon

PR #6656 (2019-10-16): scripter button events, remove comments on paste
## Description:

**Related issue (if applicable):** fixes #6643 >
section >b catches button change event   bt[x] (x=1.. MAX_KEYS) gets button state
#define USE_BUTTON_EVENT

**Related issue (if applicable):** fixes #6621>
initial support for stripping comments on pastes larger then max length

support for \n \r on commands text

small fixes

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  - [x] The code change is tested and work [... cut]
closed 2019-10-16 by arendst