arendst/Tasmota #6645: OTA Upgrade from  6.4.01/2.3.0/1.5.3 to 6.6.0.18 2_5_2/2.2.2-dev(c0eb301) fails

lobocobra, 2019-10-15:
**Bug description**
I have around 50 Shelly1 & 2 in use in the house. 
=> Shelly1 are on V.6.4.0.1 2_3_0/1.5.3
=> Shelly2 are on V6.5.0.15 2_5_2/2.2.2-dev(c0eb301) 

As I wanted to roll-out on all shelly the same firmware and shelly2 obliges to use a dev version, I planned to upgrade to: 6.6.0 / 2.5.2 from here : http://thehackbox.org/tasmota/release/020502/
1) first I wanted to upload sonoff-minimal.bin
2) then sonoff.bin to avoid problems with too big code

Step 1 succeeded on the guy and it gave a success message, but the Shelly 1 and 2 became unreachable (but still pingable). 
=> I tried it on 5 different Shelly and ended being forced to uninstall from the wall, in order to be able to reload the firmware by serial. 

**I was able to revitalize them by...**
1) esptool -cp com10 -ce
2) reload firmware 6.6.0 /2.5.2 (compiled by myself)

- [ ] Device used (e.g., Sonoff Basic): shelly 1 & shelly 2
- [ ] Tasmota binary firmware version number used: see above
  - [x ] Pre-compiled 
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: VSC
- [ ] Flashing tools used: esptool
- [ ] Provide the output of command: ``Backlog Template; Module; GPIO``:
'04:32:52 MQT: stat/y_ex_Schocklicht_s2/RESULT = {"NAME":"Shelly 2.5","GPIO":[56,255,17,255,21,83,0,0,6,82,5,22,156],"FLAG":2,"BASE":18}
04:32:52 MQT: stat/y_ex_Schocklicht_s2/RESULT = {"Module":"0 (Shelly 2.5)"}
04:32:52 MQT: stat/y_ex_Schocklicht_s2/RESULT = {"GPIO1":"0 (None)","GPIO3":"0 (None)"}
  Configuration output here:'
  
  

  ```
- [ ] Provide the output of this command: ``Status 0``:
`04:34:59 MQT: stat/y_ex_Schocklicht_s2/STATUS = {"Status":{"Module":0,"FriendlyName":["y_ex_Schocklicht_licht","y_ex_Schocklicht_sonos"],"Topic":"y_ex_Schocklicht_s2","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[1,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
04:34:59 MQT: stat/y_ex_Schocklicht_s2/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"1T06:38:46","StartupUTC":"2019-10-13T19:56:13","Sleep":50,"CfgHolder":4617,"BootCount":14,"SaveCount":1358,"SaveAddress":"F6000"}}
04:34:59 MQT: stat/y_ex_Schocklicht_s2/STATUS2 = {"StatusFWR":{"Version":"6.5.0.15(sonoff)","BuildDateTime":"2019-06-10T21:47:50","Boot":31,"Core":"2_5_2","SDK":"2.2.2-dev(c0eb301)"}}
04:34:59 MQT: stat/y_ex_Schocklicht_s2/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["domaincats",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000100060000000000000000000000","000000C1"]}}
04:34:59 MQT: stat/y_ex_Schocklicht_s2/STATUS4 = {"StatusMEM":{"ProgramSize":572,"Free":428,"Heap":23,"ProgramFlashSize":2048,"FlashSize":2048,"FlashChipId":"1540EF","FlashMode":3,"Features":["00000809","0FDAE394","041783A1","23B617CC","01003BC0"]}}
04:34:59 MQT: stat/y_ex_Schocklicht_s2/STATUS5 = {"StatusNET":{"Hostname":"y_ex_Schocklicht_s2-0324","IPAddress":"192.168.20.37","Gateway":"192.168.20.2","Subnetmask":"255.255.255.0","DNSServer":"192.168.20.2","Mac":"24:62:AB:00:81:44","Webserver":2,"WifiConfig":5}}
04:34:59 MQT: stat/y_ex_Schocklicht_s2/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.20.90","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_008144","MqttUser":"DVES_USER","MqttCount":376,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
04:34:59 MQT: stat/y_ex_Schocklicht_s2/STATUS7 = {"StatusTIM":{"UTC":"Tue Oct 15 02:34:59 2019","Local":"Tue Oct 15 04:34:59 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":99,"Sunrise":"07:44","Sunset":"18:39"}}
04:34:59 MQT: stat/y_ex_Schocklicht_s2/STATUS9 = {"StatusPTH":{"PowerDelta":0,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
04:34:59 MQT: stat/y_ex_Schocklicht_s2/STATUS10 = {"StatusSNS":{"Time":"2019-10-15T04:34:59","Switch1":"OFF","Switch2":"OFF","ANALOG":{"Temperature":53.2},"ENERGY":{"TotalStartTime":"2019-06-10T20:49:40","Total":13.633,"Yesterday":0.043,"Today":0.000,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000},"TempUnit":"C"}}
04:34:59 MQT: stat/y_ex_Schocklicht_s2/STATUS11 = {"StatusSTS":{"Time":"2019-10-15T04:34:59","Uptime":"1T06:38:46","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"domaincats","BSSId":"AC:22:05:85:AD:D3","Channel":11,"RSSI":72,"LinkCount":375,"Downtime":"0T00:13:00"}}}`
  

### TO REPRODUCE
try to upgrade a 6.4+ version to latest 6.6.

Jason2866, 2019-10-15:
Core 2.5.2 has a big issue. The memory manager is not stored in iRAM. This can result in problems.
See https://github.com/esp8266/Arduino/pull/6161
Dont use Arduino cores 2.4.x and 2.5.x. Too many unsolved issues (bugs)
Still working without major bugs is core 2.3.0. But this core has security issues
**All known (major) issues from core 2.5.2 are fixed in core pre 2.6.**

**A general point of failure, which can occur on every device, when upgrading**:
Fragments left in space in areas Tasmota actually do not use. This fragments are from the original factory firmware. Newer Tasmota versions uses now more space for settings in flash (unused before).
This settings have random values. This can result in not starting  (correct) Tasmota after reboot when new firmware is first time active.

**To do a safe upgrade:**
- Backup your settings 
- Do `reset 5`  all settings will be lost! (except wifi credentials)
- Restore your settings

**This is only neccessary for device where the flash has never been erased ONE time fully ever**

ascillato2, 2019-10-15:
Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-10-15 by ascillato2

lobocobra, 2019-10-15:
Thanks for the the detailed answer. I downgraded core to 2.3 and did a reset 5 before the firmware upgrade and it worked.