arendst/Tasmota #6647: Sensor DS18x20 is selectable, even not in compiled firmware

Jason2866, 2019-10-15:
### BUG DESCRIPTION

Choice is seen in GPIO config page even when all DS18xx drivers are **not** compiled in firmware.

[image]

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the [wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): NodeMCU
- [x] Tasmota binary firmware version number used: dev 6.6.0.18 latest
  - [ ] Pre-compiled
  - [X] Self-compiled
    - [ ] IDE / Compiler used: VSC
- [x] Flashing tools used: Esptool.py
- [x] Provide the output of command: ``Backlog Template; Module; GPIO``:

```
11:14:14 CMD: Backlog Template; Module; GPIO
11:14:14 MQT: stat/sonoff-6111ED/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
11:14:15 MQT: stat/sonoff-6111ED/RESULT = {"Module":{"18":"Generic"}}
11:14:15 MQT: stat/sonoff-6111ED/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"166":"Zigbee Rx"},"GPIO14":{"0":"None"},"GPIO15":{"165":"Zigbee Tx"},"GPIO16":{"0":"None"}}

```
```
11:15:07 CMD: status 0
11:15:07 MQT: stat/sonoff-6111ED/STATUS = {"Status":{"Module":18,"FriendlyName":["Zigbee"],"Topic":"sonoff-6111ED","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
11:15:07 MQT: stat/sonoff-6111ED/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://192.168.2.20/data/firmwares/sonoff-minimal.bin","RestartReason":"Software/System restart","Uptime":"0T00:08:27","StartupUTC":"2019-10-15T09:06:40","Sleep":50,"CfgHolder":4617,"BootCount":18,"SaveCount":36,"SaveAddress":"FA000"}}
11:15:07 MQT: stat/sonoff-6111ED/STATUS2 = {"StatusFWR":{"Version":"6.6.0.18(sonoff)","BuildDateTime":"2019-10-15T11:05:19","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)"}}
11:15:07 MQT: stat/sonoff-6111ED/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Jason_Home_WLAN",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A64000000000000","00000000"]}}
11:15:07 MQT: stat/sonoff-6111ED/STATUS4 = {"StatusMEM":{"ProgramSize":418,"Free":584,"Heap":26,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"16400E","FlashMode":3,"Features":["00000809","08800180","044003A1","00000012","000000C0","00000000"],"Drivers":"1,2,8,10,23","Sensors":"5"}}
11:15:07 MQT: stat/sonoff-6111ED/STATUS5 = {"StatusNET":{"Hostname":"sonoff-6111ED-4589","IPAddress":"192.168.2.163","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.1","Mac":"DC:4F:22:61:11:ED","Webserver":2,"WifiConfig":5}}
11:15:07 MQT: stat/sonoff-6111ED/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.114","MqttPort":1883,"MqttClientMask":"sonoff-%06X","MqttClient":"sonoff-6111ED","MqttUser":"openhabian","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
11:15:07 MQT: stat/sonoff-6111ED/STATUS7 = {"StatusTIM":{"UTC":"Tue Oct 15 09:15:07 2019","Local":"Tue Oct 15 11:15:07 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":99}}
11:15:07 MQT: stat/sonoff-6111ED/STATUS10 = {"StatusSNS":{"Time":"2019-10-15T11:15:07"}}
11:15:07 MQT: stat/sonoff-6111ED/STATUS11 = {"StatusSTS":{"Time":"2019-10-15T11:15:07","Uptime":"0T00:08:27","UptimeSec":507,"Vcc":3.078,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","BSSId":"00:A0:57:2A:BD:19","Channel":13,"RSSI":92,"LinkCount":1,"Downtime":"0T00:00:05"}}}

```
### TO REPRODUCE
Compile firmware without DS18xx driver

### EXPECTED BEHAVIOUR
Choice should not appear
[image]

Jason2866, 2019-10-15:
Uncomment in sonoff_template.h did solve
```
#if defined(USE_DS18B20) || defined(USE_DS18x20)
//  GPIO_DSB,            // Single wire DS18B20 or DS18S20
#endif
```
So it seems the if does not work

arendst, 2019-10-15:
This is as designed.

When you do not select define USE_DS18x20 it will always install the single DS18B20 driver as configured in sonoff_post.h

Jason2866, 2019-10-15:
I feel free to ask, why?

arendst, 2019-10-15:
legacy, Legacy, LEGACY!

The DS18B20 and DHT sensors were added when the Sonoff TH was introduced. This device had default support for these sensors from itead.

I agree, we are now two years ahead and might conclude to abandon the single device driver in favour of the one and only multi device DS18X20 driver.

closed 2019-10-16 by Jason2866