arendst/Tasmota #6651: Shutter position problem

vergoossen, 2019-10-15:
I have curtain motors which can start by pulling the curtain.
For example, when they're closed, i can pull them to "start" the motor to open, but tasmota doesnt recognize that and then has problems with the position.

Is this a known "bug"?

ascillato2, 2019-10-15:
Please, could you be so kind on completing the [troubleshooting template](https://github.com/arendst/Sonoff-Tasmota/issues/new/choose) in order to have more information so as to properly help you?

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

vergoossen, 2019-10-15:
Im sorry, i hope this is enough information:

Device: shelly 2.5
Firmware version: 6.6.0.18
Self compiled from development branche with USE_SHUTTER enabled:
#ifndef USE_SHUTTER
#define USE_SHUTTER      //  (+6k code)
#endif
Flashed with esphomeflasher.
Backlog template: {"NAME":"Shelly 2.5","GPIO":[56,255,17,255,21,83,0,0,6,82,5,22,156],"FLAG":2,"BASE":18}
Backlog module: {"Module":{"0":"Shelly 2.5"}}
Backlog gpio: {"GPIO1":{"0":"None"},"GPIO3":{"0":"None"}}
Backlog status
`13:55:25 RSL: stat/sonoff/STATUS = {"Status":{"Module":0,"FriendlyName":["Sonoff","Sonoff2"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[3,3,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
13:55:25 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"External System","Uptime":"0T00:54:01","StartupUTC":"2019-10-15T12:01:24","Sleep":50,"CfgHolder":4617,"BootCount":9,"SaveCount":155,"SaveAddress":"F9000"}}
13:55:25 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.6.0.18(sonoff)","BuildDateTime":"2019-10-15T10:57:32","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)"}}
13:55:25 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["WATERPAS",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000C009","2805C8000100060000005A00000000000000","40000000"]}}
13:55:25 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":567,"Free":436,"Heap":27,"ProgramFlashSize":2048,"FlashSize":2048,"FlashChipId":"1540EF","FlashMode":3,"Features":["00000809","8FDAE397","043683A1","22B617CD","01001BC0","00001181"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
13:55:25 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-5200","IPAddress":"192.168.1.111","Gateway":"192.168.1.2","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.2","Mac":"2C:F4:32:68:D4:50","Webserver":2,"WifiConfig":4}}
13:55:25 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_68D450","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
13:55:25 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Tue Oct 15 12:55:25 2019","Local":"Tue Oct 15 13:55:25 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"07:10","Sunset":"18:01"}}
13:55:25 RSL: stat/sonoff/STATUS9 = {"StatusPTH":{"PowerDelta":0,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
13:55:25 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-10-15T13:55:25","Switch1":"OFF","Switch2":"OFF","ANALOG":{"Temperature":49.2},"ENERGY":{"TotalStartTime":"2019-10-15T12:21:55","Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":[0,0],"ApparentPower":[0,0],"ReactivePower":[0,0],"Factor":[0.00,0.00],"Voltage":0,"Current":[0.000,0.000]},"Shutter1":{"Position":0,"direction":0},"TempUnit":"C"}}
13:55:25 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-10-15T13:55:25","Uptime":"0T00:54:01","UptimeSec":3241,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"WATERPAS","BSSId":"2C:30:33:DE:7E:CF","Channel":13,"RSSI":54,"LinkCount":1,"Downtime":"0T00:00:09"}}}`

ascillato, 2019-10-15:
Sorry, but we need more information, you have skipped the template.

Anyway, how your shelly can detect your manual movement of your cortain? Your cortain has others microswitches? If so, those must be connected to inform this movement to the shelly.

You should provide photos of your setup. We can't guess how is everything connected in your case

ascillato, 2019-10-15:
Checking on the official Shelly 2.5 site (https://shelly.cloud/shelly-25-wifi-smart-relay-roller-shutter-home-automation/) this device don't have any extra terminals for more switches and don't have any electronics to detect what you want to achieve (as far as the article explains)

Have you tested your setup before installing Tasmota?

ascillato2, 2019-10-15:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. 

Thanks.

---------------------------

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

### Support Information 
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-10-15 by ascillato2

joba-1, 2019-10-15:
> how your shelly can detect your manual movement of your cortain?

I don't know, but I would measure power coming from the motors due to forced movement.
Shellies can measure overcurrent, so maybe also that? Or the curtain does that itself and closes a switch? Anyways, nice feature I would like if I had electric curtains.