arendst/Tasmota #6653: Device Recovery after Application of Incompatible Template

GaryMilne, 2019-10-15:
> Please take a few minutes to complete the requested information below. Our ability to provide assistance is greatly hampered without it. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help you.

**Have you looked for this feature in other issues and in the wiki?**  
Yes and I'm surprised this issue is not more widespread. Out of 9 devices (mostly bulbs) I have 4 that exhibited this problem.

**Is your feature request related to a problem? Please describe.**  
The problem is simply that after the application of certain templates the device becomes unstable\unavailable. A constant ping will show that perhaps 85% of the packets do not get through (see below) and the web interface is only available very intermittently (rarely long enough to do much).  To recover the device I wait until I see some packets go through on the ping and I force a reset to defaults using Tasmota Device Manager. Then I start over again.

**Describe the solution you'd like**  
I'd like to be able to revert to a "Last Known Good\Alternate Config" type of scenario.
A few options here. A) Set a time interval to confirm the correct operation of the new Template.  If that time period is exceeded without console confirmation (a commit type statement) then the device would revert to the prior device Template. 

B) Utilize the method used by many manufacturers whereby multiple on/off events in a short period is an instruction to the bulb. What I'm suggesting would be reverting to "Last Known Good\Backup config" after X on/off events.

**Describe alternatives you've considered**  
This could probably be done with triggers now that I think about it but it would obviously be easier if it were built in. 

**Additional context**  
Not really other than to say I really like the Tasmota firmware but the configuration of devices really creates a time drain when you have to reset them from scratch multiple times. 

**(Please, remember to close the issue when the problem has been addressed)**  
[image]

ascillato, 2019-10-15:
Tasmota has a bootloop protection routine. if you energize your device and de-energize it in less than 2 seconds and you do that several times in a row, Tasmota will revert to some of the default configuration (rules enabled, etc etc.)

There is at this moment a proposed PR #6639 to reset Tasmota to all default values for a case like yours. You can ask there

andrethomas, 2019-10-15:
What happens if you do

`ping x.x.x.x -w 10000`

GaryMilne, 2019-10-16:
Still times out, just takes longer. It sounds as though what I thought was
me doing a reset via Tasmota Device manager was quite likely just the
normal recovery process after a boot loop.

ascillato2, 2019-10-16:
@GaryMilne We have found that some precompiled bins had an issue. Now it is fixed.

Please, try to update to latest version. If you have access by mqtt you can run reset 6 command to erase all setting but keeping WIFI and MQTT credentials, and/or do upgrades.

If you need assistance on that, please address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you.

About your request, we can continue the thread in the already open PR #6639

Thanks.

---------------------------

### Support Information 
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-10-16 by ascillato2

GaryMilne, 2019-10-16:
> FYI That I went back to version 6.1.1 and I am able to switch configs
> without losing access and getting into a boot loop.
>
> This is a livable solution for me so you can close the case.

One last thing. I was able to go to 6.5.0 and switch templates easily
without the bulbs freezing and going into a reboot cycle.