arendst/Tasmota #6665: Sonoff UK Binary is Russian

somebodywhoisntme, 2019-10-16:
Hello,

I only compiled the UK-Binary but after uploading it it was russian or something like that.

Now I downloaded the UK Version from this site and its also in russian and not english.

Am I to stupid or whats the problem here?

arendst, 2019-10-16:
Yes you are to stupid as this issue has come along before (grin) ;-)

How about UK is NOT United Kingdom but perhaps Ukrain...

arendst, 2019-10-16:
I can hear your brains crunching. Where is the United Kingdom version....

Try the default ones. Chances are they are using English.

somebodywhoisntme, 2019-10-16:
It could be hardly otherwise :D I already flashed the right binary. 

Really thanks for your fast answer and much more for the hard work you put in here.

closed 2019-10-16 by ascillato2