arendst/Tasmota #6666: Security Risk - wifi will connect without password (when configured with a password)

twinotter, 2019-10-16:
### BUG DESCRIPTION

Okay, I found something that helps me in my particular case, but is also a potential security risk.

If Tasmota is set up with a SSID and password, the unit will connect to an SSID that needs no password. This means you can hijack the unit by broadcasting from your phone the SSID that you think it's connected to. (I just did it on Android, turned on hotspot, no password.) Get the unit to reconnect and it will happily connect to your phone. You can then access its web page as if you were on its subnet.

It was incredibly useful when my unit got bad credentials. Reprogramming an Electek ESW-01 is a pain in the butt if you have to reopen its case.

### REQUESTED INFORMATION
- [X ] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X ] Searched the problem in [issues](https://github.com/arendst/Sonoff-Tasmota/issues)
- [X ] Searched the problem in the [wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X ] Device used (e.g., Sonoff Basic): ElecTek ESW-01
- [X ] Tasmota binary firmware version number used: 6.6.0
  - [ X] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [X ] Flashing tools used: NodeMCU
- [ ] Provide the output of command: ``Backlog Template; Module; GPIO``:
  ```
  Configuration output here:
  
00:00:00 CFG: Loaded from flash at F7, Count 85
00:00:00 Project sonoff Chicken Humidity Version 6.6.0(release-sonoff)-2_3_0
00:00:00 WIF: Connecting to AP1 rabbitcage in mode 11N as chickenhumidity...
00:00:07 WIF: Connect failed as AP cannot be reached
00:00:07 WIF: Connecting to AP2 dragontransit in mode 11N as chickenhumidity...
00:00:14 WIF: Connect failed as AP cannot be reached
00:00:15 WIF: Connect failed as AP cannot be reached
00:00:15 WIF: Connecting to AP1 rabbitcage in mode 11N as chickenhumidity...
00:00:22 WIF: Connect failed as AP cannot be reached
00:00:23 WIF: Connect failed as AP cannot be reached
00:00:23 WIF: Connecting to AP2 dragontransit in mode 11N as chickenhumidity...
00:00:30 WIF: Connect failed as AP cannot be reached
00:00:31 WIF: Connect failed as AP cannot be reached
00:00:31 WIF: Connecting to AP1 rabbitcage in mode 11N as chickenhumidity...
00:00:41 WIF: Connected
00:00:41 HTP: Web server active on chickenhumidity with IP address 192.168.1.31
00:00:42 MQT: Attempting connection...
00:00:44 MQT: Connected
00:00:44 MQT: tele/chickenhumidity/LWT = Online (retained)
00:00:44 MQT: cmnd/chickenhumidity/POWER = 
00:00:44 MQT: tele/chickenhumidity/INFO1 = {"Module":"ESW01-US Humid","Version":"6.6.0(release-sonoff)","FallbackTopic":"cmnd/DVES_FB0D7E_fb/","GroupTopic":"sonoffs"}
00:00:44 MQT: tele/chickenhumidity/INFO2 = {"WebServerMode":"Admin","Hostname":"chickenhumidity","IPAddress":"192.168.1.31"}
00:00:44 MQT: tele/chickenhumidity/INFO3 = {"RestartReason":"External System"}
00:00:44 MQT: stat/chickenhumidity/RESULT = {"POWER":"ON"}
00:00:44 MQT: stat/chickenhumidity/POWER = ON
18:32:10 MQT: tele/chickenhumidity/STATE = {"Time":"2019-10-16T18:32:10","Uptime":"0T00:00:51","Heap":16,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"rabbitcage","BSSId":"XX:XX:XX:XX:XX:XX","Channel":11,"RSSI":66,"LinkCount":1,"Downtime":"0T00:00:41"}}
18:32:10 MQT: tele/chickenhumidity/SENSOR = {"Time":"2019-10-16T18:32:10","ENERGY":{"TotalStartTime":"2019-10-15T19:24:00","Total":0.204,"Yesterday":0.102,"Today":0.102,"Period":0,"Power":1,"ApparentPower":744,"ReactivePower":744,"Factor":0.00,"Voltage":142,"Current":5.253},"SHT3X-0x44":{"Temperature":70.4,"Humidity":53.0},"TempUnit":"F"}
18:34:16 CMD: Backlog Template; Module; GPIO
18:34:16 MQT: stat/chickenhumidity/RESULT = {"NAME":"ESW01-US Humid","GPIO":[0,6,0,5,21,52,0,0,132,133,17,0,130],"FLAG":0,"BASE":6}
18:34:16 MQT: stat/chickenhumidity/RESULT = {"Module":"0 (ESW01-US Humid)"}
18:34:17 MQT: stat/chickenhumidity/RESULT = {"GPIO":"Not supported"}

  
  ```
- [ ] If using rules, provide the output of this command: ``Backlog Rule1; Rule2; Rule3``:
  ```
  Rules output here:
  
  
  ```
- [ ] Provide the output of this command: ``Status 0``:
  ```
  STATUS 0 output here:
  
  
  ```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ ``weblog 4`` _for more debug information)_
  ```
  Console output here:
  
  
  ```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

shantur, 2019-10-16:
I just did a test myself. I turned off my wifi router and created an open hotspot with same ssid and only Tasmota on core versions 2.3.0 connected to the hotspot but I had a nodeMcu with Stage core didn't connect. So my assumption is this bug is present in core version 2.3.0 but fixed  in or before 2.6

ascillato, 2019-10-16:
Tested on 2.4.2 and this Security risk is there too!!!!

Core Stage (or pre2.6.0) don't have this security risk

@twinotter 

Great finding !!!!!! Really Nasty Bug.

It is Core Related. Not a Tasmota issue.

So far, any Arduino Sketch that uses core 2.3.0 or 2.4.2 exhibits this security risk. Using core pre2.6.0 or STAGE this risk is not there.

Jason2866, 2019-10-16:
@TimelessNL  does not matter if core 2.5.2 has the bug or not. Dont use it.
There is no reason for using 2.5.2 It generates bigger code and has less iram and unfixed bugs.
All the drawbacks of 2.5.2 are solved with pre 2.6
Tassmota has dropped support for core 2.5.2 today

ascillato2, 2019-10-16:
Yes,

retested and core 2.3.0 and 2.4.2 only exhibits this issue.

the released 2.5.2 from arduino don't have this bug

ascillato2, 2019-10-16:
Summary of problems of arduino cores we know so far:

* **2.3.0**

Slow
High RAM usage
**No software serial support**
Sleep feature works
**KRACK security issue**
**blank password security issue(#6666) and others (#6348).**
Alexa support works

* **2.4.2** 

Faster than 2.3.0
More free RAM than 2.3.0
Software serial support
**Sleep feature don't work.**
KRACK solved
**blank password security issue(#6666) and others (#6348).**
**Alexa support don't works without a patch.** (patch included automatically on Tasmota)

* **2.5.0 and 2.5.1**

Faster than 2.3.0
More free RAM than 2.3.0
Software serial support
**Sleep feature don't work**
KRACK solved
blank password security issue solved
**other security issues not solved(#6348)**
**Memory manager not in IRAM producing weird issues like reboots.**
**Alexa support don't works without a patch.** (patch included automatically on Tasmota)

* **2.5.2**

Faster than 2.3.0
More free RAM than 2.3.0
Software serial support
Sleep feature work
KRACK solved
blank password security issue solves
**other security issues not solved(#6348)**
**Memory manager not in IRAM producing weird issues like reboots.**
Alexa support works.

* **pre2.6.0 and STAGE**

Faster than 2.3.0
More free RAM than 2.3.0
Software serial support
Sleep feature work
KRACK solved
blank password security issue solved and others solved(#6348)
Memory manager solved.
Alexa support works.

So,

We should drop all cores and only use pre2.6.0 (at least for precompiled bins and release versions)

@arendst @andrethomas

What about making a new release due this nasty security risk? and offering precompiled bins only with pre2.6.0?

qingz2004, 2019-10-16:
I don't see anything wrong here.
If web password is set in this device, nobody can get access to it remotely without the password even you fooled it to connect to your LAN.
If you have physical access to this device, well, nobody can stop you from stealing it.
You can gain access to any WiFi router by factory resetting it if you have physical access to it.

ascillato, 2019-10-16:
@qingz2004 

Please test the following:

Put a SSID and PASSWORD in your Tasmota
make a hotspot with your phone with an open wifi network called as your SSID but with your passwors blank.
Your Tasmota device will connect to it if it has core 2.3.0 or 2.4.2 and you can control it, change the config and/or put the firmware you want. That is a full control of the device. That is a security risk that is solved in core pre2.6.0

qingz2004, 2019-10-16:
@ascillato How can I control it if I don't have the login password?

qingz2004, 2019-10-16:
You will get this.
[image]

ascillato, 2019-10-16:
@qingz2004 yes, but not everyone uses the web password. So, for most users at this moment, this is a security risk and the firmware must be updated to latest with core 2.6.0

meingraham, 2019-10-16:
This is the web UI authentication, not the Wi-Fi connection credentials.

qingz2004, 2019-10-16:
Yes, and true.
If somebody leaves the door open, there is always a security issue.

meingraham, 2019-10-16:
Not deleted - marked off topic as it is about the web UI authentication and not the Wi-Fi security risk.

ascillato2, 2019-10-16:
For anyone reaching this security issue:

If you have Tasmota Firmware with core 2.3.0 or 2.4.2, please update to latest Tasmota with core pre2.6.0 (http://thehackbox.org/tasmota/) _done by @andrethomas (https://github.com/arendst/Sonoff-Tasmota/pull/6664#issuecomment-542857228)_

And add another security layer adding the webpassword (as recommended by @qingz2004)

closed 2019-10-16 by ascillato2