arendst/Tasmota #6668: Sonoff POW R2, MQTT STAT on change

spatnynick, 2019-10-16:
Hi ,Tasmota sens telemetry to MQTT periodically with help of TelePeriod parameter
It would be nice to have in additional a possibility to trigger MQTT 'OnChange' of (especially) Power value. Otherwise it is impossible to find out exact peak values..

ascillato2, 2019-10-16:
Already implemented.

Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-10-16 by ascillato2

markross42, 2019-11-16:
Hi @ascillato2 ,
Thanks for telling us, that this feature is implemented.
Where do I find the documentation on this feature?

Best regards

Mark

meingraham, 2019-11-16:
Power#State with Status 8