arendst/Tasmota #6669: Add another switch hold timer

SimonHarrisonSH, 2019-10-16:
> Please take a few minutes to complete the requested information below. Our ability to provide assistance is greatly hampered without it. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help you.

**Have you looked for this feature in other issues and in the wiki?**  
Yes

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
No

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

It would be useful to have multiple (2/3) hold timers. Due to the fact that Rules using hold timers are triggered after expiry of the timer, even though the switch is still held down, this makes it possible to have, say, two rules that are triggered at, HOLD_TIME and (HOLD_TIME * 2), for the same button-press. 
My use-case is at Christmas; Single-press turns on the main light (Not often used, direct-connect) long-press (1s) turns on the floor lamps, extra-long-press(2s) turns on the Christmas Tree etc. lights, both via MQTT.

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._
Add extra switch-state (4)
Add hold_timer2 (2x length of switch_timer) to switch struct
SendKey(4) when timer2 expires
This avoids having an extra SetOption, although we could add a SetOption, as a multiplier, for extra flexibility. Not sure it's really necessary though.

**Additional context**  
_Add any other context or screenshots about the feature request here._
Changes required for the simple doubled timer are minimal, I've built and used this successfully on my Sonoff Mini. Will submit a pull request, if others think this is useful.

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2019-10-17:
If you need 3 different actions for the same switch or button, you can use the following example of the wiki at rules:

https://github.com/arendst/Sonoff-Tasmota/wiki/Rule-Cookbook#button-with-single-press-double-press-and-hold

andrethomas2, 2019-10-17:
This is a very unique use case but I agree with @ascillato that it can be done by using rules.

I'm not sure that many users would benefit from a PR relating to this feature so either try to use rules or if the code modification is simple you're welcome to keep it in your own repository.

Closing this issue for now as a work around has been suggested.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-10-17 by andrethomas2

SimonHarrisonSH, 2019-10-17:
OK. I don't think the workaround is quite as spouse-friendly, but suspected that it was quite a niche requirement. I'll keep it to myself, then.