arendst/Tasmota #6670: Sonoff TX 1 wire - no switch status reported? (Time-delay After Switch Off example).

ahostn, 2019-10-17:
Hello, I just flashed tasmota (sonoff.bin Sonoff-Tasmota 6.6.0.18) to my Sonoff TX 1wire - almost everything works fine, except switch status:

01:10:38 CMD: Switch1#State
01:10:38 MQT: sonoff/1/RESULT = {"Counter1":0}

Yes. I did run switchmode1 1 command too.
Heres the code I need to run:

`on switch1#state=1 do backlog
  power1 on;
  ruletimer1 0
endon
on switch1#state=0 do
  ruletimer1 300
endon
on rules#timer=1 do
  power1 0
endon`
This code doesn't even trigger when I press the switch.

The only state I can get is POWER state, but I don't know how to make delay script with this state:
This is my failed attempt:
`rule1
on power1#state=1 do backlog
	power1 on;
	ruletimer1 0
endon
on power1#state=0 do backlog
	// turn it back on...
	power1 on; // this resets timer to 0 and the light stays on
	ruletimer1 20;	
endon
on rules#timer=1 do
  power1 0
endon`

Thank you.

ascillato2, 2019-10-17:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-10-17 by ascillato2

meingraham, 2019-10-17:
`01:10:38 CMD: Switch1#State`

Switch#State is a rule trigger. It is not a command.

As suggested, if you have further questions, please address them to the Discord chat.