arendst/Tasmota #6674: Support color temperature for RGB lights

mijofa, 2019-10-17:
My main purpose for having any sort of RGB light globe is so I can set the colour temperature for different times of the day. I got a cheap RGBW light globe and didn't realise this wouldn't support Tasmota's CT features until after I spent an hour or more trying to figure out why I kept getting "Command:Unknown" when I tried to use them.

Turns out that since this particular globe isn't predefined as an RGBWC globe, it doesn't work. My main confusion here is, why doesn't it use the red/blue channels to get the same effect? I personally don't know the math calculations to match up colour temperature with the RGB channels, but I know it's possible. And that's how redshift does it on my desktop.

For now what I've implemented is some hacky things with an already running setup with redshift setting my desktop screen's colour temperature, then pulling the gamma ramps out of the display system and applying them to the light globe via a HTTP request. It's a fairly botched together thing (I can attach this hook script if requested) but it does the job for now... assuming my desktop is powered on and running in my Debian environment instead of running one of the other dual-booted OSes I have on the system.
Hence it would be much nicer to be able to do this math within Tasmota itself, then I believe I could use the timers functionality to preschedule a serious of colour temperatures around the sunset/sunrise times and never have to think about it again.

Jason2866, 2019-10-17:
Our ability to provide assistance is greatly hampered since you did not take a few minutes to complete the [troubleshooting template](https://github.com/arendst/Sonoff-Tasmota/issues/new/choose) with the requested information. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help you.

Please close this issue and address this to the [Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are best used for Tasmota **_software feature requests and bug reporting_**. Troubleshooting and setup assistance is more effective using an interactive forum.
 
Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).
 
Thanks.
 
### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-10-17 by ascillato2

s-hadinger, 2019-10-17:
The `CT` command only works when you have 2 channels for CW/WW (Cold White/Warm White).

In my experience, it does not work very well to use the RGB part to turn a Cold White into Warm White. The main reason is that often white leds are much brighter than RGB. So to get a correct colortone, you need to drastically reduce the overall brightness. Is that what you observe too?

Adapting color tone based on sunset/sunrise may be doable with scripting.

mijofa, 2019-10-17:
On Thu, Oct 17, 2019 at 11:18:01AM -0700, s-hadinger wrote:
> The `CT` command only works when you have 2 channels for CW/WW (Cold White/Warm White).
>
> In my experience, it does not work very well to use the RGB part to turn a Cold White into Warm White. The [...]

Yeah, I found the 100% brightness white to be way too bright anyway, so
am quite happy to reduce it down to ~75% or so.
However if it's possible to make it brighter when cold-white that might
be reasonable as well, reducing the brightness as the colour temperature
goes up.

> Adapting color tone based on sunset/sunrise may be doable with scripting.

Yeah, I've got something like this done with scripting on the computer,
but would be much better if I could do some of this scripting on the
light globe.

-- 
Mike Abrahall