arendst/Tasmota #6684: light and fan not working together - iFan03

GiladShoham, 2019-10-18:
### ISSUE DESCRIPTION - TROUBLESHOOTING
So I have an iFan03 flashed with tasmota 6.6.0.11.
I have an issue with turn on the light and fan together.
When turning on the fan on power 1 (lowest) the light sometimes stayed on but with lower power. when turning on the fan on power 2/3 the light turned off completely.
In the interfact it's still marked as on, also when I turned off the fan the light is back on.
So it looks like the iFan is not strong enough to power them on both together.
The same happens no matter if I use the remote or the web interface.  
I made a quick video demonstrating the problem here:
https://youtu.be/MMcTmsUMUVY

Also here is a picture of the fan specification (I might translate some of the terms incorrect, but I believe the values will be enough to understand)
[image]

Here are the specs of the original command unit that comes with the fan:
[image]

I'm pretty sure it's a hardware problem, but if you can help me understand if there is something I should replace on the board or something else I can do, it will be great!

Jason2866, 2019-10-18:
Please update to latest development version. There where changes in ifan driver.
Maybe your issue is solved with

closed 2019-10-18 by ascillato2

GiladShoham, 2019-10-18:
Ok thanks,
I tried to flash the latest version (6.6.0.19) (The new beeps are cool).
But it still doesn't work. 
I'll try in the support chat.

Jason2866, 2019-10-18:
Solved in Discord. Wrong wiring