arendst/Tasmota #6685: General(18) doesnt publish switches in auto discovery

magiva, 2019-10-18:
If i set the type to generic(18) and then set some gpio's to switches, they are not published in home assistant with auto discovery.

it would be great if they were auto published so that auto discovery made home assistant config-less

ascillato2, 2019-10-18:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-10-18 by ascillato2

magiva, 2019-10-18:
i did that and it was suggested that this is not a current functionality on
tasmota, hence the suggestion post

br
Mark-Andrew

ascillato, 2019-10-18:
Sorry, It is already implemented. See Request #6530 and Code Merge: #6548

ascillato, 2019-10-18:
update to latest Tasmota v6.6.0.20

magiva, 2019-10-18:
my appologies and thanks for the help

PhilNS, 2022-02-17:
Hi. 
It  looks like  the specified problem is still there with latest Tasmota Release: 11.0.0 as General (18) doesn't publish switches in auto discovery :

Program Version :      11.0.0(tasmota)
Build Date and Time: 2022-02-12T14:13:50
Core/SDK version:      2_7_4_9/2.2.2-dev(38a443e)

From Console we only get:

12:24:24.420 MQT: tele/tasmota_3387E1/SENSOR = {"Time":"2022-02-17T12:24:24","Switch1":"ON","Switch2":"ON"}