arendst/Tasmota #6686: Problems getting device Mac Address value in a rule (StatusNET#Mac)

scanos, 2019-10-18:
> **GUIDE**
>
> This BUG issue template is meant to REPORT Tasmota software BUGS ONLY>
>  
> Please DO NOT OPEN AN ISSUE:
>  - If your Tasmota version is not the latest from the development branch, please update your device before submitting your issue. Your problem might already be solved. The latest precompiled binaries of Tasmota can be downloaded from http://thehackbox.org/tasmota/  
>  - If your issue is a flashing issue, please address it to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4)  
>  - If your issue is compilation problem, please address it to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4)  
>  - If your issue has been addressed before (i.e., duplicated issue), please ask in the original issue  
>  - If your issue is a Wi-Fi problem or MQTT problem, please try the steps provided in the [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) and troubleshooting wiki articles  
>
> Please take a few minutes to complete the requested information below. Our ability to provide assistance is greatly hampered without it. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help you.

### BUG DESCRIPTION
_A clear and concise description of what the bug is._
I cannot get the value of the device macaddress in a rule using a construct such as ON StatusNET#Mac DO Websend [XXXXX] %value% ENDON. Note, that I can do this for other items such as rule1 ON AM2301#Temperature DO Websend [XXXX] %value% ENDON 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [X ] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X ] Searched the problem in [issues](https://github.com/arendst/Sonoff-Tasmota/issues)
- [ X] Searched the problem in the [wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X ] Device used (e.g., Sonoff Basic): _____Generic Module with Witty ESP32
- [ X] Tasmota binary firmware version number used: _____

Program Version | 6.6.0(release-sonoff)
-- | --
2019-07-06T13:10:20
2_3_0/1.5.3(aec24ac9)

  - [x ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x ] Flashing tools used: _____Direct USB
- [ ] Provide the output of command: ``Backlog Template; Module; GPIO``:
  ```  Configuration output here:
  18:30:25 MQT: stat/sonoff_node_1/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
18:30:25 MQT: stat/sonoff_node_1/RESULT = {"Module":"18 (Generic)"}
18:30:26 MQT: stat/sonoff_node_1/RESULT = {"GPIO0":"0 (None)","GPIO1":"0 (None)","GPIO2":"0 (None)","GPIO3":"0 (None)","GPIO4":"0 (None)","GPIO5":"0 (None)","GPIO12":"0 (None)","GPIO13":"0 (None)","GPIO14":"2 (AM2301)","GPIO15":"0 (None)","GPIO16":"0 (None)"}
  
  ```
- [X ] If using rules, provide the output of this command: ``Backlog Rule1; Rule2; Rule3``:
Rules output here:
18:32:16 MQT: stat/sonoff_node_1/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":450,"Rules":"ON StatusNET#Mac DO Websend [XXXXX] %value% ENDON"}
  ```
  
  
  
  ```
- [ X] Provide the output of this command: ``Status 0``:
  ```
  STATUS 0 output here:
  8:33:07 MQT: stat/sonoff_node_1/STATUS = {"Status":{"Module":18,"FriendlyName":["Sonoff_upstairs"],"Topic":"sonoff_node_1","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
18:33:08 MQT: stat/sonoff_node_1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:09:57","StartupUTC":"2019-10-18T17:23:10","Sleep":50,"CfgHolder":4617,"BootCount":87,"SaveCount":974,"SaveAddress":"FA000"}}
18:33:08 MQT: stat/sonoff_node_1/STATUS2 = {"StatusFWR":{"Version":"6.6.0(release-sonoff)","BuildDateTime":"2019-07-06T13:10:20","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
18:33:08 MQT: stat/sonoff_node_1/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["BTHub6-WQRX",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000100060000000000000000000000","00000001"]}}
18:33:08 MQT: stat/sonoff_node_1/STATUS4 = {"StatusMEM":{"ProgramSize":503,"Free":500,"Heap":13,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":3,"Features":["00000809","0FDAE396","001783A0","22B617CD","01003BC0"]}}
18:33:08 MQT: stat/sonoff_node_1/STATUS5 = {"StatusNET":{"Hostname":"sonoff_node_1-5225","IPAddress":"192.168.1.135","Gateway":"192.168.1.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.254","Mac":"60:01:94:5E:54:69","Webserver":2,"WifiConfig":4}}
18:33:08 MQT: stat/sonoff_node_1/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.197","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_5E5469","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
18:33:08 MQT: stat/sonoff_node_1/STATUS7 = {"StatusTIM":{"UTC":"Fri Oct 18 17:33:08 2019","Local":"Fri Oct 18 18:33:08 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"07:14","Sunset":"17:55"}}
18:33:08 MQT: stat/sonoff_node_1/STATUS10 = {"StatusSNS":{"Time":"2019-10-18T18:33:08","AM2301":{"Temperature":12.0,"Humidity":94.7},"TempUnit":"C"}}
18:33:08 MQT: stat/sonoff_node_1/STATUS11 = {"StatusSTS":{"Time":"2019-10-18T18:33:08","Uptime":"0T00:09:58","Heap":13,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":21,"Wifi":{"AP":1,"SSId":"BTHub6-WQRX","BSSId":"A0:39:EE:5B:CC:2C","Channel":1,"RSSI":68,"LinkCount":1,"Downtime":"0T00:00:07"}}}
  
  ```
- [X ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ ``weblog 4`` _for more debug information)_
  ```
  Console output here:
  Saved to flash at F9, Count 975, Bytes 3584
  
  ```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Turned rules off and on repeatedly.
Changed rule 1 to my sensor values AM2301#Temperature and Checked that my sensor values continued to be processed by the rule
i.e. 
rule1 ON AM2301#Temperature DO Websend [XXXXX] %value% ENDON
outputted 
18:37:02 SRC: Rule
18:37:02 RSL: Received Topic /Websend, Data Size 24, Data [XXXXX] 12.0
18:37:02 RSL: Group 0, Index 1, Command WEBSEND, Data [XXXX] 12.0
18:37:02 MQT: stat/sonoff_node_1/RESULT = {"WebSend":"Done"}

Again, there was no output when I replaced with AM2301#Temperature with StatusNET#Mac in rule1
### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
I expected an output from the console window when rule1 was triggered. Also, I am monitoring the logs on my web server and whilst they show the value of AM2301#Temperature, they show nothing when rule1 contains

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

jziolkowski, 2019-10-18:
Status messages and their values are not valid rule trigger. See the related rules article for a list of possible triggers.

ascillato2, 2019-10-18:
Hi,

Sorry. Not a bug. That trigger is not supported.

closed 2019-10-18 by ascillato2