arendst/Tasmota #6690: DS18B20 Temperature Dont' work

sriccardi-invn, 2019-10-19:
The sonoff TH16 send correct message :
14:27:09 MQT: tele/sonoff_water_heater/SENSOR = {"Time":"2019-10-19T14:27:09","DS18B20":{"Temperature":18.2},"TempUnit":"C"}

But on Jeedom platform with MQTT plugin, the topic is not discovered with the good format:
tele/sonoff_water_heater/SENSOR{DS18B20}

Jason2866, 2019-10-19:
Tasmota sends a valid Json mqtt string. There will be no change.
Please close since it is NOT a issue in Tasmota.

Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

sriccardi-invn, 2019-10-19:
Tamosta only send sensor data ?
It don't send the list of topics for automatic discover?

Jason2866, 2019-10-19:
Read wiki, all explained

closed 2019-10-19 by sriccardi-invn