arendst/Tasmota #6692: Send web requests via Tasmota to IFTTT( Webhooks)

hwarang098tkd, 2019-10-19:
So i have a simple wemos d1 mini with tasmota software and i wae wandering if it is possible to send via RULES a   post request to IFTTT.

```
on power2#state=1 do
	https://maker.ifttt.com/trigger/david/with/key/mykey
endon 
```

thnx in advance

Jason2866, 2019-10-19:
Our ability to provide assistance is greatly hampered since you did not take a few minutes to complete the [troubleshooting template](https://github.com/arendst/Sonoff-Tasmota/issues/new/choose) with the requested information. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help you.

Please close this issue and address this to the [Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are best used for Tasmota **_software feature requests and bug reporting_**. Troubleshooting and setup assistance is more effective using an interactive forum.
 
Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).
 
Thanks.
 
### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

jziolkowski, 2019-10-19:
Also no, you can't do it

closed 2019-10-19 by ascillato2

FrankBolleri, 2021-01-01:
Hello All, 
just a comment to help people searching for this matter and landing on this page in the future.

Something like the example reported above will currently work, if using WebSend command and putting the host portion between square brackets:

_on power2#state=1 do
	**WebSend** **[**maker.ifttt.com**]** **/trigger/user/with/key/mykey**
endon_ 

I take the chance of this comment to say **thanks to all the Tasmota team for the great job.**