arendst/Tasmota #6693: Enhanced lighting control

naylin-mi, 2019-10-19:
Hi, 
First let me say that I love Tasmota! It's an excellent piece of software/firmware for home automation.

Now, on to the nitty-gritty. 

Would it be possible to add a way to control all aspects of a light (brightness, color, etc.) in one go? Basically like set a status and then execute the status. For example: (probably not exactly like this but something simillar)
`LIGHT/SET {dimmer 50, color red, power on}` 
and then 
`LIGHT/EXEC` 
and it executes with the light in the configured status.

I know that there is the Backlog command but, imo, while it is an excellent feature for it's ability to execute commands to multiple devices, it's not very elegant for controlling a single light. It's pretty ugly in fact. Say I have a light that is red at 100% brightness and I want to go to white at 50%, using backlog, I either get it white @ 100% and then it goes to 50% or I get red @ 50% and then switches to white. While it works, it's not much of a smooth transition. But, if one could set all the attributes and then execute it in one go, it should be able to "just change."

I've done a lot of searching and reading and I can't find any way to do this (if I'm missing something PLEASE let me know!)

Thanks so much for your time and effort into making this project such an excellent one,

~Naylin

ascillato2, 2019-10-19:
Hi, 

For bright transitions on lights, you can use command fade and speed. Please, check the wiki at commands. If you use a backlog set of commands, you can set the delay between them. See related setoption at commands in the wiki. 

Please, if you have further questions on Tasmota usage, address those to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-10-19 by ascillato2

s-hadinger, 2019-10-19:
The closest to what you are looking for is the `Color` command which lets you set all channels with a single command and with no transition. Ex: `Color 40408000`

naylin-mi, 2019-10-19:
@ascillato2 Considering this wasn't a question about Tasmota usage nor a support request, but rather a feature request, I figured this was the appropriate place to post and marked it as such. If this wasn't a proper place to make such a request, perhaps that option should be removed from the list and should be set to ONLY for "Software Bug Reporting."

Thank you @s-hadinger that's the way I'm using it currently (actually using HsbColor but...) in combination with backlog. I haven't placed any delay commands between each command and there is about a 1 second delay between each command actually executing. Perhaps I can try putting a "delay 0" between each one and see if that removes the apparently inherent delay built-in. I just assumed that without a delay command between, there should be no delay and adding a delay would do just that considering there is no mention of a built-in delay in the [wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#using-backlog). HENCE the feature request.

Thanks again for your time,

~Naylin

s-hadinger, 2019-10-19:
Try SetOption34 to reduce the inter- commands delay with backlog.

ascillato, 2019-10-19:
>Considering this wasn't a question about Tasmota usage nor a support request, but rather a feature request, I figured this was the appropriate place to post and marked it as such.

Sorry for the misunderstanding. I didn't meant to upset you. Your feature request is already doable in Tasmota but in a different way, that is why the link to the support chat was added in order to explain to you all those things. 

As explained in the wiki, color2 command will change the color without changing the brightness. Fade command will make brightness transitions smoothly as defined with the speed command. The backlog delay is controlled by setoption34.

If you need further assistant, please find us in the Tasmota support chat. 

Thanks