arendst/Tasmota #6694: FLHS-ZN04 Power Monitoring Plug not working with 6.6.0.20

stepanov1975, 2019-10-19:
### BUG DESCRIPTION
Today I upgraded my 3 FLHS-ZN04 plugs to 6.6.0.20 version. After the upgrade, all rebooted with exception and gone back to Sonoff basic module. The update was done using Tasmoadmin.
When I tried to downgrade to version 6.6.0.0 2 of 3 didn't come online after uploading sonoff-minimal.bin. One was downgraded to 6.6.0.0 successfully.
Additionally, I have 2 FLHS-ZN04 plugs workings with 6.6.0.15 firmware. But they not from the same batch.

- [v ] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [v ] Searched the problem in [issues](https://github.com/arendst/Sonoff-Tasmota/issues)
- [ v] Searched the problem in the [wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ v] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ v] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ v] Device used (e.g., Sonoff Basic): FLHS-ZN04
- [v ] Tasmota binary firmware version number used: 6.6.0.20
  - [ v] Pre-compiled
- [ v] Flashing tools used: OTA
- [ v] Provide the output of command: ``Backlog Template; Module; GPIO``:
  ```
  Configuration output here:
  15:56:38 CMD: Backlog Template; Module; GPIO
15:56:38 MQT: stat/FLHS_ZN04_3/RESULT = {"NAME":"Israel plug","GPIO":[57,0,56,131,0,134,0,0,0,17,132,21,0],"FLAG":0,"BASE":45}
15:56:38 MQT: stat/FLHS_ZN04_3/RESULT = {"Module":{"1":"Sonoff Basic"}}
15:56:38 MQT: stat/FLHS_ZN04_3/RESULT = {"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO14":{"0":"None"}}
  
  ```
- [ ] Provide the output of this command: ``Status 0``:
  ```
  STATUS 0 output here:
  15:57:13 CMD: Status 0
15:57:13 MQT: stat/FLHS_ZN04_3/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff"],"Topic":"FLHS_ZN04_3","ButtonTopic":"0","Power":1,"PowerOnState":1,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
15:57:13 MQT: stat/FLHS_ZN04_3/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://192.168.0.7:9999/data/firmwares/sonoff.bin","RestartReason":"Exception","Uptime":"0T07:33:54","StartupUTC":"2019-10-19T07:23:19","Sleep":50,"CfgHolder":4617,"BootCount":246,"SaveCount":371,"SaveAddress":"F7000"}}
15:57:13 MQT: stat/FLHS_ZN04_3/STATUS2 = {"StatusFWR":{"Version":"6.6.0.20(d62d33b-sonoff)","BuildDateTime":"2019-10-18T19:00:12","Boot":4,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)"}}
15:57:13 MQT: stat/FLHS_ZN04_3/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["IoT_FL1",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","000000C0"]}}
15:57:13 MQT: stat/FLHS_ZN04_3/STATUS4 = {"StatusMEM":{"ProgramSize":561,"Free":440,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144068","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00007081"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
15:57:13 MQT: stat/FLHS_ZN04_3/STATUS5 = {"StatusNET":{"Hostname":"FLHS_ZN04_3-7494","IPAddress":"192.168.0.31","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"CC:50:E3:14:1D:46","Webserver":2,"WifiConfig":4}}
15:57:13 MQT: stat/FLHS_ZN04_3/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.7","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_141D46","MqttUser":"mqttuser","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
15:57:13 MQT: stat/FLHS_ZN04_3/STATUS7 = {"StatusTIM":{"UTC":"Sat Oct 19 14:57:13 2019","Local":"Sat Oct 19 15:57:13 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"07:16","Sunset":"17:54"}}
15:57:13 MQT: stat/FLHS_ZN04_3/STATUS10 = {"StatusSNS":{"Time":"2019-10-19T15:57:13"}}
15:57:13 MQT: stat/FLHS_ZN04_3/STATUS11 = {"StatusSTS":{"Time":"2019-10-19T15:57:13","Uptime":"0T07:33:54","UptimeSec":27234,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"IoT_FL1","BSSId":"2A:E8:29:EA:8D:67","Channel":13,"RSSI":48,"LinkCount":1,"Downtime":"0T00:00:07"}}}
  
  ```

### TO REPRODUCE
Upgrade FLHS-ZN04 to 6.6.0.20

### EXPECTED BEHAVIOUR
Expected to work as usual

ascillato2, 2019-10-19:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-10-19 by ascillato2

Jason2866, 2019-10-19:
Just upgraded a Blitzwolf SHP 6 to 6.6.0.20.
Works without a issue

stepanov1975, 2019-10-19:
So if someone will have same problem. I flashed via serial and it works now with 6.6.0.20. Probably as @Faziaty mentioned in the chat problems with leftovers after OTA conversion. (This device was flashed originally using tuya convert)