arendst/Tasmota #6695: ESP AP mode

au190, 2019-10-19:
First of all I want to THANK for those who wrote this great software.

In tasmota its hard to switch to AP mode.
This is a idea what would be great in tasmota. This part of the software is critical and must be raliable.

Some devices does not have external button or switch - for these devices is hard to change the WIFI. For these devices you can just plug and unpuld the device (switch the power ON/OFF ) 5 times and the ESP goes in AP mode and not return from this mode until you do not saved the WIFI config.
If you plug the device increments a counter (AP_counter), after 5 seconds this counter (AP_counter) set to zero, if you repeat this (AP_counter =5) 5 times automaticaly goes to AP mode, and remain in this state until you do not saved the WIFI config, even if you unplug the device and plug it again.

SwitchTOAP  = 0 - (default) - if the button pressed 4 times goes in AP mode and not return from this mode until you do not saved the WIFI config or if you switch the power ON/OFF 5 time goes in AP mode and not return from this mode until you do not saved the WIFI config
SwitchTOAP  = 1 - if the button pressed 4 times goes in AP mode and not return from this mode until you do not saved the WIFI config
SwitchTOAP  = 2 - if you switch the power ON/OFF 5 time goes in AP mode and not return from this mode until you do not saved the WIFI config

I have some concesrns about the AP mode. Let me describe.
1. If the router is break or goes off the wifi the ESP goes in AP mode - in this case there is a vourneability - somebody can connect to it and can do anyithing. 
2. If the ESP has weak WIFI strehngt and semetimes can not see the reouter - ESP goes in AP mode - in this case there is a vourneability - somebody can connect to it and can do anyithing. 
3. If I have chaged the router and I have different SSID, I just can not set easiliy the new WIFI. Not allways goes in AP mode.
4. If I bring the ESP to my freind, I just can not set easiliy the new WIFI. Not allways goes in AP mode.

If WifiConfig = 1 = set SmartConfig (Android/iOS) as the current configuration tool, start SmartConfig for 3 minutes, and then reboot and try to connect to your Wi-Fi network again

SetOption13 = 1 = only single press action for immediate response (i.e., disable multipress detection). Disable by holding for 4 x button hold time (see SetOption32).
I use thsi config because immediate response - but in this case. But if I do not have button ?

If plug and unpuld the device (switch the power ON/OFF ) 5 times - now its resetting the  SONOFF_BASIC;  Its not clear for me why this setting- this (RtcReboot.fast_reboot_count ) should be using only for AP mode, all other config can be done form the web interface.
What do you think ?

if (RtcReboot.fast_reboot_count > 4) {      // Restarted 5 times
Settings.module = SONOFF_BASIC;

andrethomas, 2019-10-19:
RtcReboot.fast_reboot_count will not be triggered by power cycle - It requires the RTC ram to be active during restarts - RTC mem is cleared when power is removed.

au190, 2019-10-19:
Yes, but you can use the EEPROM to save this value(AP_counter).

andrethomas, 2019-10-19:
I'm not totally sure I understand what you mean because you seem to be mixing two different issues and you're reminding me of http://xyproblem.info/

For device recovery there is already functionality added for this in the last few days and documented on https://github.com/arendst/Sonoff-Tasmota/wiki/Device-Operational-Recovery

The functionality of the boot counter relates to instances where there is usually a configuration problem or a rules problem that was detected because of fast reboot cycles so that its not directly related to device recovery as such.

au190, 2019-10-19:
It was not clear for me the fast reboot cycles. I try to explain again.
It would be grat if would be a methond to set ESP in AP mode any time when a user what to do it. In this case do not need to reset any configuration just you want to set new WIFI SSID and password.
Some devices does not have external button or switch - for these devices is hard to change the WIFI. For that I thought to do such configuraton as I discribed above.

andrethomas, 2019-10-19:
From the wiki link

[image]

au190, 2019-10-19:
That is great, let me try. From where can I download ?
I can see just https://github.com/arendst/Sonoff-Tasmota/releases/tag/v6.6.0

Jason2866, 2019-10-19:
No, here http://thehackbox.org/tasmota/sonoff.bin

andrethomas, 2019-10-19:
It is not included in the release yet.

You may access the current development binaries by reading this : https://github.com/arendst/Sonoff-Tasmota#development

Please consider reading more as you keep asking questions which are already in the Wiki.

Can you close this issue now since there is nothing further? Thanks.

au190, 2019-10-19:
Thanks for your help.

closed 2019-10-19 by au190