arendst/Tasmota #6700: Cannot store TLS key to configure AWS IoT

cschnidr, 2019-10-20:
Hi all,
I compiled last dev version 6.6.0.20 and enabled AWS IOT as described in https://github.com/arendst/Sonoff-Tasmota/wiki/AWS-IoT.
TLSKey command is known but I cannot store any key in the keystore by using Webconsole:
`12:08:23 CMD: TLSKey 0
12:08:23 RSL: stat/sonoff/RESULT = {"TLSKey1":-1,"TLSKey2":-1}
12:08:59 CMD: TLSKey1 UvBgyCuPr/lKSgwumf/8o/mIsKQPBHn3ZZAGZl4ui9E=
12:08:59 RSL: stat/sonoff/RESULT = {"TLSKey1":-1,"TLSKey2":-1}`

Note: I converted the key using both methods in the wiki. Here I use the key from the wiki just to see if I can upload a key.
I also erased the whole flash in my Wemos when uploading the firmware. Core is 2.5.2.

ascillato2, 2019-10-20:
Please, could you be so kind on completing the [troubleshooting template](https://github.com/arendst/Sonoff-Tasmota/issues/new/choose) in order to have more information so as to properly help you?

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

s-hadinger, 2019-10-20:
I typed exactly your command above:

`TLSKey1 UvBgyCuPr/lKSgwumf/8o/mIsKQPBHn3ZZAGZl4ui9E=`

And it worked fine:
```
TLSKey1 UvBgyCuPr/lKSgwumf/8o/mIsKQPBHn3ZZAGZl4ui9E=
15:19:38 CMD: TLSKey1 UvBgyCuPr/lKSgwumf/8o/mIsKQPBHn3ZZAGZl4ui9E=
15:19:38 RSL: stat/sonoff/RESULT = {"TLSKey1":32,"TLSKey2":-1}
```

What version are you using?

ascillato2, 2019-10-21:
@cschnidr 

Hi, 

Any news on this?

ascillato2, 2019-10-21:
When you have time, please complete the template. Thanks.

closed 2019-10-21 by ascillato2

cschnidr, 2019-10-22:
Hi here is the completed template. Thank you!

### ISSUE DESCRIPTION - TROUBLESHOOTING

I compiled last dev version 6.6.0.20 and enabled AWS IOT as described in https://github.com/arendst/Sonoff-Tasmota/wiki/AWS-IoT.
Here the adjusted parameters:
#define USE_MQTT_TLS
#define USE_MQTT_TLS_CA_CERT
#define USE_MQTT_TLS_FORCE_EC_CIPHER
#define USE_MQTT_AWS_IOT
//#define USE_DISCOVERY

TLSKey command is known but I cannot store any key in the keystore by using Webconsole. No error occur, the key is just not stored.
12:08:23 CMD: TLSKey 0
12:08:23 RSL: stat/sonoff/RESULT = {"TLSKey1":-1,"TLSKey2":-1}
12:08:59 CMD: TLSKey1 UvBgyCuPr/lKSgwumf/8o/mIsKQPBHn3ZZAGZl4ui9E=
12:08:59 RSL: stat/sonoff/RESULT = {"TLSKey1":-1,"TLSKey2":-1}

Note: I converted the key using both methods in the wiki. In the example above I use the key from the wiki just to see if I can upload a key.
I also erased the whole flash in my Wemos when uploading the firmware. Core is 2.5.2.

What might go wrong that I cannot store the key?
  

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Wemos D1 mini (ESP8266 with 4M Mem)
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [X] Self-compiled 6.6.0.20
    - [X] IDE / Compiler used: Arduino IDE
- [X] Flashing tools used: Arduino IDE
- [ ] Provide the output of this command: ``Backlog Template; Module; GPIO``:
  ```
  Configuration output here:

06:28:18 CMD: Backlog Template; Module; GPIO
06:28:18 RSL: stat/sonoff/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
06:28:19 RSL: stat/sonoff/RESULT = {"Module":{"18":"Generic"}}
06:28:19 RSL: stat/sonoff/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"1":"DHT11"},"GPIO5":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}
  
  ```
- [ ] If using rules, provide the output of this command: ``Backlog Rule1; Rule2; Rule3``:
  ```
  Rules output here:
  
06:28:55 CMD: Backlog Rule1; Rule2; Rule3
06:28:55 RSL: stat/sonoff/RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
06:28:56 RSL: stat/sonoff/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
06:28:56 RSL: stat/sonoff/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
  
  ```
- [ ] Provide the output of this command: ``Status 0``:
  ```
  STATUS 0 output here:
06:29:42 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.6.0.20(sonoff)","BuildDateTime":"2019-10-20T12:55:45","Boot":31,"Core":"2_5_2","SDK":"2.2.1(cfd48f3)"}}
06:29:42 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":2,"LogHost":"","LogPort":514,"SSId":["version2","<your_ssid2>"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000"]}}
06:29:42 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":579,"Free":3496,"Heap":20,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164020","FlashMode":2,"Features":["00000809","8FD8E3C7","003683A0","22B617C5","010008C0","00007081"],"Drivers":"1,2,3,4,6,7,8,9,10,12,16,18,19,20,21,22,24,26","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
06:29:42 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-4499","IPAddress":"192.168.37.231","Gateway":"192.168.37.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.37.1","Mac":"80:7D:3A:6E:71:93","Webserver":2,"WifiConfig":4}}
06:29:42 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"DVES_USER","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_6E7193","MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
06:29:42 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Tue Oct 22 05:29:42 2019","Local":"Tue Oct 22 06:29:42 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"07:21","Sunset":"17:48"}}
06:29:42 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-10-22T06:29:42","DHT11":{"Temperature":23.9,"Humidity":67.0},"TempUnit":"C"}}
06:29:42 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-10-22T06:29:42","Uptime":"0T00:07:54","UptimeSec":474,"Heap":20,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":0,"Wifi":{"AP":1,"SSId":"version2","BSSId":"18:E8:29:FE:05:0A","Channel":6,"RSSI":54,"LinkCount":1,"Downtime":"0T00:00:05"}}}
  
  ```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ ``weblog 4`` _for more debug information)_
  ```
  Console output here:
06:31:41 CMD: Group 0, Index 1, Command "WEBLOG", Data "4"
06:31:41 RSL: stat/sonoff/RESULT = {"WebLog":4}
06:31:44 CMD: TLSKey1 UvBgyCuPr/lKSgwumf/8o/mIsKQPBHn3ZZAGZl4ui9E=
06:31:44 SRC: WebConsole from 192.168.37.221
06:31:44 CMD: Group 0, Index 1, Command "TLSKEY", Data "UvBgyCuPr/lKSgwumf/8o/mIsKQPBHn3ZZAGZl4ui9E="
06:31:44 RSL: stat/sonoff/RESULT = {"TLSKey1":-1,"TLSKey2":-1}  
  
  ```

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2019-10-22:
It looks like the data was correctly parsed, but not correctly written to flash.

Can you enable `#define DEBUG_DUMP_TLS ` (xdrv_02_mqtt.ino, line 22) and type `TlsKey0` before and after the `TlsKey1 ...` command.

Also, what device are you using? Are you using any specific linker file, like 4M?

ascillato, 2019-10-22:
Core 2.5.2 has memory issues due to memory manager is not in IRAM, please use core pre2.6.0 and standard 1M memory map

cschnidr, 2019-10-23:
Okay, I recompiled by using actual pre2.6.0 core with standard memory map. and pulled a new Tasmota source (6.6.0.21) too.
Parameters in user config overrride are:
#define USE_MQTT_TLS
#define USE_MQTT_TLS_CA_CERT
#define USE_MQTT_TLS_FORCE_EC_CIPHER
#define USE_MQTT_AWS_IOT
#undef USE_DISCOVERY

I added #define DEBUG_DUMP_TLS in xdrv_02_mqtt.ino too.

Device is a Wemos D1 mini. I do not use a specific linker file.

22:31:44 CMD: Group 0, Index 1, Command "TLSKEY", Data "0"
22:31:44 RSL: stat/sonoff/RESULT = {"TLSKey1":-1,"TLSKey2":-1}
22:31:47 CMD: TLSKey1 UvBgyCuPr/lKSgwumf/8o/mIsKQPBHn3ZZAGZl4ui9E=
22:31:47 CMD: Group 0, Index 1, Command "TLSKEY", Data "UvBgyCuPr/lKSgwumf/8o/mIsKQPBHn3ZZAGZl4ui9E="
22:31:47 RSL: stat/sonoff/RESULT = {"TLSKey1":-1,"TLSKey2":-1}
22:31:51 CMD: TLSKey 0
22:31:51 CMD: Group 0, Index 1, Command "TLSKEY", Data "0"
22:31:51 RSL: stat/sonoff/RESULT = {"TLSKey1":-1,"TLSKey2":-1}

Command is still not storing the key into flash. Any other ideas?

s-hadinger, 2019-10-24:
Can you please type `TlsKey0` before and after storing the key and paste the output here?

cschnidr, 2019-10-24:
`TlsKey0` is giving me an error... That's why I used `Tlskey 0` above.

06:50:26 CMD: Tlskey0
06:50:26 SRC: WebConsole from 192.168.37.221
06:50:26 CMD: Group 0, Index 0, Command "TLSKEY", Data ""
06:50:26 RSL: stat/sonoff/RESULT = {"Command":"Error"}
06:50:28 CMD: TLSKey1 UvBgyCuPr/lKSgwumf/8o/mIsKQPBHn3ZZAGZl4ui9E=
06:50:28 SRC: WebConsole from 192.168.37.221
06:50:28 CMD: Group 0, Index 1, Command "TLSKEY", Data "UvBgyCuPr/lKSgwumf/8o/mIsKQPBHn3ZZAGZl4ui9E="
06:50:28 RSL: stat/sonoff/RESULT = {"TLSKey1":-1,"TLSKey2":-1}
06:50:33 CMD: TlsKey0
06:50:33 SRC: WebConsole from 192.168.37.221
06:50:33 CMD: Group 0, Index 0, Command "TLSKEY", Data ""
06:50:33 RSL: stat/sonoff/RESULT = {"Command":"Error"}

s-hadinger, 2019-10-24:
It means that there is something weird with your firmware or the way you are compiling it. When you add `#define DEBUG_DUMP_TLS`, `TlsKey0` should dump the flash memory location where the TLS keys are stored.

Beware, `TlsKey0` dumps the flash while `TlsKey 0` clears the keys in flash...

s-hadinger, 2019-10-24:
Ah sorry, I realize now that the dump is output on Serial, not in the console... Do you have a serial terminal attached to it?

cschnidr, 2019-10-24:
Yeah, realized this too... sorry. The TlsKey1 command doesn't seem to change something...

07:24:15 CMD: Tlskey0
405fe400:  00801500 00801500 00801500 00801500
405fe410:  00802d00 00802d00 00802d00 00802d00
405fe420:  00805700 00805700 00805700 00805700
405fe430:  00807e00 00807e00 00807e00 00807e00
405fe440:  00807e00 00807e00 00807e00 00807e00
405fe450:  00807e00 00807e00 00807e00 00807e00
405fe460:  00807e00 00807e00 00807e00 00807e00
405fe470:  00807e00 00807e00 00807e00 00807e00
405fe480:  00807e00 00807e00 00807e00 00807e00
405fe490:  00807e00 00807e00 00807e00 00807e00
405fe4a0:  00807e00 00807e00 00807e00 00807e00
405fe4b0:  00807e00 00807e00 00807e00 00807e00
405fe4c0:  00807e00 00807e00 00807e00 00807e00
405fe4d0:  00807e00 00807e00 00807e00 00807e00
405fe4e0:  00807e00 00807e00 00807e00 00807e00
405fe4f0:  00807e00 00807e00 00807e00 00807e00
405fe500:  00807e00 00807e00 00807e00 00807e00
405fe510:  00807e00 00807e00 00807e00 00807e00
405fe520:  00807e00 00807e00 00807e00 00807e00
405fe530:  00807e00 00807e00 00807e00 00807e00
405fe540:  00807e00 00807e00 00807e00 00807e00
405fe550:  00807e00 00807e00 00807e00 00807e00
405fe560:  00807e00 00807e00 00807e00 00807e00
405fe570:  00807e00 00807e00 00807e00 00807e00
405fe580:  00807e00 00807e00 00807e00 00807e00
405fe590:  00807e00 00807e00 00807e00 00807e00
405fe5a0:  00807e00 00807e00 00807e00 00807e00
405fe5b0:  00807e00 00807e00 00807e00 00807e00
405fe5c0:  00807e00 00807e00 00807e00 00807e00
405fe5d0:  00807e00 00807e00 00807e00 00807e00
405fe5e0:  00807e00 00807e00 00807e00 00807e00
405fe5f0:  00807e00 00807e00 00807e00 00807e00
405fe600:  00807e00 00807e00 00807e00 00807e00
405fe610:  00807e00 00807e00 00807e00 00807e00
405fe620:  00807e00 00807e00 00807e00 00807e00
405fe630:  00807e00 00807e00 00807e00 00807e00
405fe640:  00807e00 00807e00 00807e00 00807e00
405fe650:  00807e00 00807e00 00807e00 00807e00
405fe660:  00807e00 00807e00 00807e00 00807e00
405fe670:  00807e00 00807e00 00807e00 00807e00
405fe680:  00807e00 00807e00 00807e00 00807e00
405fe690:  00807e00 00807e00 00807e00 00807e00
405fe6a0:  00807e00 00807e00 00807e00 00807e00
405fe6b0:  00807e00 00807e00 00807e00 00807e00
405fe6c0:  00807e00 00807e00 00807e00 00807e00
405fe6d0:  00807e00 00807e00 00807e00 00807e00
405fe6e0:  00807e00 00807e00 00807e00 00807e00
405fe6f0:  00807e00 00807e00 00807e00 00807e00
405fe700:  00807e00 00807e00 00807e00 00807e00
405fe710:  00807e00 00807e00 00807e00 00807e00
405fe720:  00807e00 00807e00 00807e00 00807e00
405fe730:  00807e00 00807e00 00807e00 00807e00
405fe740:  00807e00 00807e00 00807e00 00807e00
405fe750:  00807e00 00807e00 00807e00 00807e00
405fe760:  00807e00 00807e00 00807e00 00807e00
405fe770:  00807e00 00807e00 00807e00 00807e00
405fe780:  00807e00 00807e00 00807e00 00807e00
405fe790:  00807e00 00807e00 00807e00 00807e00
405fe7a0:  00807e00 00807e00 00807e00 00807e00
405fe7b0:  00807e00 00807e00 00807e00 00807e00
405fe7c0:  00807e00 00807e00 00807e00 00807e00
405fe7d0:  00807e00 00807e00 00807e00 00807e00
405fe7e0:  00807e00 00807e00 00807e00 00807e00
405fe7f0:  00807e00 00807e00 00807e00 00807e00
07:24:15 RSL: stat/sonoff/RESULT = {"Command":"Error"}
07:24:23 CMD: TLSKey1 UvBgyCuPr/lKSgwumf/8o/mIsKQPBHn3ZZAGZl4ui9E=
07:24:23 RSL: stat/sonoff/RESULT = {"TLSKey1":-1,"TLSKey2":-1}
[... 41 lines cut ...]
405fe680:  00807e00 00807e00 00807e00 00807e00
405fe690:  00807e00 00807e00 00807e00 00807e00
405fe6a0:  00807e00 00807e00 00807e00 00807e00
405fe6b0:  00807e00 00807e00 00807e00 00807e00
405fe6c0:  00807e00 00807e00 00807e00 00807e00
405fe6d0:  00807e00 00807e00 00807e00 00807e00
405fe6e0:  00807e00 00807e00 00807e00 00807e00
405fe6f0:  00807e00 00807e00 00807e00 00807e00
405fe700:  00807e00 00807e00 00807e00 00807e00
405fe710:  00807e00 00807e00 00807e00 00807e00
405fe720:  00807e00 00807e00 00807e00 00807e00
405fe730:  00807e00 00807e00 00807e00 00807e00
405fe740:  00807e00 00807e00 00807e00 00807e00
405fe750:  00807e00 00807e00 00807e00 00807e00
405fe760:  00807e00 00807e00 00807e00 00807e00
405fe770:  00807e00 00807e00 00807e00 00807e00
405fe780:  00807e00 00807e00 00807e00 00807e00
405fe790:  00807e00 00807e00 00807e00 00807e00
405fe7a0:  00807e00 00807e00 00807e00 00807e00
405fe7b0:  00807e00 00807e00 00807e00 00807e00
405fe7c0:  00807e00 00807e00 00807e00 00807e00
405fe7d0:  00807e00 00807e00 00807e00 00807e00
405fe7e0:  00807e00 00807e00 00807e00 00807e00
405fe7f0:  00807e00 00807e00 00807e00 00807e00
07:24:30 RSL: stat/sonoff/RESULT = {"Command":"Error"}

cschnidr, 2019-10-24:
I actually also changed the hardware to rule out anything there, but I have the exact same problem.

s-hadinger, 2019-10-24:
This is weird and seems to indicate the address calculation of the flash area is wrong.

You should see something like this:

```
402ff400:  ffffffff ffffffff ffffffff ffffffff
402ff410:  ffffffff ffffffff ffffffff ffffffff
402ff420:  ffffffff ffffffff ffffffff ffffffff
402ff430:  ffffffff ffffffff ffffffff ffffffff
402ff440:  ffffffff ffffffff ffffffff ffffffff
402ff450:  ffffffff ffffffff ffffffff ffffffff
402ff460:  ffffffff ffffffff ffffffff ffffffff
402ff470:  ffffffff ffffffff ffffffff ffffffff
...
```

Strangely your `Status4` indicates 4M flash size:
` stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":579,"Free":3496,"Heap":20,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164020","FlashMode":2,"Features":["00000809","8FD8E3C7","003683A0","22B617C5","010008C0","00007081"],"Drivers":"1,2,3,4,6,7,8,9,10,12,16,18,19,20,21,22,24,26","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}`

What tool are you using to flash the Wemo D1 mini?

@arendst I'm not very familiar with linker scripts. Any idea?

cschnidr, 2019-10-24:
Thank you very munch for your effort and your prompt feedbacks!
The Wemo D1 I'm using has 4M flash size. It's auto-detected by the Arduino IDE.
I flash them by using the Arduino IDE.

ascillato, 2019-10-24:
@cschnidr

>The Wemo D1 I'm using has 4M flash size. It's auto-detected by the Arduino IDE.

That is your issue. Wrong memory map. As explained before:

>please use core pre2.6.0 and standard 1M memory map

The easiest way is to use GITPOD (online and fast) or VSC+PIO (local)

See wiki for more details. If you need further help on compiling, please, use the Tasmota Support Chat

cschnidr, 2019-10-24:
Okay guys - thanks a lot for helping. The problem is fixed!
I used "Wemos R1 D1" as the board type in the Arduino IDE. This did not allow me to change the Flash size to 1MB. I changed now the board type to "Generic ESP8266 Module" and choose the default flash size 1MB (FS:64KB). Recompile and re-flash and I can store my keys and cert...

20:45:00 CMD: TLSKey 0
20:45:00 RSL: stat/sonoff/RESULT = {"TLSKey1":-1,"TLSKey2":-1}
20:45:11 CMD: TLSKey1 UvBgyCuPr/lKSgwumf/8o/mIsKQPBHn3ZZAGZl4ui9E=
20:45:11 RSL: stat/sonoff/RESULT = {"TLSKey1":32,"TLSKey2":-1}

ascillato, 2019-10-24:
A little detail. You need to use **1MB no spiffs** (those 64kb of spiffs will brake your OTA)

cschnidr, 2019-10-25:
Thank you guys! I used the private key from the [official docu](https://github.com/arendst/Sonoff-Tasmota/wiki/AWS-IoT). So no key revealed from me.
BTW: there is some old wording in the AWS IoT documentation. In the caveats it says that the keys need to embedded in the code, which is obviously no longer needed.

s-hadinger, 2019-10-25:
Lol I didn't recognize the key from the doc.
Correct, it is no longer necessary to embed the key in the code.

s-hadinger, 2019-10-25:
I tried both `esp8266.flash.4m.ld` and `eagle.flash.4m.ld` and the address looks ok, but the content is does not seem to be flash:

```
tlskey0
00:00:08 CMD: tlskey0
405ff400:  00801500 00801500 00801500 00801500
405ff410:  00802d00 00802d00 00802d00 00802d00
405ff420:  00805700 00805700 00805700 00805700
405ff430:  00807e00 00807e00 00807e00 00807e00
```

Tasmota parameters are not written to flash at all. Either my Wemo D1 Mini does not have 4M, or there is an issue in the script.

Also `Status 4` shows: `"ProgramSize":551,"Free":3524`. 551+3524=4075 (which is more than 4096)

I need to dive deeper into it.

s-hadinger, 2019-10-25:
I found the issue. I thought naively that the entire flash was mapped to memory, but only the first 1M is. The rest needs to be read with `spi_flash_read()`.

I need to add some code to support Flash greater than 1M.

arendst, 2019-10-25:
I'm glad you found it. Perhaps move all spi_write calls to spi_flash calls in line with settings code.

s-hadinger, 2019-10-25:
Actually that's not that simple. The goal of reading the TLS key directly from flash is to avoid to copy it to RAM, saving 1k of RAM. But 0x405ff400 cannot be read by the TLS library.

I need the TLS keys to stay at location 0x402ff400. For Flash >1M, that would mean patching the linker scripts to save the last 4k of the first 1M of flash. I'm not sure if there's a generic way to do it.