arendst/Tasmota #6701: Unique topic for each device

jj-uk, 2019-10-20:
6.6.0(release-sonoff) 
2019-07-06T13:10:20
2_3_0/1.5.3(aec24ac9)

**Have you looked for this feature in other issues and in the wiki?**  
yes.
The wiki states: https://github.com/arendst/Sonoff-Tasmota/wiki/MQTT

"For a basic setup you only need to set Host, User and Password but it is recommended to change Topic to avoid issues. Each device should have a unique Topic."

**Is your feature request related to a problem? Please describe.**  
Yes.
My issue is that since the topic should be unique, why does the firmware default this to "sonoff" ?

If a user forgets to set this topic as unique, then adding multiple devices causes them all to report to the same topic, overwriting each other.

**Describe the solution you'd like**  
I think this should be defaulted to "tasmota_%06X" so that each device reports to a unique topic.

Changing the default in the firmware to the MAC address "%06X", or "tasmota_%06X", rather than relying on the user to change this for correct operation seems like the odious thing to do, unless there is another reason they all default to the same topic value "sonoff" ?

I've just set up 10 new tasmota devices. I configured one, then backed up / restored the config to the other 9 so they were all the same.
Whilst googling why this was happening, I found quite a few forum posts that all say the same. "Change the topic to a unique topic". Hence my suggestion to let the firmware set this, by default, to "tasmota_%06X".

Jason2866, 2019-10-20:
I use this setup for my self compiled firmware:
```
// -- Setup your own MQTT settings  ---------------
#undef  MQTT_HOST
#define MQTT_HOST         "mqtt.lan"       // [MqttHost]

#undef  MQTT_PORT
#define MQTT_PORT         1883                   // [MqttPort] MQTT port (10123 on CloudMQTT)

#undef  MQTT_USER
#define MQTT_USER         ""         // [MqttUser] Optional user

#undef  MQTT_PASS
#define MQTT_PASS         ""         // [MqttPassword] Optional password

#undef  MQTT_TOPIC
#define MQTT_TOPIC        PROJECT "-%06X"      // [Topic] (unique) MQTT device topic

#undef  MQTT_CLIENT_ID
#define MQTT_CLIENT_ID    "sonoff-%06X"        // [MqttClient] Also fall back topic using Chip Id = last 6 characters of MAC address
```

arendst, 2019-10-20:
Very, very, very old issue as you confirmed by googling.

The simple reason is that once you setup a default device it will always listen to topic "sonoff". Using the max address as part of the topic makes it impossible for a user to know what the default topic is as he doesn't know the mac address.

Lucky for you this issue has also a soluton you'll like. Every device also subscribes to a backup topic in case a user screws up like you do.

The backup topic contains, oh joy, part of the MAC address so you can still find the device usinf the backup topic.

Happy MQTT'ing.

ascillato2, 2019-10-20:
Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-10-20 by ascillato2

jj-uk, 2019-10-20:
Thanks for the reply but I don't quite understand.  After changing the topic to the mac address, I can now control each device individually. I don't see why this is not the default?