arendst/Tasmota #6703: Show Gosund SP1 monthly usage

meierchen006, 2019-10-20:
Hi,

I run a Gosund SP1 socket with Tasmota firmware 6.6.0.20. 
This shows me on the overview page many measurement data and consumption data:
e.g.
Energy today
Energy Yesterday
Energy overall.

Here I miss the display energy in the month, so that one could see how much energy was consumed so far in the month.

Could you bring this monthly consumption into one of the next firmware?

Kind regards
meierchen006

andrethomas, 2019-10-20:
You can reset the energy totals using the commands documented here https://github.com/arendst/Sonoff-Tasmota/wiki/Power-Monitoring-Calibration

Some people will do this weekly, monthly, annually etc so there's no specific way to know all the possible combinations but using the reset function you can schedule this from your home automation software.

ascillato, 2019-10-20:
Or you can use a home automation software like home assistant or OpenHab

closed 2019-10-20 by ascillato2