arendst/Tasmota #6706: iFan03 - where are the GPIOs????

rhamblen, 2019-10-20:
### ISSUE DESCRIPTION - TROUBLESHOOTING
I've flashed the iFan03 with latest version of the firmware 6.6.0.15
Set up the module and the fan works fine in the web interface.
 
[image]

However, the GPIOs are not listed in the module view (they are there in the template view) and as such when I use the BrettSheleski drivers to use with smarththings, there is an error when looking at the GPIO list. (ie it is a Null because it doesnt report existing).

Any thoughts as to why?  Its not urgent (tbh I got the iFan03 to test and I wont be fitting till later this year)  Just wanted to know if it is me or the f/w?
 
[image]

ascillato, 2019-10-20:
If you select the ifan module using the default module selection, Tasmota won't show any extra gpio as those are not easily accessible for adding sensors or switches. So what you are seeing is the default and normal behavior.

If you are going to use the ifan in a non standard way (at your own risk) you will need to use templating and define there, what gpios you want to expose.

ascillato2, 2019-10-20:
If you need further assistant, please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-10-20 by ascillato2