arendst/Tasmota #6710: Support for Blitzwolf BW-SHP7 3680W

peterforeman, 2019-10-21:
The Blitzwolf BW-SHP7 (dual outlet) is based on a ESP8266 as well. I did not find any documentation online, therefore I wanted to let you know the pins I found so far:

GPIO0 = button2 (left/bottom)
GPIO1 = blue led (inverted)
GPIO13 = red led (inverted)
GPIO12 = button1 (right/top)
GPIO14 = relay1 (top)
GPIO16 = relay2 (bottom)

It should have power monitoring but I have yet to find out how this is connected.

jziolkowski, 2019-10-21:
Thanks for the data. Please add it to our official device template repository at https://blakadder.github.io

jziolkowski, 2019-10-21:
Sorry, pasted link was too short.

https://blakadder.github.io/templates/

closed 2019-10-21 by ascillato2

Ronald1817, 2019-11-22:
I tried this template but:

the led does not switch to red when socket is enabled.
The power measurement  is way off, cos phi 0..02 and high current 15.575 ampere with 75 watt lamp.
i opened the device , it's a BL0937 on the board.
Is there a update fot this profile?
THere's little info, the esp is a 8285 module not a 8266, is has 13 pins 7 on one side and 6 at the other, tried to fiend a pinout but did not find the right one.
tried test template and got one of the leds light up red but the other led did not light up red, , i will look into it further this weekend.

meingraham, 2019-11-22:
We have found that manufacturers change the electronics but retain the same model numbers. So a template that worked in the past may not work on newer devices. Unfortunately it takes some additional investigation to find the new GPIO setup. Since you know most of what is working with the current template, finding a missing LED should be relatively easy to do. Follow [this procedure](https://github.com/arendst/Tasmota/wiki/Configuration-Procedure-for-New-Devices) to find the "missing" LED.

As for the power readings, you need to [calibrate the device](https://github.com/arendst/Tasmota/wiki/Power-Monitoring-Calibration).

Ronald1817, 2019-11-22:
Hi,

Thanx i will try that.

What i think, the cos phi is 1-0.02=0.98 because 0.02 makes no sense with  a 75 watt light bulb attached to it.

I will try this weekend if i can puzzle out the pinout and will post a template.

 

Regards,

Ronald.

Ronald1817, 2019-11-23:
hi, 

i've a small question.
i have found out the following with blitwolf SHP7

The gpio1 pin is needed constantly ON and with gpio4 you control red led 2 for socket 2 and with gpio6 you can control the red led 1 for socket 1.
allso when gpio3=off then with gpio4 you control the blue led 2 for socket 2 and with gpio13 you can control the blu led for socket 1

it's no big deal if the blue led isn't controlled it's fine when socket off = led off and socket on  = led red on for that socket.
can you give me a pointer how to accomplish that gpio 3 = always on?

thanx.

Regards,
Ronald.

meingraham, 2019-11-23:
Your screenshots did not come through.

To control the state of a GPIO, configure it as a relay (e.g., Relay3). Then, when the device starts up, you can control the relay with a Rule:
`On Power3#Boot DO Power3 ON ENDON`

One downside of adding phantom relays is that the web UI will show an additional button for this relay which you may or may not want. But, since the web UI is not intended to be the user interface (it's intended for configuration and management) this is is usually not "seen" anyway.

Mike

Ronald1817, 2019-11-23:
Thanx, i wil try that.

big difference between SHP6 wich works with template and the SHP7 , 
I allso noticed that output 6 needs to be 1 because of the enable of the BL0937 and cos phi works differently.

Ronald1817, 2019-11-23:
Hi,

 

After some trial and error i got the SHP7 working

 

The configuration is based on the Blitwolf(45) because then the power reading is right on target, still must calibrate it, but it’s close to what it should be.

 

The controlling of the leds i did with a rule set.

 

The configuration looks like this

 

 

The ruleset:

 

Rule1 On Power3#Boot DO Power3 ON ENDON ON power1#state=1 do power5 off endon on power1#state=0 do power5 on endon ON power2#state=1 do power4 off endon on power2#state=0 do power4 on endon

 

 

 

The setup is like this:

 

#red led socket 1 is on gpio-13

#red led socket 2 is on gpio-02

 

#gpio-13 is configured as relay4

#gpio-02 is configured as relay5

 

#relay socket 1 is connected to gpio-14

#relay socket 2 is conneected to gpio-16

 

#gpio-3 is common for the leds, 1= control red leds, set it to one with a rule

 

I am glad you gave me the pointers for learning how to implement rules.

 

I don’t know how to extract the template and put is on the template repo, could you this for me?

 

What ‘ve done is just bsased on the Blitzwolfshp(45) and just changed the name in SHP7.

 

I am glad that i got it working and now other people can use it to.

 

Regards,

Ronald.

meingraham, 2019-11-24:
@Ronald1817 

Please enter `Template 255` in the Console and paste the result here.

Ronald1817, 2019-11-24:
Hi,

 

The result is:

 

{"NAME":"Merged","GPIO":[17,23,24,131,134,132,0,0,18,25,21,0,22],"FLAG":0,"BASE":45}

 

 

 

Regards,

Ronald.

meingraham, 2019-11-24:
Sorry, I had edited my post...

Also, the results of:
`Backlog Rule1;Rule2;Rule3`

Ronald1817, 2019-11-24:
Hi,

 

Template:

 

{"NAME":"SHP7","GPIO":[17,23,24,131,134,132,0,0,18,25,21,0,22],"FLAG":0,"BASE":45}

 

Rule1

 

{"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":308,"Rules":"On Power3#Boot DO Power3 ON;Power4 ON;Power5 ON ENDON ON power1#state=1 do power5 off endon on power1#state=0 do power5 on endon ON power2#state=1 do power4 off endon on power2#state=0 do power4 on endon"}

 

Rule2

 

{"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}

 

Rule3

 

{"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}

 

Regards,

Ronald.

meingraham, 2019-11-24:
I think Some combination of `LED<x>` will be better that the current relay setup. Even if not, I think GPIO2 and GPIO13 should be inverted relays (`Relay<x>i`). To make it easier in my mind, I paired Relay1/4 and Relay2/5.

`{"NAME":"SHP7","GPIO":[17,23,57,131,134,132,0,0,18,56,21,0,22],"FLAG":0,"BASE":45}`

There is also typically an LED that indicates the Wi-Fi connectivity (link) status. When the device boots, do you know which of the two LEDs blinked?

Can you explain again how GPIO1 (Relay3) controls the LEDs and the color selection?

This is going to change... but for now, change your rule.
`Rule1 ON Power3#Boot DO Power3 on`

Mike

P.S. FYI - Since you were setting Relay3/4/5 at boot, use `System#Boot` instead of `Power3#Boot`. You are also missing a backlog. But this is going away anyway.

Ronald1817, 2019-11-24:
Hi,

 

#red led socket 1 is on gpio-13

#red led socket 2 is on gpio-02

 

#gpio-13 is configured as relay4

#gpio-02 is configured as relay5

 

#relay socket 1 is connected to gpio-14

#relay socket 2 is conneected to gpio-16

 

#gpio-3 is common for the leds, 1= control red leds, set it to one with a rule

 

The blue leds are controller when gpio3=off

 

Gpio1 = off,  blue led socket 1 is gpio 13, blue led socket 2 is gpio 2 ( both non inverted)

 

It is not possible to control red and blue seperatly for both.

 

When gpio1 is 0ff and gpio  2 and 13 are on , then both blue leds are lit.

 

When gpio1=off and gpio 2 and 13 are off , both red and blue are lit or for one of them., maybe there’s 1 other gpio gpio pin involved to control the blue leds, i have to look for that.

 

Regards,

Ronald.

meingraham, 2019-11-24:
Did you try the updated template and the rule I sent?

Ronald1817, 2019-11-24:
HI, 

 

No i am gonna try right now

 

The switches work, the leds color are off because the rule is not complete yet.

 

Power measurement is good.

 

 

Regards,

Ronald.

meingraham, 2019-11-24:
Using LEDs, you don't need the old rules. It's a matter of arriving are the right GPIO configuration. Tasmota will then manage the LEDs to follow the relays.

When you say that the LED color is off, can you describe further?

I want to pin the template down properly so we can post an updated version. The template as you had it is not the optimal configuration due to having to use multiple rules.

Thanks for working with me to get this pinned down.

Mike

Ronald1817, 2019-11-24:
Hi,

 

I understand

 

What i mean with the color is off , is that at boot both leds are now blue

When i activate socket1 , the led for socket1 is both for a second red and blue after that it remains blue.

 

 

When gpio1 = 0 then the blue leds are controlled with gpio 3 and 13 but the gpio pins of the red leds can not be zero otherwise of a socket red and blue light up.

 

Now when socket 1 is off the led of socket 1 is blue when i switch on socket 1, de led of socket 2 light up red and blue.

The state of led 1 remains blue it is not switched, but when pressing switch 1 shortly the led blue and red of socket 1

 

There’s something strange with the controlling of the leds.

 

Regards,

Ronald.

 

 

vi

meingraham, 2019-11-24:
I want to confirm that you are using the simplified rule that just turns on Relay3 when the device boots; yes?

Are you using MQTT? If not, go into `Configure Other` and disable MQTT.

When the device is booting and it is trying to connect to Wi-Fi, do either of the LEDs blink? Which one? What color?

Ronald1817, 2019-11-24:
Hi,

 

Yes i just disabled mqtt and i use the simplified rule you gave me.

 

The rest stays the same regarding pressing button and led status.

 

Regards,

Ronald.

meingraham, 2019-11-24:
When the device is booting and it is trying to connect to Wi-Fi, do either of the LEDs blink? Which one? What color?

Ronald1817, 2019-11-24:
Yes socket-1 blue- lit, red alernates.

Blue led socket-2 constantly lit.

Ronald1817, 2019-11-24:
I think, the leds are controlled something like this

 

Dont’s know for sure but concerning the red leds this looks like it is the case.

Maybe they used a transistor to drive the leds i don’t know i did not check that.

 

But from playing around with the gpio’s this is the behaviour.

 

 

 

Regards,

Ronald.

Ronald1817, 2019-11-24:
The rule was not ok endon was missing so tried again

 

On boot toggle3 = on

 

Press button 1, led 1 light up red when pressing the button and goes off and relais one switches.

 

Press button 2, led 2 stays, red relais-2 switches

 

Regards

Ronald

Van: Ronald Hofman <ronald@rhofman.net> 
Verzonden: zondag 24 november 2019 20:08
Aan: 'arendst/Tasmota' <reply@reply.github.com>
Onderwerp: RE: [arendst/Tasmota] Support for Blitzwolf BW-SHP7 3680W (#6710), update

 

I think, the leds are controlled something like this

 

Dont’s know for sure but concerning the red leds this looks like it is the case.

Maybe they used a transistor to drive the leds i don’t know i did not check that.

 

But from playing around with the gpio’s this is the behaviour.

 

 

 

Regards,

Ronald.

Ronald1817, 2019-11-24:
little update

The rule was not ok endon was missing so tried again

On boot toggle3 = on

Press button 1, led 1 light up red when pressing the button and goes off and relais one switches.

Press button 2, led 2 becomes red and blue, red relais-2 switches

i think the circuit for the leds looks something like this:

[image]

Ronald1817, 2019-11-24:
gpio 1 is controlling when led red or led blue is selected.

meingraham, 2019-11-25:
Apologies for the rule syntax error!

We might be able to play a "trick" with `LedLink` on `GPIO1`. Use link to "select" the led color. May be a little too clever, but turn off the rule (`Rule1 0`) and try this template.

`{"NAME":"SHP7","GPIO":[17,157,57,131,134,132,0,0,18,56,21,0,22],"FLAG":0,"BASE":45}`

If this doesn't work, try

`{"NAME":"SHP7","GPIO":[17,158,57,131,134,132,0,0,18,56,21,0,22],"FLAG":0,"BASE":45}`

Let me know what those LedLink templates do.

If this trick doesn't work, we'll put `LedLink` on an unused GPIO, reinstate the rule. Then The LEDs will follow the relays and you can select which color by whether you turn `Relay3` on or off.

Mike

Ronald1817, 2019-11-25:
No problem,

 

I will try this this evening, i will let you know.

 

Regards,

Ronald.

Ronald1817, 2019-11-25:
Hi, 

 

Had time to try it and  this works

 

{"NAME":"SHP7","GPIO":[17,158,57,131,134,132,0,0,18,56,21,0,22],"FLAG":0,"BASE":45}

 

 

Rule1=off

 

But mqtt is disabled and the blu leds do not flash.

 

Both leds are off

 

When i press button 1 relais-1 switches and led 1 goes red

 

When i press button 2 relais-2 switches and led 2 goes red

 

 

When i press the switches they shortly are red and blue but only when it is switched.

 

So it is better , no rules needed.

 

Only thing when no mqtt leds are off

 

Power measurement is working fine.

 

The volt reading is off when both relais are off when one the relais are switched to on , de power measurement is enabled.

 

 

 

Regards,

Ronald.

meingraham, 2019-11-25:
Power readings when relays off - `SetOption21`

https://github.com/arendst/Tasmota/wiki/Status-LED

The "trick" with `LedLink` is that I'm using Tasmota to turn on GPIO1 once the device has a connection. Depending on what color you like for the relay LEDs, one can use `LedLink` or `LedLinki`. One aspect of `LedLink` is that it is a signal for BOTH Wi-Fi and MQTT (if enabled). If either one is not connected, then LedLink will blink. 

Are you using MQTT? If not, you need to be sure that you disable it via `Configure Other`. I think we did this earlier, but we've done a lot of testing.

> But mqtt is disabled and the blu leds do not flash.
> Only thing when no mqtt leds are off

Can you explain? Please understand that I have zero visual reference so please be as descriptive as possible.

Thanks.

Mike

Ronald1817, 2019-11-25:
Hi,

Yes mqtt is disabled as per your request with testing.
Rule1 is disabled.

The device has wifi connection but no mqtt zo in principle the blue led's should flash , that is not the case.
is does not worry me when the blue leds are not flashing , because when the connection with ap is lost, the device is acting as a ap so one could always connect to it.

But it is nice to have when the blu led's are flashing ....

The rest is working fine.

Regards,
Ronald.

meingraham, 2019-11-25:
OK... the problem is there is no "dedicated" link LED on the device.

So there are two alternatives:  
1. No link indicator when using the latest template.
2. Revert to the previous template with `Relay3` on `GPIO1` and enable the rule. This should make `Led1` indicate link status. `Led1` will blink if no connection. Blinking LED color may change during boot until the Rule runs to set `GPIO1` state.

Which do you think is preferable?

Mike

Ronald1817, 2019-11-25:
HiI think option 1 because there are no rules needed and the led status correspond with the relay statusI.do not think people will mind.that the leds are not blinkingRegardsRonald

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/arendst/Tasmota/issues/6710?email_source=notifications\u0026email_token=AMTJQPMUXAR3UHMWBFJYCO3QVPTURA5CNFSM4JC4FSZ2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEFCXFQA#issuecomment-558199488",
"url": "https://github.com/arendst/Tasmota/issues/6710?email_source=notifications\u0026email_token=AMTJQPMUXAR3UHMWBFJYCO3QVPTURA5CNFSM4JC4FSZ2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEFCXFQA#issuecomment-558199488",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

meingraham, 2019-11-25:
OK. Could you submit

`{"NAME":"SHP7","GPIO":[17,158,57,131,134,132,0,0,18,56,21,0,22],"FLAG":0,"BASE":45}`

as an new "Blitzwolf SHP7 v2" entry in the [templates repository](https://blakadder.github.io/templates/)?

You have the links for where you ordered it, a photo (which you can copy from the other template if necessary), etc.

Thanks.

Mike

Ronald1817, 2019-11-25:
Hi,

 

I filled in the form and is waits for approval.

 

I wil flash tamosta in a D1 mini  i want to learn the options in the firmware.

I purchased this one on Banggood, put the link in de form and should be good.

 

Thanx for helping me out with the template and other people will be happy they can use it.

 

 

Regards,

Ronalsl

Ronald1817, 2019-11-25:
I tried something 

 

Filled in wrong credentials in mqtt and both blue leds are blinking.

So that allso may be the case when wifi is gone

 

Disabling mqtt sets the led to off.

 

If you switch the reley’s on then both the led are red and blue blinks that’s good.

 

Looks good !

 

Regards

Ronald.

Ronald1817, 2019-11-30:
Hi,

 

I have a small question.

 

I am trying yo use decode-config.py but i keep getting Either argument -d <host> or -f <filename> must be given.

 

I have a dmp file from the tasmota and i want to convert it , what is the correct commandline syntax for this?

 

Thanx.

 

Regards,

Ronald.

 

https://github.com/arendst/Tasmota/wiki/Status-LED

meingraham, 2019-11-30:
Please take this question to Discord.

Ronald1817, 2020-08-02:
Hi,

sorry for the very late reply.
SetOption21 On is indeed working, forgotten to reply to this.
in the meantime a am switched to discord.

regards,
Ronald.