arendst/Tasmota #6711: KNX: How to handle scenes?

egeu5, 2019-10-21:
I did not found anything in the documentation about the implementation and use of scenes in KNX. Scenes are a really basic and necessary feature in KNX and I had no idea how to handle them on a tasmota device. Could somebody please add an example how to programm them or if it is not possible a workaround?

ascillato, 2019-10-21:
KNX Scenes are automations that are configured in your KNX network. Tasmota is an actuator in the KNX network. That configuration is from the KNX Network side. Tasmota will listen to the group addresses you set.

So, you need to configure the SCENE in KNX ([ETS software](https://www.youtube.com/watch?v=TOjemIe8Wgo)) and make there, the list of commands you want to execute when that SCENE is triggered. One of those commands can be Turn ON, Turn OFF your Tasmota device (to the KNX Address you previously set up in your Tasmota from the WEB UI)

You can trigger those scenes also from home automation softwares like [OpenHab](https://community.openhab.org/t/knx-scenes/12936) and [HomeAssistant](https://www.home-assistant.io/integrations/scene.knx/)

Tasmota KNX support is meant for just receiving the KNX commands and executing them, but it can also listen to the group address of the scene if you set it as a KNX RX Slot from the Tasmota KNX WEB UI. In that case, it will trigger a rule EVENT#KNXRX_VAL1 and there you can set the behaviour you want.

So,

Scenes are from the KNX side. You configure them in your KNX Network and those scenes should send commands to the configured **group address to listen** in Tasmota.

ascillato2, 2019-10-21:
If you have further questions, feel free to ask here or in the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). 

Thanks.

---------------------------

### Support Information 
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

egeu5, 2020-04-06:
> KNX Scenes are automations that are configured in your KNX network. Tasmota is an actuator in the KNX network. That configuration is from the KNX Network side. Tasmota will listen to the group addresses you set.
> 
> So, you need to configure the SCENE in KNX ([ETS software - Scene Config Video](https://www.youtube.com/watch?v=TOjemIe8Wgo)) and make there, the list of commands you want to execute when that SCENE is triggered. One of those commands can be Turn ON, Turn OFF your Tasmota device (to the KNX Address you previously set up in your Tasmota from the WEB UI)

Hi, sorry for the late reply. You're video shows a button which is able to do more than sending a scene but that's not the way how scenes working! Furthermore you need a special device which is able to do this. Only some devices can send more values if a scene is selected.

Let's look how scenes are working: You send on a group address a simple number, e.g. 0 for scene 1, 1 for scene 2 and so on. Every device which is listening on this group address react on the scene. How it is reacting is programmed into the different actuators and not the button. On a dim-actuator you could say how bright the light should be on every scene. On a switch actuator you could say if it is turned on or off.

Why should you prefer to do it on this way and not on the one which is shown in the video? Let's look at a typical KNX use case: In every room (which have lots of actuators) you have a off, normal light, and evening-light scene programmed. You can switch those scenes from a switch in the room, a global one for the whole building or one just for the floor. 
Sometimes you have to change a value, e.g. the reading light should be turned on in the normal light scene. If you set the reactions in the different actuators you just have to make one change (go in the actuator and say you have to be on in scene 1). 

If you set the reactions in the different switchs you have to adjust the value in every switch which is not really nice.

egeu5, 2020-04-30:
@ascillato2 @ascillato  Please reopen.

ascillato, 2020-04-30:
Hi, why you want to reopen? 

Scenes are supported. I use them at home. Just use rules to set the behaviour you want in your Tasmota when receiving the scene command. What extra behavior or config do you need?

egeu5, 2020-04-30:
Could you give an example?

ascillato, 2020-04-30:
Yes, of course.

One of the scenes I have is:

## 1 - THE TRIGGER

When the TV is turned on, my home automation software (Home Assistant) detects that in the wifi network and triggers a KNX scene (TV Time) that sends a 1 to the group address 3.1.1

## 2 - DEVICE 1

There is a Tasmota device listening to 3.1.1
That address is configured in Tasmota in a KNX slot (KNX_RX1)
Then, in that device, there is a rule (in rule1) that is triggered when there is a reception on that KNX SLOT and then checks for sunset. If it is night, it will turn on a light. If it is not night, It will enable an internal automation (that is in rule2) to turned on that light when the sun comes down. (there is also another scene that is triggered when the TV is turned off that will disable this internal automation (that is in rule3) ) - This light is very confortable to be turned on while watching TV.

* KNX

[image]

* RULES

```lua
rule1 on event#knxrx_val1 do backlog event checksunrise=%time%;event checksunset=%time%;var1 1 endon on event#checksunrise<%sunrise% do power1 1 endon on event#checksunset>%sunset% do power1 1 endon
rule2 on power1#boot do var1 0 endon on Clock#Timer=5 do event turnon=%var1% endon on event#turnon=1 do power1 1 endon
rule3 on event#knxrx_val2 do var1 0 endon
```

* TIMERS

[image]
_There is also, another Timer that turn off this light 1 hour after sunrise, just in case._

## 3 - DEVICE 2

Also that scene command is received by another Tasmota device that turn off a speaker (that is playing a local radio most of the day but not needed when watching TV)

[image]

```lua
rule1 on event#knxrx_val1 do power1 0 endon
```

------

If you want to make a specific scene, explain it and I will try to help you.

As you can see, Tasmota KNX configuration is manual (you have to set what that device must do). So far is not supported the configuration thru ETS. In ETS a Tasmota device must be set as DUMMY.

------

_**NOTE:** As you can see, I'm not using in these rules the 1 sended in the scene command. **_So far is not implemented processing the scene number._** It is in the TO DO list in order to be used to trigger different things like:_
```
  on event#knxrx_val1=1 do ...
  on event#knxrx_val1=2 do ...
```
_or to be used for another things like:_
```
  on event#knxrx_val1 do dimmer %value% endon
```

egeu5, 2020-05-04:
Hi,
thanks for you're reply. What you describe is a workaround for a scene but not a real scene. You should have a look at KNX datatype 17.001 which describes scene-numbers. In my last reply I discribed the usecase of scenes and how they are working.

> If you want to make a specific scene, explain it and I will try to help you.

Thanks! I need that the Tasmota-device is listening on a group adress which receive different scene numbers, lets say group adress 1/1/1 and scene numbers 0-3. If it is receiving scene number 0 it should turn off, on 1 and 2 it should turn on and on 3 it should turn off. How is that possible?

Tanks for you're work!

ascillato, 2020-05-04:
> I need that the Tasmota-device is listening on a group adress which receive different scene numbers, lets say group adress 1/1/1 and scene numbers 0-3. If it is receiving scene number 0 it should turn off, on 1 and 2 it should turn on and on 3 it should turn off. How is that possible?

From the last part of my last comment. 

_**NOTE:** As you can see, I'm not using in these rules the 1 sended in the scene command. **_So far is not implemented processing the scene number._** It is in the TO DO list in order to be used to trigger different things like:_
```
  on event#knxrx_val1=1 do ...
  on event#knxrx_val1=2 do ...
```
_or to be used for another things like:_
```
  on event#knxrx_val1 do dimmer %value% endon
```

ascillato2, 2020-05-04:
Reopening for tracking the feature request of scene number support.

ascillato2, 2020-05-04:
.

egeu5, 2020-05-05:
Hi,
thanks!
Is there any way to contact you? I want to help implementing more KNX features. I'm not a perfect programmer but probably I can do some work if you give me a short introduction into xdrv_11_knx.ino. 
Greetings

ascillato, 2020-05-06:
> Is there any way to contact you? I want to help implementing more KNX features. 

Cool, we can talk at [Discord](https://discord.gg/Ks2Kzd4). Please PM me. Thanks.

ascillato2, 2020-06-18:
Closing this issue as Scenes has been added on the PR #8728

PR #8728 (2020-06-18): KNX: Add Scene Support
## Description:

Add KNX Scene Support.

If rules are enabled, now the user can define a KNX address for a new KNX SLOT called Scene. This new slot is available for receiving a scene number and also for transmitting a scene number, using KNX DPT 17.001 (1 byte - integer). The Scene number can be any number from 0 to 63 as defined in the KNX standard.

When Tasmota receives a scene number, it will trigger the rule event `KNX_SCENE` with the number of scene received. Example:

```lua
Rule1 1
Rule1 on EVENT#KNX_SCENE=1 do dimmer 50 endon
```

If the user wants to send a sce [... cut]
closed 2020-06-18 by ascillato2