arendst/Tasmota #6713: Templates doesn´t work anymore on the power sockets

nightfly0815, 2019-10-21:
> **GUIDE**
>
> This BUG issue template is meant to REPORT Tasmota software BUGS ONLY>
>  
> Please DO NOT OPEN AN ISSUE:
>  - If your Tasmota version is not the latest from the development branch, please update your device before submitting your issue. Your problem might already be solved. The latest precompiled binaries of Tasmota can be downloaded from http://thehackbox.org/tasmota/  
>  - If your issue is a flashing issue, please address it to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4)  
>  - If your issue is compilation problem, please address it to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4)  
>  - If your issue has been addressed before (i.e., duplicated issue), please ask in the original issue  
>  - If your issue is a Wi-Fi problem or MQTT problem, please try the steps provided in the [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) and troubleshooting wiki articles  
>
> Please take a few minutes to complete the requested information below. Our ability to provide assistance is greatly hampered without it. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help you.

### BUG DESCRIPTION
After the newest update the power Sockets doesn,t work anymore withe the templates.
I have two differnt sockets and the problem is the same at both of them.
One socket is this one : https://github.com/arendst/Sonoff-Tasmota/issues/2223

The other one is this: https://github.com/arendst/Sonoff-Tasmota/wiki/RGB-Smart-Plug-16A
It´s not possible to active the template in the dropdown menu.
When I set a template, the socket reboots and set the template back to sonoff basic modul.

My self compiled image is standard! I just change this line:
env_default = sonoff-minimal, sonoff-DE, sonoff-DE-4MB 
All other settings are standard from the git!

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Sonoff-Tasmota/issues)
- [ x] Searched the problem in the [wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): _____
- [ x] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [x ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x ] Flashing tools used: web
- [ ] Provide the output of command: ``Backlog Template; Module; GPIO``:
  ```
  Configuration output here:
  
  
  ```
- [ ] If using rules, provide the output of this command: ``Backlog Rule1; Rule2; Rule3``:
  ```
  Rules output here:
  
  
  ```
- [ x] Provide the output of this command: ``Status 0``:
  ```
  STATUS 0 output here:
  
19:46:07 MQT: stat/Geschirrspüler/STATUS = {"Status":{"Module":1,"FriendlyName":["Geschirrspüler"],"Topic":"Geschirrspüler","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
19:46:07 MQT: stat/Geschirrspüler/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://192.168.0.30:9999/data/firmwares/sonoff.bin","RestartReason":"Exception","Uptime":"0T00:01:05","StartupUTC":"2019-10-21T17:45:02","Sleep":50,"CfgHolder":4617,"BootCount":52,"SaveCount":299,"SaveAddress":"F9000"}}
19:46:07 MQT: stat/Geschirrspüler/STATUS2 = {"StatusFWR":{"Version":"6.6.0.20(sonoff)","BuildDateTime":"2019.10.21 19:17:26","Boot":4,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)"}}
19:46:07 MQT: stat/Geschirrspüler/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Openwrt_sonoff",""],"TelePeriod":300,"Resolution":"55C180C0","SetOption":["00008009","2805C8000100060000005A00000000170700","00100000"]}}
19:46:07 MQT: stat/Geschirrspüler/STATUS4 = {"StatusMEM":{"ProgramSize":565,"Free":436,"Heap":22,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144068","FlashMode":3,"Features":["00000407","8FDAE397","043683A0","22B617CD","01001BC0","00007081"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
19:46:07 MQT: stat/Geschirrspüler/STATUS5 = {"StatusNET":{"Hostname":"Geschirrspüler","IPAddress":"192.168.0.106","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"EC:FA:BC:2E:04:27","Webserver":2,"WifiConfig":4}}
19:46:07 MQT: stat/Geschirrspüler/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.145","MqttPort":1883,"MqttClientMask":"Geschirrspüler","MqttClient":"Geschirrspüler","MqttUser":"sonoff","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
19:46:07 MQT: stat/Geschirrspüler/STATUS7 = {"StatusTIM":{"UTC":"Mon Oct 21 17:46:07 2019","Local":"Mon Oct 21 19:46:07 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":99,"Sunrise":"08:19","Sunset":"18:50"}}
19:46:07 MQT: stat/Geschirrspüler/STATUS10 = {"StatusSNS":{"Time":"2019-10-21T19:46:07"}}
19:46:07 MQT: stat/Geschirrspüler/STATUS11 = {"StatusSTS":{"Time":"2019-10-21T19:46:07","Uptime":"0T00:01:0
  
  ```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ ``weblog 4`` _for more debug information)_
  ```
  Console output here:
  
  
  ```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
This is the socket 
https://github.com/arendst/Sonoff-Tasmota/wiki/RGB-Smart-Plug-16A
This was the working template for the Vansware RGB Socket
Template: 
{"NAME":"RGB Smart Plug","GPIO":[37,0,39,0,38,134,0,0,131,17,132,21,0],"FLAG":0,"BASE":45}

This was the working eemplate for the Gosund Socket
{"NAME":"Blitz_Test","GPIO":[52,255,57,255,0,134,0,0,131,17,132,21,0],"FLAG":0,"BASE":45}

**(Please, remember to close the issue when the problem has been addressed)**

meingraham, 2019-10-21:
Did you upgrade a working device or is this a new device that you flashed with Tasmota and have specified a template that worked on another similar device you set up in the past?

ascillato2, 2019-10-21:
As you said that you are using all default settings from the codebase, have you tried the precompiled bins (http://thehackbox.org/tasmota/sonoff.bin)

Remember that you can not use 4MB firmware on GOSUND. If you upload a 4MB to a 1MB, then you will need to erase all flash using esptool.py as explained in the wiki and flash again with esptool.py

nightfly0815, 2019-10-21:
No I upgraded both working devices. The templates have worked in the past without any problems.
Yes I know that the 4MB doesn´t work on the socket. The 4MB is for a basic modul where I have changed the flash! ;-) The precompiled minimal image with core 2.5.2 killed the device complete. So now it´s not possible to reach it through the webif....

ascillato, 2019-10-21:
May be you already know, but I will clarify just for anyone else reading this. The minimal firmware is only for 1MB devices. Using it on firmwares with other memory maps like 4MB will stuck the device (as it has 1mb memory map) and the only way to recover is by erasing and flashing by serial with esptool.py as explained in the wiki.
Using 4mb memory map firmware is not really necessary. The precompiled bins of 1mb can be used in all your devices. If you have 4mb flash, you can use the 1mb and will work just fine.

Now, coming back to your issue. If you can't reach the device, I'm sorry but you will need to erase and flash by serial.

The core 2.5.2 has a bug in the memory manager. That is why is not longer supported in Tasmota. The cores 2.4.2 and 2.3.0 have security issues as explained in the comments in platformio.ini file. That is why the core used in the precompiled bins (http://thehackbox.org/tasmota/) is only arduino core pre2.6.0 that is faster, more reliable than the previous cores and don't have any known bug or security issues. If you want to know more about that, please see issue #6666

nightfly0815, 2019-10-21:
Ok I understand. Yes I have to flash the device by serial again. I know that there are some problem's, when you change the core... The running core bevor I tried the precompiled image was STAGE/2.2.2-dev(38a443e).( That one is standard activated at the git)  The only strange thing is that all other devices keep there templates when I had updated it. For exampel the 1 Channel Modul with this template:
{"NAME":"1_Channel_AM23","GPIO":[17,0,0,0,0,0,0,0,21,56,2,0,0],"FLAG":0,"BASE":12}
Or a B1 fake bulb
{"NAME":"B1 Fake","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":1,"BASE":18}

Just the power sockets won´t work with the templates .
Most of the time I compile the firmware and flash it through tasmoadmin, to update alle devices at once with the same firmware.

Jason2866, 2019-10-21:
If flash was not erased completely before flashing Tasmota all kind of strange issues can occur.
This is always the case when Tuya-Convert was used. 
`Reset 5 `should always be done as first step when Tasmota is first time active.
When Tasmota is back, configure the device

nightfly0815, 2019-10-21:
Yes I understand that erasing is most of the time the best decision. But I don´t want do dismount a device to flash it through serial interface! And not every device is easy to dismount. Look at the link in my first post. (The RGB socket is glued) I really wanted to avoid having to open the device. This socket has no screws. I hope now I am able to open it without to destroy it!! 
I will dismount the gosund socket tomorrow and flash it through serial and report my results.

Jason2866, 2019-10-22:
@nightfly0815  There is ALWAYS the risk to brick the device when using Tuya Convert.
This is not a issue from Tasmota. It is the way how Tuya Convert (has) to work.
You are switching firmware, while device is running!, to a firmware which is complete different
from the original. You are operating on open heart....

arendst, 2019-10-22:
Just went over an upgrade from 6.5.0.15 to 6.6.0.20 with your template Blitz-test active without any problem:
```
00:00:00 CFG: Loaded from flash at F4, Count 1408
00:00:00 SRC: Restart
00:00:00 Project sonoff Homecube Version 6.5.0.15(sonoff)-2_4_2
00:00:00 WIF: Attempting connection...
00:00:01 WIF: Network (re)scan started...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Attempting connection...
00:00:02 RSL: tele/homecube/SENSOR = {"Time":"1970-01-01T00:00:02","ENERGY":{"TotalStartTime":"2018-10-31T16:57:01","Total":3.304,"Yesterday":0.016,"Today":0.000,"Power":3.80,"ApparentPower":8.74,"ReactivePower":7.90,"Factor":0.43,"Voltage":224.1,"Current":0.039}}
00:00:03 WIF: Attempting connection...
00:00:04 WIF: Network 0, AP1, SSId indebuurt1, Channel 1, BSSId 24:D3:F2:97:C0:A1, RSSI -76, Encryption 1
00:00:04 WIF: Network 1, AP2, SSId indebuurt2, Channel 5, BSSId A0:AB:1B:7D:42:AC, RSSI -50, Encryption 1
00:00:04 WIF: Network 2, AP-, SSId indebuurt3, Channel 12, BSSId 60:E3:27:58:77:E6, RSSI -85, Encryption 1
00:00:04 WIF: Connecting to AP2 indebuurt2 in mode 11N as homecube...
00:00:04 WIF: Attempting connection...
00:00:05 WIF: Attempting connection...
00:00:06 WIF: Attempting connection...
00:00:07 WIF: Attempting connection...
00:00:08 WIF: Connected
00:00:08 HTP: Web server active on homecube with IP address 192.168.2.168
00:00:08 APP: (UTC) Tue Oct 22 09:12:37 2019, (DST) Sun Mar 31 02:00:00 2019, (STD) Sun Oct 27 03:00:00 2019
11:12:39 MQT: Attempting connection...
11:12:39 MQT: Connected
11:12:39 MQT: tele/homecube/LWT = Online (retained)
11:12:39 MQT: cmnd/homecube/POWER = 
11:12:39 MQT: Subscribe to cmnd/homecube/#
11:12:39 MQT: Subscribe to cmnd/sonoffs/#
11:12:39 MQT: Subscribe to cmnd/DVES_7ED306_fb/#
11:12:39 MQT: tele/homecube/INFO1 = {"Module":"Blitz_Test","Version":"6.5.0.15(sonoff)","FallbackTopic":"cmnd/DVES_7ED306_fb/","GroupTopic":"sonoffs"}
11:12:39 MQT: tele/homecube/INFO2 = {"WebServerMode":"Admin","Hostname":"homecube","IPAddress":"192.168.2.168"}
11:12:39 MQT: tele/homecube/INFO3 = {"RestartReason":"Software/System restart"}
11:12:39 MQT: stat/homecube/RESULT = {"POWER":"ON"}
11:12:39 MQT: stat/homecube/POWER = ON
11:12:39 APP: Boot Count 84
11:12:39 CFG: Saved to flash at FB, Count 1409, Bytes 3584
11:12:47 MQT: tele/homecube/STATE = {"Time":"2019-10-22T11:12:47","Uptime":"0T00:00:18","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":2,"SSId":"indebuurt2","BSSId":"A0:AB:1B:7D:42:AC","Channel":5,"RSSI":92,"LinkCount":1,"Downtime":"0T00:00:08"}}
11:12:47 MQT: tele/homecube/SENSOR = {"Time":"2019-10-22T11:12:47","ENERGY":{"TotalStartTime":"2018-10-31T16:57:01","Total":3.304,"Yesterday":0.016,"Today":0.000,"Period":0.01,"Power":3.70,"ApparentPower":8.79,"ReactivePower":8.00,"Factor":0.42,"Voltage":225.4,"Current":0.039}}
11:12:48 CMD: template 
11:12:48 SRC: WebConsole from 192.168.2.1
11:12:48 RSL: Group 0, Index 1, Command TEMPLATE, Data 
11:12:48 MQT: stat/homecube/RESULT = {"NAME":"Blitz_Test","GPIO":[52,255,57,255,0,134,0,0,131,17,132,21,0],"FLAG":0,"BASE":45}
11:13:11 CMD: upload 1
11:13:11 SRC: WebConsole from 192.168.2.1
11:13:11 RSL: Group 0, Index 1, Command UPLOAD, Data 1
11:13:11 MQT: stat/homecube/RESULT = {"Upload":"Version 6.5.0.15 from http://domus1:80/api/arduino/sonoff.bin"}
11:13:12 CFG: Saved to flash at FB, Count 1410, Bytes 3584

00:00:00 CFG: Loaded from flash at FB, Count 1411
00:00:00 CFG: Saved to flash at FB, Count 1412, Bytes 4096
00:00:00 QPC: Reset
00:00:00 SRC: Restart
00:00:00 Project sonoff Homecube Version 6.6.0.20(sonoff)-STAGE
00:00:00 WIF: Attempting connection...
00:00:01 WIF: Network (re)scan started...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Attempting connection...
00:00:03 WIF: Attempting connection...
00:00:04 WIF: Network 0, AP1, SSId indebuurt1, Channel 1, BSSId 24:D3:F2:97:C0:A1, RSSI -77, Encryption 1
00:00:04 WIF: Network 1, AP2, SSId indebuurt2, Channel 5, BSSId A0:AB:1B:7D:42:AC, RSSI -39, Encryption 1
00:00:04 WIF: Network 2, AP-, SSId indebuurt3, Channel 12, BSSId 60:E3:27:58:77:E6, RSSI -87, Encryption 1
00:00:04 WIF: Connecting to AP2 indebuurt2 in mode 11N as homecube...
00:00:04 WIF: Attempting connection...
00:00:05 WIF: Attempting connection...
00:00:06 WIF: Attempting connection...
00:00:07 WIF: Connected
00:00:07 HTP: Web server active on homecube with IP address 192.168.2.168
11:13:48 NTP: Drift 0, (UTC) Tue Oct 22 09:13:48 2019, (DST) Sun Mar 31 02:00:00 2019, (STD) Sun Oct 27 03:00:00 2019
11:13:48 APP: Boot Count 85
11:13:49 MQT: Attempting connection...
11:13:49 MQT: Connected
11:13:49 MQT: tele/homecube/LWT = Online (retained)
11:13:49 MQT: cmnd/homecube/POWER = 
11:13:49 MQT: Subscribe to cmnd/homecube/#
11:13:49 MQT: Subscribe to cmnd/sonoffs/#
11:13:49 MQT: Subscribe to cmnd/DVES_7ED306_fb/#
11:13:49 MQT: tele/homecube/INFO1 = {"Module":"Blitz_Test","Version":"6.6.0.20(sonoff)","FallbackTopic":"cmnd/DVES_7ED306_fb/","GroupTopic":"sonoffs"}
11:13:49 MQT: tele/homecube/INFO2 = {"WebServerMode":"Admin","Hostname":"homecube","IPAddress":"192.168.2.168"}
11:13:49 MQT: tele/homecube/INFO3 = {"RestartReason":"Software/System restart"}
11:13:49 MQT: stat/homecube/RESULT = {"POWER":"ON"}
11:13:49 MQT: stat/homecube/POWER = ON
11:13:49 CFG: Saved to flash at FA, Count 1413, Bytes 4096
11:13:57 MQT: tele/homecube/STATE = {"Time":"2019-10-22T11:13:57","Uptime":"0T00:00:19","UptimeSec":19,"Heap":29,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":2,"SSId":"indebuurt2","BSSId":"A0:AB:1B:7D:42:AC","Channel":5,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:09"}}
11:13:57 MQT: tele/homecube/SENSOR = {"Time":"2019-10-22T11:13:57","ENERGY":{"TotalStartTime":"2018-10-31T16:57:01","Total":3.304,"Yesterday":0.016,"Today":0.000,"Period":0.01,"Power":4.00,"ApparentPower":8.74,"ReactivePower":7.80,"Factor":0.46,"Voltage":224.1,"Current":0.039}}
```
So expect it to work just fine.

ascillato2, 2019-10-23:
Putting this issue ON HOLD until the user has time to perform the test. Thanks.

closed 2019-10-23 by ascillato2

nightfly0815, 2019-10-24:
Hey, I could fix the two sockets!
I flashed the pre compiled Image on the Gosund through serial und try the templates without any problems.
Then I used my compiled version from the git and updated it through the webif. The socket also worked without any problems with the templates.
The RGB socket isn´t repairable because it´s not possible to dismount it without destroying the housing. But I have two of them with the same effect with the non working templates.
I could repair the other one with flashing an older self compiled image (Sonoff-Tasmota-782563b3ae1301b91759bcdc8f7cfcea221f3be0) This Version is one week old it is 6.6.0.18.
I first flashed the self compiled minimal image and then the normal version. With this version the templates worked as it should. After that I flashed the actual self compiled firmware 6.6.0.21. First the minimal then the normal version through the web interface.
And it looks like that it works again.
So I´m not sure where the failure come from. I tried to reproduce it. I changed the template to the generic modul. At this template I tryied to set the GPIO´s Manual (Relay, Button etc.)
I was able to set alle settings except the HLWBL CF1 and the BL0937 CF. When i set one of them an reboot the socket the socket jumps back to sonoff basic and lost all settings for buttons, relay, led etc...
I hope the failure doesn´t come again with the new versions. I will give feedback here if the failure comes again!

Thanks for helping.