arendst/Tasmota #6714: Setoption68  2 = RGB+W option ( Treat PWM1-3 and 4 as two separate channels.

homeseer666, 2019-10-21:
**Have you looked for this feature in other issues and in the wiki?**   yes

**Is your feature request related to a problem? Please describe.**  

The use case is : using a RGBW controller with a RGB led strip on PWM1-3 and a single white strip on PWM4.

When using RGBW (PWM 1-4 -setoptions 15, 68 enabled),  channel4 resets to zero when a RRGGBB command is sent (without WW).
Also, "HSBcolor" is disabled when setoption68 is enabled.

**Describe the solution you'd like**  

Option 1 : 

- Add Setoption68  2 = RGB+W option ( Treat PWM1-3 and 4 as two separate channels. )
- Color still works to set all channels at once.
- HSB still works to set PWM1-3

Option 2:

with RGBW device (PWM 1-4)
when "color RRGGBB"  command sent (WW undefined) keep the WW value **unchanged**. (vs resetting WW to 0)

Option 3: 
option to have an "unchanged" value token (U or other)  Ex: "color 0,255,0,U" which would turn RGB color to green without touching the existing WW value.

Option 4: any other option, ex: amending setoption37 as already proposed by @s-hadinger on the Tasmota support chat

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

Setoption37 128 + using HSBcolor for PWM1-3 and Channel4 for PWM4
downside : losing ability to Power on/off RGB and W independantly .
(the benefit with option68 is that you can turn off rgb or white separately and when you turn on again it keeps the existing value.  with option37 you can only turn off all at once or use 0/100 values on channel 4 but obviously lose the dim value. )

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2019-10-22:
As discussed on Discord, I will look at it in the coming days.

homeseer666, 2019-10-24:
Sounds good. Thank you for the prompt response, 
let me know once it is ready for testing, I'll be glad to do some testing on my end. 
Again not familiar with the Github platform, so maybe I'll be notified automatically...
Cheers.

Gio-dot, 2019-10-26:
I, is this issue fixed in 6.7.1?

ascillato, 2019-10-26:
No, that is why this issue is still open.

Gio-dot, 2019-10-26:
Mmmh so i didn't understand the 6.7.1 change here
https://github.com/arendst/Sonoff-Tasmota/commit/7bafb0f897fbbd5ae8a0dac039b4545318bd7733 at row 290?

ascillato, 2019-10-26:
powerdelta exception has nothing to do with setoption68 and RGB

Gio-dot, 2019-10-26:
@ascillato powerdelta?? I'm not speaking about it but about the next row... 290-291 whre option 68 is mentioned... 😥

ascillato, 2019-10-26:
No, that line is not related to this issue. Again. This issue is open because is not implemented yet.

PR #6799 (2019-10-31): Fix better control of RGB/White when SetOption37 >128, added Dimmer1 and Dimmer2 commands
## Description:

- Better control of RGB/White when SetOption37 >128. Now you see two disctinct devices that you can power on/off with `Power1` and `Power2`.
- I added `Dimmer2` command to change only dimmer of white channels. `Dimmer` or `Dimmer1` will only change RGB brightness.
- Changes to `SetOption37` will trigger a reboot, it is required in most cases.
- Also added `Channel<x> +` and `Channel<x> -` to add/substract 10 to brightness channel (range is still 0..100). This is an old request from @jziol [... cut]
closed 2019-10-31 by arendst